realsenseai/realsense-ros #2999: unable to find package realsense2_camera / realsense2_camera_node

IamShubhamGupto, 2024-02-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435                       | 
| Firmware Version                   | 2.55.0                        | 
| Operating System & Version |   { Linux (Ubuntu 22)  } | 
| Kernel Version (Linux Only)    |  5.15.122-tegra aarch64                                                                    | 
| Platform                                 | NVIDIA Jetson AGX Orin                           |
| Librealsense SDK Version       |  { 2.55.0 }                                                    | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  Robot                           | 
| ROS Distro			         |  Iron                                        | 
| RealSense ROS Wrapper Version   |  ros2-development                                              | 

### Issue Description
I am building a docker image which will run ros2 iron wrapper of realsense. my base image : nyuultraviolet/cv-dockerfile:jp60-pytorch21-iron-tensorrt-realsense

Since I already have ros2 iron and librelsense installed using the repository https://github.com/dusty-nv/jetson-containers, I have only Step 3 of the installation required for my docker image. However, it does not work when I copy the commands into the Dockerfile.

When I try option1 of Step 3: `apt install install ros-iron-realsense2-*` the command executes successfully but I cannot run any of the Usage commands given in the readme.md

When I try option2 of Step 3, building from source, I get the following error:
```
colcon ERROR colcon build: Duplicate package names not supported:
- librealsense2
- librealsense
- usr/local
```

I am unable to install the ROS2 wrapper using either methods in my dockerfile.

I tested Option2 of Step 3 inside the shell of nyuultraviolet/cv-dockerfile:jp60-pytorch21-iron-tensorrt-realsense and it worked so it seems to be a path issue when directly running these commands inside the Dockerfile.

Thank you for your help

MartyG-RealSense, 2024-02-08:
Hi @IamShubhamGupto  It sounds as though you may have more than one installation of librealsense on your computer that are conflicting with each other.  If you already have librealsense installed before installing the ROS wrapper then I recommend building the wrapper from source code instead of using the `apt install` package installation method (which installs both librealsense and the wrapper at the same time from packages).

IamShubhamGupto, 2024-02-08:
@MartyG-RealSense thanks, I'll update this thread soon. im planning to follow the instructions on this Dockerfile https://github.com/2b-t/realsense-ros2-docker

MartyG-RealSense, 2024-02-08:
@IamShubhamGupto  Thanks very much.  I look forward to your next report.  Good luck!

IamShubhamGupto, 2024-02-09:
@MartyG-RealSense I was able to run ros2 wrapper and realsense SDK inside a docker container using ros2 humble and the above repository, for whatever reason Iron did not want to be compiled on the AGX orin.

closed 2024-02-09 by IamShubhamGupto

MartyG-RealSense, 2024-02-09:
I'm pleased to hear that you were able to achieve a solution with Humble instead of Iron.  Thanks very much for the update!