realsenseai/realsense-ros #3003: Realsense D435i keeps disconnecting on Orange Pi 5

cml1010101, 2024-02-11:
We have been having issues with using the D435i via the ROS2 Wrapper. The camera works for say a minute or two, disconnects, and won't reconnect, even when plugged in for a while. librealsense is at version 2.54.1, built from source and using -DFORCE_LIBUVC. Realsense ros is built from source, in ros2 iron, and using 4.54.1. I am launching the node via a composable node. I have set align_depth.enable to true. I have also set rgb_camera.profile to 1280,720,15. The depth image is set to defaults. This is running in a privileged docker container, and the host has also libuvc built so that the udev rules work. I have tried powered usb hubs and cables that are most likely capable of carrying the necessary amperage. We are using an orange pi 5, with the debian orange pi image and thus cannot use the kernel back end. The kernel is I believe 5.10.?-rockchip. It is also recognized as usb 2.1 most of the time. I have also seen this issue using libuvc backend on my pc. This is being used in a robotics application.

MartyG-RealSense, 2024-02-12:
Hi @cml1010101  libuvc or its successor **-DFORCE_RSUSB_BACKEND=TRUE** are the recommended methods of installing the librealsense SDK on RockChip-based computing devices.  This is because these installation methods bypass the Linux kernel, making the build non-dependent on Linux versions or kernel versions and not requiring the application of a kernel patch.

If the composable node launch works correctly for a minute or two however then it sounds as though the launch method works and some factor is causing the camera to become disconnected.

Are you starting docker in sudo admin powers mode with **sudo docker run** please?

cml1010101, 2024-02-12:
I am using docker compose with docker granted sudo permissions via groupadd. I have set privileged to true via the docker-compose.yaml for the ros2_swervy container. The ros2_devel container is used for vscode, and I removed the privileged option from it to prevent the development container from grabbing the camera accidentally.

I will try rebuilding librealsense to use the -DFORCE_RSUSB_BACKEND to avoid these issues later today.

Here is the docker-compose.yaml and Dockerfile right now:
```services:
  ros2_devel:
    build:
      context: docker
    image: ghcr.io/northernforce/pifiles:main
    command: tail -f /dev/null
    working_dir: /workspaces/rosnfr
    volumes:
      - .:/workspaces/rosnfr
      - /tmp/.X11-unix:/tmp/.X11-unix
      - ~/.Xauthority:/root/.Xauthority
    network_mode: host
    # privileged: true
    environment:
      DISPLAY: $DISPLAY
  ros2_swervy:
    image: ghcr.io/northernforce/pifiles:main
    command: ros2 launch nfr_crescendo swervy.launch.py
    working_dir: /workspaces/rosnfr
    volumes:
      - .:/workspaces/rosnfr
    network_mode: host
    privileged: true
  ros2_test:
    image: ghcr.io/northernforce/pifiles:main
    command: ros2 launch nfr_crescendo sim.launch.py
    working_dir: /workspaces/rosnfr
    volumes:
      - .:/workspaces/rosnfr
    privileged: true

FROM ubuntu:22.04
RUN apt update && apt install -y locales && locale-gen en_US en_US.UTF-8 && update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
ENV LANG=en_US.UTF-8
RUN apt install -y software-properties-common
RUN apt-add-repository universe
RUN apt update && apt install curl -y && curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key \
    -o /usr/share/keyrings/ros-archive-keyring.gpg && echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] \
    http://packages.ros.org/ros2/ubuntu $(. /etc/os-release && echo $UBUNTU_CODENAME) main" | tee /etc/apt/sources.list.d/ros2.list > /dev/null
ENV DEBIAN_FRONTEND=NONINTERACTIVE
RUN apt update && apt upgrade -y && apt install -y ros-iron-ros-base python3-argcomplete && apt update && apt install -y ros-dev-tools
WORKDIR /opt/realsense
RUN git clone https://github.com/IntelRealSense/librealsense -b v2.54.1
RUN apt update && apt install git cmake libssl-dev freeglut3-dev libusb-1.0-0-dev pkg-config libgtk-3-dev unzip -y
RUN cd librealsense && mkdir build && cd build && cmake .. -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release
RUN cd librealsense/build && make -j6
RUN cd librealsense/build && make install
WORKdir /opt/ros/realsense
RUN mkdir -p src && cd src && git clone https://github.com/IntelRealSense/realsense-ros -b 4.54.1
RUN . /opt/ros/iron/setup.sh && rosdep init && rosdep update && rosdep install --ignore-src --from-paths src -y --skip-keys librealsense2
RUN . /opt/ros/iron/setup.sh && colcon build --merge-install --install-base /opt/ros/iron
RUN rm -rf src build log
RUN apt update && apt install -y ros-iron-apriltag
RUN apt update && apt install -y ros-iron-apriltag-ros
RUN apt update && apt install -y ros-iron-image-proc
RUN apt update && apt install -y ros-iron-navigation2
RUN apt update && apt install -y python3 python3-pip libssh-dev libprotobuf-dev
RUN apt update && apt install -y generate-ninja ninja-build protobuf*
RUN pip3 install jinja2
WORKDIR /tmp
RUN git clone https://github.com/wpilibsuite/allwpilib.git -b v2024.2.1
RUN cd allwpilib && mkdir build && cd build && cmake .. -DWITH_JAVA=OFF -DWITH_WPILIB=OFF -DWITH_EXAMPLES=OFF \
    -DWITH_TESTS=OFF -DWITH_GUI=OFF -DWITH_SIMULATION_MODULES=OFF -DWITH_JAVA_SOURCE=OFF
RUN cd allwpilib/build && make -j$(nproc)
RUN cd allwpilib/build && make install
RUN rm -rf allwpilib
RUN apt update && apt install -y nlohmann-json3-dev
RUN apt update && apt install -y ros-iron-usb-cam ros-iron-v4l2-camera
RUN apt update && apt install -y ros-iron-fuse*
#RUN ln -s /opt/ros/iron/lib/x86_64-linux-gnu/liburdfdom_model.so.3.0 /opt/ros/iron/lib/liburdfdom_model.so.3.0
RUN ln -s /opt/ros/iron/lib/aarch64-linux-gnu/liburdfdom_model.so.3.0 /opt/ros/iron/lib/liburdfdom_model.so.3.0
RUN ldconfig
RUN mkdir -p /workspaces
COPY scripts/entrypoint.sh /workspaces/entrypoint.sh
RUN chmod +x /workspaces/entrypoint.sh
ENTRYPOINT [ "/workspaces/entrypoint.sh" ]

MartyG-RealSense, 2024-02-12:
Thanks very much for the update and Dockerfile, @cml1010101 - I look forward to your next report. Good luck!

cml1010101, 2024-02-12:
Hi @MartyG-RealSense,

I rebuilt on my pc for now (I won't be able to use the pi until later today) using the rsusb backend option. I did the same for the docker image. However, it does not disconnect, but it only sends frames every two seconds or so, and prints these error messages in the meantime. Thanks so much for your help.

[log.txt](https://github.com/IntelRealSense/realsense-ros/files/14250802/log.txt)

cml1010101, 2024-02-12:
I believe I have found the cause of this new issue. I believe that the camera, due to a possibly bad cable, is still being detected as usb 2.1, and there is too much data trying to be transferred over usb 2.1. I adjusted the parameters to specify that I only wanted depth and color, and I was able to achieve a consistent stream. I might investigate more into possibilities for getting usb 3.0, but for now the solution appears to have worked. Thanks so much for your help @MartyG-RealSense

closed 2024-02-12 by cml1010101

MartyG-RealSense, 2024-02-12:
It's great to hear that you achieved a solution, @cml1010101 - thanks very much for the update!