realsenseai/realsense-ros #3006: Unsupported device! Product ID: 0x0B4B, while running roslaunch realsense2_camera rs_camera.launch

proxi666, 2024-02-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D405 / D415 / D435 / D435i / D455 / D457 }                          | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |   { Win (10/11) / Linux (Ubuntu 18/20/22) / MacOS } | 
| Kernel Version (Linux Only)    |  (e.g. 5.4)                                                                       | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..                           |
| Librealsense SDK Version       |  { 2.<?>.<?> }                                                    | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  {Iron/Humble/Rolling/etc.. }                                           | 
| RealSense ROS Wrapper Version   |  {4.51.1, 4.54.1, etc..}                                                  | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hello, I am trying to make my Intel RealSense D435i RGBD camera but I am getting an error "Unsupported device! Product ID: 0x0B4B"
Please do guide me to resolve this error 

Regards.

Here is the terminal copy for reference:

**roslaunch realsense2_camera rs_camera.launch**

... logging to /home/proxi/.ros/log/63aeebb4-cb50-11ee-9495-1f95b65f82db/roslaunch-proxi-pc-6003.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://proxi-pc:45759/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
[... 210 lines cut ...]
    Depth	  640x360	@ 90Hz	   Z16
    Depth	  640x360	@ 60Hz	   Z16
    Depth	  640x360	@ 30Hz	   Z16
    Depth	  640x360	@ 15Hz	   Z16
    Depth	  640x360	@ 6Hz	   Z16
    Depth	  480x270	@ 90Hz	   Z16
    Depth	  480x270	@ 60Hz	   Z16
    Depth	  480x270	@ 30Hz	   Z16
    Depth	  480x270	@ 15Hz	   Z16
    Depth	  480x270	@ 6Hz	   Z16
    Depth	  424x240	@ 90Hz	   Z16
    Depth	  424x240	@ 60Hz	   Z16
    Depth	  424x240	@ 30Hz	   Z16
    Depth	  424x240	@ 15Hz	   Z16
    Depth	  424x240	@ 6Hz	   Z16
    Depth	  256x144	@ 300Hz	   Z16
    Depth	  256x144	@ 90Hz	   Z16

Stream Profiles supported by Motion Module
 Supported modes:
    stream       resolution      fps       format   
    Accel	 N/A		@ 200Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 100Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 400Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 200Hz	   MOTION_XYZ32F

MartyG-RealSense, 2024-02-14:
Hi @proxi666  The camera's firmware driver is identifying your D435i as a **D430i**.  This means that the firmware is not detecting the camera's RGB sensor.  Whilst the librealsense SDK recognizes the D430i mode, the RealSense ROS1 wrapper (2.3.2) does not support D430i.

D430i support was added to the ROS2 wrapper in its 4.54.1 version but this support is not included in the ROS1 wrapper as its development ceased after the 2.3.2 version in November 2021 and only the ROS2 wrapper continues to be updated.  You will therefore need to try to get the camera to be detected as D435i.

I note that the camera is also detected as D430i in the **rs-enumerate-devices** tool.

Your camera firmware version is 5.15.1.0.  Using this firmware with SDK 2.50.0 will cause errors.  The firmware version that should be used with 2.50.0 is **5.13.0.50**, so please downgrade your camera's firmware to that version.

proxi666, 2024-02-14:
Thank you for your fast response, I'll try what you said, and get back to you!

proxi666, 2024-02-15:
@MartyG-RealSense 
Hello,
I did as what you have mentioned, but still its coming as D430i, please do guide me what else I can do...
[image]

I tried unplugging and restarting the realsense-viewer but still getting it as D430i 
[image]

MartyG-RealSense, 2024-02-15:
Although your firmware is 5.13.0.50, it is being used in the above image with the **2.54.2** version of the RealSense Viewer instead of the 2.50.0 version of it.

When the ROS1 wrapper and librealsense are installed together at the same time with apt-get (Method 1) then the RealSense Viewer tool (the 2.50.0 version of it) is not included with librealsense.  

The easiest way to check whether or not the camera can be detected as D435i when using the 5.13.0.50 firmware would be to repeat the **rs-enumerate-devices** command to see whether it still lists the camera as D430i.

proxi666, 2024-02-15:
Hey, thankyou for your quick responce

Here is what I get, apparently its showing me D430i...
I also although deleted all the packages related to ROS wrapper, and rn I only have l Intel® RealSense™ SDK 2.0 pkgs, Please do guide me how to resolve this error...

Thanks.

**rs-enumerate-devices**

Device info: 
    Name                          : 	Intel RealSense D430I
    Serial Number                 : 	152222072718
    Firmware Version              : 	5.13.0.50
    Recommended Firmware Version  : 	5.15.1
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb4/4-1/4-1:1.0/video4linux/video0
    Debug Op Code                 : 	15
    Advanced Mode                 : 	NO
    Product Id                    : 	0B4B
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	152523062056
    Firmware Update Id            : 	152523062056

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format   
    Infrared 1	  1280x800	@ 30Hz	   Y8
    Infrared 1	  1280x800	@ 25Hz	   Y16
    Infrared 1	  1280x800	@ 15Hz	   Y16
    Infrared 1	  1280x800	@ 15Hz	   Y8
    Infrared 1	  1280x720	@ 30Hz	   Y8
    Infrared 1	  1280x720	@ 15Hz	   Y8
    Infrared 1	  1280x720	@ 6Hz	   Y8
    Infrared 1	  848x480	@ 90Hz	   Y8
    Infrared 1	  848x480	@ 60Hz	   Y8
    Infrared 1	  848x480	@ 30Hz	   Y8
    Infrared 1	  848x480	@ 15Hz	   Y8
    Infrared 1	  848x480	@ 6Hz	   Y8
    Infrared 1	  848x100	@ 300Hz	   Y8
    Infrared 1	  848x100	@ 100Hz	   Y8
    Infrared 1	  640x480	@ 90Hz	   Y8
    Infrared 1	  640x480	@ 60Hz	   Y8
    Infrared 1	  640x480	@ 30Hz	   Y8
    Infrared 1	  640x480	@ 15Hz	   Y8
    Infrared 1	  640x480	@ 6Hz	   Y8
    Infrared 1	  640x400	@ 25Hz	   Y16
    Infrared 1	  640x400	@ 15Hz	   Y16
    Infrared 1	  640x360	@ 90Hz	   Y8
    Infrared 1	  640x360	@ 60Hz	   Y8
    Infrared 1	  640x360	@ 30Hz	   Y8
    Infrared 1	  640x360	@ 15Hz	   Y8
    Infrared 1	  640x360	@ 6Hz	   Y8
    Infrared 1	  480x270	@ 90Hz	   Y8
    Infrared 1	  480x270	@ 60Hz	   Y8
    Infrared 1	  480x270	@ 30Hz	   Y8
    Infrared 1	  480x270	@ 15Hz	   Y8
    Infrared 1	  480x270	@ 6Hz	   Y8
    Infrared 1	  424x240	@ 90Hz	   Y8
    Infrared 1	  424x240	@ 60Hz	   Y8
    Infrared 1	  424x240	@ 30Hz	   Y8
    Infrared 1	  424x240	@ 15Hz	   Y8
    Infrared 1	  424x240	@ 6Hz	   Y8
    Infrared 2	  1280x800	@ 30Hz	   Y8
    Infrared 2	  1280x800	@ 25Hz	   Y16
    Infrared 2	  1280x800	@ 15Hz	   Y16
    Infrared 2	  1280x800	@ 15Hz	   Y8
    Infrared 2	  1280x720	@ 30Hz	   Y8
    Infrared 2	  1280x720	@ 15Hz	   Y8
    Infrared 2	  1280x720	@ 6Hz	   Y8
    Infrared 2	  848x480	@ 90Hz	   Y8
    Infrared 2	  848x480	@ 60Hz	   Y8
    Infrared 2	  848x480	@ 30Hz	   Y8
    Infrared 2	  848x480	@ 15Hz	   Y8
    Infrared 2	  848x480	@ 6Hz	   Y8
    Infrared 2	  848x100	@ 300Hz	   Y8
    Infrared 2	  848x100	@ 100Hz	   Y8
    Infrared 2	  640x480	@ 90Hz	   Y8
    Infrared 2	  640x480	@ 60Hz	   Y8
    Infrared 2	  640x480	@ 30Hz	   Y8
    Infrared 2	  640x480	@ 15Hz	   Y8
    Infrared 2	  640x480	@ 6Hz	   Y8
    Infrared 2	  640x400	@ 25Hz	   Y16
    Infrared 2	  640x400	@ 15Hz	   Y16
    Infrared 2	  640x360	@ 90Hz	   Y8
    Infrared 2	  640x360	@ 60Hz	   Y8
    Infrared 2	  640x360	@ 30Hz	   Y8
    Infrared 2	  640x360	@ 15Hz	   Y8
    Infrared 2	  640x360	@ 6Hz	   Y8
    Infrared 2	  480x270	@ 90Hz	   Y8
    Infrared 2	  480x270	@ 60Hz	   Y8
    Infrared 2	  480x270	@ 30Hz	   Y8
    Infrared 2	  480x270	@ 15Hz	   Y8
    Infrared 2	  480x270	@ 6Hz	   Y8
    Infrared 2	  424x240	@ 90Hz	   Y8
    Infrared 2	  424x240	@ 60Hz	   Y8
    Infrared 2	  424x240	@ 30Hz	   Y8
    Infrared 2	  424x240	@ 15Hz	   Y8
    Infrared 2	  424x240	@ 6Hz	   Y8
    Depth	  1280x720	@ 30Hz	   Z16
    Depth	  1280x720	@ 15Hz	   Z16
    Depth	  1280x720	@ 6Hz	   Z16
    Depth	  848x480	@ 90Hz	   Z16
    Depth	  848x480	@ 60Hz	   Z16
    Depth	  848x480	@ 30Hz	   Z16
    Depth	  848x480	@ 15Hz	   Z16
    Depth	  848x480	@ 6Hz	   Z16
    Depth	  848x100	@ 300Hz	   Z16
    Depth	  848x100	@ 100Hz	   Z16
    Depth	  640x480	@ 90Hz	   Z16
    Depth	  640x480	@ 60Hz	   Z16
    Depth	  640x480	@ 30Hz	   Z16
    Depth	  640x480	@ 15Hz	   Z16
    Depth	  640x480	@ 6Hz	   Z16
    Depth	  640x360	@ 90Hz	   Z16
    Depth	  640x360	@ 60Hz	   Z16
    Depth	  640x360	@ 30Hz	   Z16
    Depth	  640x360	@ 15Hz	   Z16
    Depth	  640x360	@ 6Hz	   Z16
    Depth	  480x270	@ 90Hz	   Z16
    Depth	  480x270	@ 60Hz	   Z16
    Depth	  480x270	@ 30Hz	   Z16
    Depth	  480x270	@ 15Hz	   Z16
    Depth	  480x270	@ 6Hz	   Z16
    Depth	  424x240	@ 90Hz	   Z16
    Depth	  424x240	@ 60Hz	   Z16
    Depth	  424x240	@ 30Hz	   Z16
    Depth	  424x240	@ 15Hz	   Z16
    Depth	  424x240	@ 6Hz	   Z16
    Depth	  256x144	@ 300Hz	   Z16
    Depth	  256x144	@ 90Hz	   Z16

Stream Profiles supported by Motion Module
 Supported modes:
    stream       resolution      fps       format   
    Accel	 N/A		@ 200Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 100Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 400Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 200Hz	   MOTION_XYZ32F

MartyG-RealSense, 2024-02-15:
In the past, a downgrade of the firmware has been one way to correct the D430i mis-detection.  You have downgraded the firmware though and the issue has not been resolved.

Has the camera always been detected as D430i since it was purchased or did it initially detect as D435i and later become D430i please?  

There is an extremely rare circumstance where the RGB sensor may not be detected if a small cable inside the camera that connects the RGB sensor to the camera circuit board has become detached.  If the camera is less than a year old though then opening up the camera to check the cable would likely invalidate the camera's sales warranty.

proxi666, 2024-02-17:
This product is brand new, yet I'm encountering the same issue while using ROS2 Foxy. I would greatly appreciate your guidance in resolving this error.

Thanks

MartyG-RealSense, 2024-02-17:
Which ROS2 wrapper version are you using Foxy with, please?  The current latest version **4.54.1** is the only one that has D430i support, and that wrapper version does not support being used with Foxy (it supports only Humble and Iron).

proxi666, 2024-02-18:
I was going with this link
"https://github.com/IntelRealSense/realsense-ros/tree/ros2-legacy"
and tried both ways to run my sensor...
Will switching to Humble might solve my error?

MartyG-RealSense, 2024-02-18:
If downgrading your firmware driver has not resolved the issue and the camera has been detected as D430i since the day that it was purchased then it is likely a hardware problem that cannot be fixed and requires the return of the camera to the retailer that you purchased it from to obtain a replacement camera under warranty.

proxi666, 2024-02-19:
alright, let me try switching to ROS2 Humble, if that didnt work, then i'll try returning/exchanging the product

proxi666, 2024-02-21:
I have given the product for replacement.

MartyG-RealSense, 2024-02-21:
Thanks very much for the update!

Coinstine, 2024-07-17:
hi mate, did you solve the problem? My d430i happend a same problem and the function is good on windows with realsense-viewer, but when i run roslaunch realsense....  on ubuntu20.04 system, just like the problem u described: Unsupported device! Product ID: 0x0B4B. plz help

MartyG-RealSense, 2024-07-17:
Hi @Coinstine  Support was added for D430i in the RealSense ROS2 wrapper from wrapper version 2.54.1 onwards.

Support for D430i was not added for the ROS1 wrapper 2.3.2 though as development of the ROS1 wrapper has ceased and only the ROS2 wrapper continues to be updated, as advised at https://github.com/IntelRealSense/realsense-ros/pull/2321

If a camera is detected as D430i then it usually indicates that an RGB sensor in a D435i camera has not been detected.  However, there are RealSense-equipped robot products that are purposely D430i because they do not include an RGB sensor in the robot.