realsenseai/realsense-ros #3009: Stereo IR using USB 2.0

dakejahl, 2024-02-16:
Is it possible to stream both infra1 and infra2 using USB 2.0? It does not appear to be possible, but I haven't found any documentation about it. I've tried enabling the streams and reducing the resolution and framerate but I still only see infra1
```
    realsense_camera_node = Node(
        name='camera',
        namespace='camera',
        package='realsense2_camera',
        executable='realsense2_camera_node',
        parameters=[{
                'enable_infra1': True,
                'enable_infra2': True,
                'enable_color': False,
                'rgb_camera.profile': '640x480x30',
                'enable_depth': False,
                'depth_module.emitter_enabled': 0,
                'depth_module.profile': '480x270x15',
                'enable_gyro': False,
                'enable_accel': False,
        }]
    )
```

MartyG-RealSense, 2024-02-17:
Hi @dakejahl  I can confirm that Infra2 is only available on a USB3 connection.

dakejahl, 2024-02-17:
@MartyG-RealSense is there any way to enable both? Or is this a limitation of the hardware?

MartyG-RealSense, 2024-02-17:
It is a limitation.  Both Infra1 and Infra2 are only available on  USB3.

closed 2024-02-20 by dakejahl