realsenseai/realsense-ros #3010: set rgb_camera.exposure value in rs_launch

patrickwasp, 2024-02-18:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i                   | 
| Firmware Version                   | 5.14.0                         | 
| Operating System & Version |   Ubuntu 22.04 | 
| Kernel Version (Linux Only)    |  6.1.35                                                                    | 
| Platform                                 | PC                           |
| Librealsense SDK Version       |  v2.54.1                                                   | 
| Language                               |  n/a        | 
| Segment			         |  others                             | 
| ROS Distro			         |  Humble                                   | 
| RealSense ROS Wrapper Version   |  4.54.1-1jammy                                                 | 

### Issue Description

I see that the new version of the wrapper has added the option to set the "enable_auto_exposure" value, but I don't see how I would set it to a specific value if it was set to false. Is there a corresponding argument like `rgb_camera.exposure`?

And since it seems https://github.com/IntelRealSense/realsense-ros/issues/2798 you can't set an upper limit when auto exposure is enabled, are there any suggested strategies to implement that functionality? 

```yaml
launch: 

- include: 
    file: "$(find-pkg-share realsense2_camera)/launch/rs_launch.py"
    arg:
    -
        name: "enable_color"
        value: "true"
    - 
        name: "rgb_camera.profile"
        value: "1920x1080x6"
    -
        name: "rgb_camera.enable_auto_exposure"
        value: "false"
```

MartyG-RealSense, 2024-02-18:
Hi @patrickwasp  Yes, you should be able to set a manual RGB exposure value with `rgb_camera.exposure`

I am not aware of a method that will allow limits to be set for RGB exposure when auto-exposure is enabled, unfortunately.

patrickwasp, 2024-02-19:
`rgb_camera.exposure` doesn't seem to change anything, can you point me to the code that reads these arguments? 

This is the launch file I use:

```yaml

launch: 

- include: 
    file: "$(find-pkg-share realsense2_camera)/launch/rs_launch.py"
    arg:
    - 
        name: "json_file_path"
        value: "$(find-pkg-prefix bringup)/config/realsense_high_accuracy.json"
    - 
        name: "initial_reset"
        value: "true"
    -
        name: "camera_name"
        value: "depth"
    -
        name: "enable_depth"
        value: "true"
    -
        name: "align_depth.enable"
        value: "true"
    -
        name: "enable_color"
        value: "true"
    - 
        name: "rgb_camera.profile"
        value: "1920x1080x6"
    -
        name: "rgb_camera.enable_auto_exposure"
        value: "false"
    -
        name: "rgb_camera.exposure"
        value: "1"
    - 
        name: "depth_module.profile"
        value: "848x480x6"
    -
        name: "enable_infra1"
        value: "false"
    -
        name: "enable_infra2"
        value: "false"
    -
        name: "infra_rgb"
        value: "false"
    - 
        name: "colorizer.enable"
        value: "false"
    -
        name: "clip_distance"
        value: "-2."
    -
        name: "enable_gyro"
        value: "true"
    - 
        name: "enable_accel"
        value: "true"
    -
        name: "enable_pose"
        value: "true"
    -
        name: "gyro_fps"
        value: "0"
    -
        name: "accel_fps"
        value: "0"
    -
        name: "pointcloud.enable"
        value: "false"
    -
        name: "pointcloud.stream_filter"
        value: "2"
    -
        name: "pointcloud.stream_index_filter"
        value: "0"
```

MartyG-RealSense, 2024-02-19:
Most of the functioning of the ROS wrapper happens in the **base_realsense_node** file of its source code.

https://github.com/IntelRealSense/realsense-ros/blob/c0a674cc0c829b8cf171210e2120e2c7273525bc/realsense2_camera/src/base_realsense_node.cpp

MartyG-RealSense, 2024-02-27:
Hi @patrickwasp  Do you require further assistance with this case, please?  Thanks!

tonynajjar, 2024-05-31:
> [realsense2_camera_node-2] [WARN] [1717159431.231466651] [realsense_left]: Could not set param: rgb_camera.exposure with nan Range: [nan, nan]: parameter 'rgb_camera.exposure' could not be set: Parameter {rgb_camera.exposure} doesn't comply with floating point range.

I'm on 4.55.1 and get this warning. `rgb_camera.exposure` is not even available in the [launch file](https://github.com/IntelRealSense/realsense-ros/blob/4.55.1/realsense2_camera/launch/rs_launch.py)

MartyG-RealSense, 2024-05-31:
Hi @tonynajjar  Are you able to set the RGB exposure during runtime after launch using the command below, please?

`ros2 param set /camera/camera rgb_camera.exposure 78`

tonynajjar, 2024-05-31:
No, I get

> ros2 param set /realsense_left rgb_camera.exposure 78
Setting parameter failed: parameter 'rgb_camera.exposure' cannot be set because it was not declared

MartyG-RealSense, 2024-05-31:
I will enquire to one of my Intel RealSense colleagues.

@Nir-Az  4.55.1 wrapper user @tonynajjar has been unable to set the RGB exposure parameter, even with **ros2 param set**. Do you have advice that you can offer, please?  Thanks!

SamerKhshiboun, 2024-06-02:
Hi @tonynajjar,

Make sure your are in a well sourced new terminal ( `source /opt/ros/humble/setup.bash` ) and run this command:
`ros2 param list <your_node>`

Please print out the output here.

Thanks,
Samer

tonynajjar, 2024-06-02:
```
ros2 param list /perception/front_left | grep rgb_camera
  rgb_camera.auto_exposure_roi.bottom
  rgb_camera.auto_exposure_roi.left
  rgb_camera.auto_exposure_roi.right
  rgb_camera.auto_exposure_roi.top
  rgb_camera.color_format
  rgb_camera.color_profile
  rgb_camera.enable_auto_exposure
  rgb_camera.frames_queue_size
  rgb_camera.gain
  rgb_camera.global_time_enabled
```

Here is the result. I really don't think it's a local issue of mine, aren't you able to reproduce it? I don't see where the param `rgb_camera.exposure` is declared

SamerKhshiboun, 2024-06-02:
A small note, regarding the launch file, it is not mandatory that every parameter should be there.
Usually we put parameters there which we need them to be editable via the launch command.

Anyway,
Can you please try these 2 things:

1)
terminal1: `ros2 run realsense2_camera realsense2_camera_node`
terminal2: `ros2 param list /camera/camera`
and send the output

2)
another thing I want you to try, is to launch the node with debug information and provide me the output in a file:
add this param to your launch command:
`--log-level:=debug`

the output can be huge, so you can redirect it to a file and attach it to here

Thanks

tonynajjar, 2024-06-02:
1.
 ```
ros2 param list /camera/camera | grep rgb_camera
  rgb_camera.auto_exposure_roi.bottom
  rgb_camera.auto_exposure_roi.left
  rgb_camera.auto_exposure_roi.right
  rgb_camera.auto_exposure_roi.top
  rgb_camera.color_format
  rgb_camera.color_profile
  rgb_camera.enable_auto_exposure
  rgb_camera.frames_queue_size
  rgb_camera.gain
  rgb_camera.global_time_enabled
```

2. Here is the point of interest

```
[DEBUG] [1717321249.764950797] [rcl]: Subscription taking message
[DEBUG] [1717321249.764967309] [rcl]: Subscription take succeeded: true
[DEBUG] [1717321249.765069742] [camera.camera]: setParam::Setting parameter: depth_module.auto_exposure_roi.right
[DEBUG] [1717321249.765126607] [rcl]: Subscription taking message
[DEBUG] [1717321249.765141584] [rcl]: Subscription take succeeded: true
[DEBUG] [1717321249.765248497] [camera.camera]: setParam::Setting parameter: depth_module.auto_exposure_roi.top
[DEBUG] [1717321249.765305586] [rcl]: Subscription taking message
[DEBUG] [1717321249.765320946] [rcl]: Subscription take succeeded: true
[DEBUG] [1717321249.765430292] [camera.camera]: setParam::Setting parameter: depth_module.auto_exposure_roi.bottom
[DEBUG] [1717321249.765486325] [rcl]: Subscription taking message
[DEBUG] [1717321249.765501237] [rcl]: Subscription take succeeded: true
[DEBUG] [1717321249.765660791] [camera.camera]: Set RGB Camera as VideoSensor.
[DEBUG] [1717321249.769635154] [camera.camera]: Declare: DOUBLE::rgb_camera.exposure=nan  [0, 0]
[DEBUG] [1717321249.769711091] [camera.camera]: setParam::Setting parameter: rgb_camera.exposure
[WARN] [1717321249.769943831] [camera.camera]: Could not set param: rgb_camera.exposure with nan Range: [nan, nan]: parameter 'rgb_camera.exposure' could not be set: Parameter {rgb_camera.exposure} doesn't comply with floating point range.
[DEBUG] [1717321249.798109913] [camera.camera]: Declare: INT::rgb_camera.gain=0  [0, 128]
[DEBUG] [1717321249.798175162] [camera.camera]: setParam::Setting parameter: rgb_camera.gain
[DEBUG] [1717321249.798345404] [rcl]: Subscription taking message
[DEBUG] [1717321249.798413757] [rcl]: Subscription take succeeded: true
[DEBUG] [1717321249.800550557] [camera.camera]: Declare: BOOL::rgb_camera.enable_auto_exposure=True  [0, 1]
[DEBUG] [1717321249.800627902] [camera.camera]: setParam::Setting parameter: rgb_camera.enable_auto_exposure
```

SamerKhshiboun, 2024-06-02:
from this `[DEBUG] [1717321249.769635154] [camera.camera]: Declare: DOUBLE::rgb_camera.exposure=nan  [0, 0]`
I can see that rgb_camera.exposure is trying to be declared as DOUBLE, although it should be INTEGER
Let me debug this further and update you asap.

SamerKhshiboun, 2024-06-02:
Meanwhile, can you check if you can set the color exposure through the realsense-viewer GUI tool ?
Let me know also if you get any debug/error message in the debug window/pane in the viewer

tonynajjar, 2024-06-02:
Unfortunately I can't check with realsense viewer, I only have ssh access to the computer connected to the realsense. Maybe consider reopening this issue until this is investigated?

Arun-Prasad-V, 2024-06-04:
@tonynajjar, could you also please share below info:

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D405 / D415 / D435 / D435i / D455 / D457 }                          | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |   { Win (10/11) / Linux (Ubuntu 18/20/22) / MacOS } | 
| Kernel Version (Linux Only)    |  (e.g. 5.4)                                                                       | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..                           |
| Librealsense SDK Version       |  { 2.<?>.<?> }                                                    | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  {Iron/Humble/Rolling/etc.. }                                           | 
| RealSense ROS Wrapper Version   |  {4.51.1, 4.54.1, etc..}                                                  |

tonynajjar, 2024-06-04:
Required Info	
Camera Model	 D457 
Firmware Version	5.16.0.1
Operating System & Version	Ubuntu 20.04.6 LTS
Kernel Version (Linux Only)	5.10.104-tegra
Platform	NVIDIA Orin NX
Librealsense SDK Version	2.55.1
Language	ROS driver
Segment	Robot
ROS Distro	Humble
RealSense ROS Wrapper Version	 4.55.1

Arun-Prasad-V, 2024-06-04:
@tonynajjar, thanks for the info.

For D457 model, there is already an open PR https://github.com/IntelRealSense/librealsense/pull/12251 related to this issue.
If possible, please try it.

MartyG-RealSense, 2024-06-16:
Case closed due to no further comments received.

closed 2024-06-16 by MartyG-RealSense

Nir-Az, 2024-06-16:
The PR was merged, this should hopefully take care of this issue.
Please build librealsense development branch from source to get it or pend for a next ROS release including it.