realsenseai/realsense-ros #3011: Custom node that publishes the Realsense's Image data

henryburon, 2024-02-18:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i                          | 
| Firmware Version                   | 5.15.1                          | 
| Operating System & Version |   Ubuntu 22.04.3 LTS | 
| Kernel Version (Linux Only)    |  5.15.0-1045-raspi                                                                       | 
| Platform                                 | Raspberry Pi 4 Model B 8 GB                           |
| Librealsense SDK Version       |  2.54.1                                                    | 
| Language                               |  Python         | 
| Segment			         |  Robot                             | 
| ROS Distro			         |  Iron                                           | 

### Issue Description

I am attempting to create a ROS2 node that used the librealsense Python API to fetch image data from the IntelRealsense camera and publish it on a topic.

I was successful in creating the node, and when I run it on my laptop, I am able to do: `ros2 run rqt_image_view rqt_image_view` and see the video.

However, when I run it on the Raspberry Pi, it sort of publishes one message, then further messages get dropped, and I am never able to view the video feed.

So, my question is, has anyone been able to successfully write a custom node that uses the librealsense Python API to fetch the image data, publish it, and view it somewhere else?

The issue might be that the Image message is too computationally intensive for the Rapberry Pi to publish, even though I adjusted the FPS down to 15 and the publishing frequency to 1 Hz. Is there a way to compress/modify the image from the Intel Realsense so it is lightweight and can be published easily by the RPi? Any input appreciated, thank you.

### When I echo the topic the image is published on, this is what I see:

header:
  stamp:
    sec: 1708245965
    nanosec: 203004265
  frame_id: color_frame
height: 480
width: 640
encoding: bgr8
is_bigendian: 0
step: 1920
data:
- 127
- 91
- 29
- 127
- 91
- 29
- 127
- 87
- 37
- 130
- 90
- 39
- 135
- 91
- 46
- 134
- 90
- 45
- 131
- 90
- 42
- 128
- 88
- 40
- 133
- 89
- 32
- 133
- 89
- 32
- 138
- 91
- 33
- 138
- 91
- 33
- 138
- 92
- 31
- 138
- 92
- 31
- 136
- 90
- 43
- 136
- 90
- 43
- 129
- 92
- 46
- 129
- 92
- 46
- 127
- 91
- 42
- 129
- 93
- 44
- 137
- 92
- 44
- 139
- 94
- 47
- 143
- 96
- 45
- 141
- 95
- 43
- 152
- 103
- 31
- 148
- 100
- 28
- 155
- 104
- 12
- 162
- 111
- 19
- 173
- 120
- 14
- 171
- 117
- 12
- 175
- 121
- 4
- 175
- 121
- 4
- 154
- 110
- 26
- 154
- 110
- 26
- 138
- 100
- 43
- 133
- 96
- 38
- 131
- 94
- 41
- 131
- 94
- 41
- 128
- 94
- 38
- 128
- 94
- 38
- 129
- 95
- 41
- 129
- 95
- 41
- 133
- 92
- '...'
---
A message was lost!!!
	total count change:2
	total count: 2---
A message was lost!!!
	total count change:2
	total count: 4---
A message was lost!!!
	total count change:4
	total count: 8---
A message was lost!!!
	total count change:4
	total count: 12---

MartyG-RealSense, 2024-02-18:
Hi @henryburon  A Pi 4 user at https://github.com/IntelRealSense/realsense-ros/issues/2329#issuecomment-1109819912 found that installing the **image_transport** plugin enabled compressed topics to be published.  

The official ROS2 documentation page at the link below suggests installing this plugin with a command for your particular version of ROS2.  For example:

**Foxy**
`sudo apt install ros-foxy-image-transport`

**Foxy**
`sudo apt install ros-humble-image-transport`

henryburon, 2024-02-18:
I was able to finally get the Raspberry Pi to publish the Intel Realsense's image with a custom ros2 node. The final issue was that messages were being dropped, and the solution ended up being lowering the resolution that the Intel Realsense was publishing at down to 424, 240.

I have copied my node down below if anyone would like to see it for reference:
```
import rclpy
from rclpy.node import Node
from sensor_msgs.msg import Image
import pyrealsense2 as rs
import cv2
import numpy as np

class Fetch_Camera(Node):
   def __init__(self):
      super().__init__("fetch_camera")

      # Publishers
      self.camera_pub = self.create_publisher(Image, "robot_camera", 10)

      # Timers
      self.camera_timer = self.create_timer(1/15, self.publish_image)

      # Initialization

      # Configure depth and color streams
      self.pipeline = rs.pipeline()
      self.config = rs.config()

      # Get device product line for setting a supporting resolution
      # self.pipeline_wrapper = rs.pipeline_wrapper(self.pipeline)
      # self.pipeline_profile = self.config.resolve(self.pipeline_wrapper)
      # self.device = self.pipeline_profile.get_device()
      # self.device_product_line = str(self.device.get_info(rs.camera_info.product_line))

      # self.config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

      # This works on the Raspberry Pi when streaming at 424, 240
      self.config.enable_stream(rs.stream.color, 424, 240, rs.format.bgr8, 6) # (424, 240)

      self.pipeline.start(self.config)

   # Called at 100 Hz
   def publish_image(self):
      frames = self.pipeline.wait_for_frames()
      color_frame = frames.get_color_frame()
      if not color_frame:
         return
      
      # Convert images to numpy arrays
      color_image = np.asanyarray(color_frame.get_data())

      # Create Image message
      msg = Image()
      msg.header.stamp = self.get_clock().now().to_msg()
      msg.header.frame_id = "color_frame"
      msg.height = color_image.shape[0]
      msg.width = color_image.shape[1]
      msg.encoding = 'bgr8'
      msg.is_bigendian = False
      msg.step = color_image.shape[1] * color_image.shape[2]
      msg.data = color_image.tobytes()

      self.camera_pub.publish(msg)
      
      

def fetch_camera_entry(args=None):
   rclpy.init(args=args)
   node = Fetch_Camera()
   rclpy.spin(node)
   rclpy.shutdown()```

MartyG-RealSense, 2024-02-27:
Hi @henryburon  Do you require further assistance with this case, please?  Thanks!

henryburon, 2024-02-28:
No, I don't. Thank you!

closed 2024-02-28 by henryburon

MartyG-RealSense, 2024-02-28:
You are very welcome, @henryburon - thanks very much for the update!

Jabtwin, 2024-08-21:
Hi @henryburon , I also have the same problem as you. I've already changed the resolution like yours,( 424, 240). However, the problems remain, this is the post from my friend, please support us if you are there
https://github.com/IntelRealSense/librealsense/issues/13272
This is my final project, so pls :<<