realsenseai/realsense-ros #3012: node stops publishing data

patrickwasp, 2024-02-19:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i                   | 
| Firmware Version                   | 5.15.1                         | 
| Operating System & Version |   Ubuntu 22.04 | 
| Kernel Version (Linux Only)    |  6.1.35                                                                    | 
| Platform                                 | PC                           |
| Librealsense SDK Version       |  v2.54.1                                                   | 
| Language                               |  n/a        | 
| Segment			         |  others                             | 
| ROS Distro			         |  Humble                                   | 
| RealSense ROS Wrapper Version   |  4.54.1-1jammy                                                 | 

### Issue Description

When running a realsense node, the camera publishes an image or two and then stops. The camera works fine in `realsense-viewer`.

This is my setup:

```dockerfile
FROM ros:humble

# use bash instead of sh
SHELL ["/bin/bash", "-c"]
ENV DEBIAN_FRONTEND=noninteractive

RUN sudo apt-get update && \
    sudo apt-get install -y \
    ros-humble-realsense2-camera \ 
    ros-humble-realsense2-description \
    && rm -rf /var/lib/apt/lists/*

# add packages to path
RUN echo "source /opt/ros/$ROS_DISTRO/setup.bash" >> ~/.bashrc
```

output of `ros2 launch realsense2_camera rs_launch.py`:
```
# ros2 launch realsense2_camera rs_launch.py 
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-02-19-03-04-57-571767-6b5e89bc827d-232
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [233]
[camera.camera]: RealSense ROS v4.54.1
[camera.camera]: Built with LibRealSense v2.54.1
[camera.camera]: Running with LibRealSense v2.54.1
[camera.camera]: Device with serial number 108322072360 was found.

[camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video0 was found.
[camera.camera]: Device with name Intel RealSense D435I was found.
[camera.camera]: Device with port number 2-1.3 was found.
[camera.camera]: Device USB type: 3.2
[camera.camera]: getParameters...
[camera.camera]: JSON file is not provided
[camera.camera]: Device Name: Intel RealSense D435I
[camera.camera]: Device Serial No: 108322072360
[camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video0
[camera.camera]: Device FW version: 5.15.1
[camera.camera]: Device Product ID: 0x0B3A
[camera.camera]: Sync Mode: Off
[camera.camera]: Set ROS param depth_module.profile to default: 848x480x30
[camera.camera]: Set ROS param rgb_camera.profile to default: 1280x720x30
[camera.camera]: Set ROS param gyro_fps to default: 200
[camera.camera]: Set ROS param accel_fps to default: 63
[camera.camera]: Stopping Sensor: Depth Module
[camera.camera]: Starting Sensor: Depth Module
[camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[camera.camera]: Stopping Sensor: RGB Camera
[camera.camera]: Starting Sensor: RGB Camera
[camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1]  19/02 03:05:03,199 WARNING [140467224172096] (backend-v4l2.cpp:1510) Frames didn't arrived within 5 seconds
[realsense2_camera_node-1] [WARN] [1708311903.199449414] [camera.camera]: XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.70831e+12,Warn,Frames Timeout
```

patrickwasp, 2024-02-19:
I replaced the USB cable and now the node works fine. The exact same type of USB cable, I bought 2.

closed 2024-02-19 by patrickwasp

MartyG-RealSense, 2024-02-19:
It's great to hear that a cable change resolved your node problem.  Thanks very much for the update!