realsenseai/realsense-ros #3013: roslaunch realsense2_camera rs_camera.launch

Meiyi6, 2024-02-20:
When i launch a launch file using command roslaunch realsense2_camera re_camera.launch then recieving this error ,here are the complete terminal output.

... logging to /home/cmy/.ros/log/ff1fb86c-cfc9-11ee-ab00-a86daa550b9f/roslaunch-cmy-Alienware-m15-27630.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cmy-Alienware-m15:37613/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: 848
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/ordered_pc: False
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/reconnect_timeout: 6.0
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/stereo_module/exposure/1: 7500
 * /camera/realsense2_camera/stereo_module/exposure/2: 1
 * /camera/realsense2_camera/stereo_module/gain/1: 16
 * /camera/realsense2_camera/stereo_module/gain/2: 16
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method: 
 * /camera/realsense2_camera/usb_port_id: 
 * /camera/realsense2_camera/wait_for_device_timeout: -1.0
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[camera/realsense2_camera_manager-1]: started with pid [27663]
process[camera/realsense2_camera-2]: started with pid [27664]
[ INFO] [1708418187.838340736]: Initializing nodelet with 12 worker threads.
[ERROR] [1708418187.871474501]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Failed to load library /home/cmy/realsense_ws/devel/lib//librealsense2_camera.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libopencv_imgcodecs.so.3.2: cannot open shared object file: No such file or directory)
[ERROR] [1708418187.871527567]: The error before refreshing the cache was: Failed to load library /home/cmy/realsense_ws/devel/lib//librealsense2_camera.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libopencv_imgcodecs.so.3.2: cannot open shared object file: No such file or directory)
[FATAL] [1708418187.871682585]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera-2] process has died [pid 27664, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/cmy/.ros/log/ff1fb86c-cfc9-11ee-ab00-a86daa550b9f/camera-realsense2_camera-2.log].
log file: /home/cmy/.ros/log/ff1fb86c-cfc9-11ee-ab00-a86daa550b9f/camera-realsense2_camera-2*.log

What should i do?

MartyG-RealSense, 2024-02-20:
Hi @Meiyi6  Which procedure did you use to install the librealsense SDK and the RealSense ROS1 wrapper, please?

Did you use **Method 1** at the link below to install the wrapper with **sudo apt-get install** please? 

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy?tab=readme-ov-file#method-1-the-ros-distribution

If you did use sudo apt-get install then did you install librealsense first before installing the wrapper?  If you did then this could cause a problem.  This is because Method 1 installs both librealsense and the wrapper at the same time.  So if librealsense was installed first and then Method 1 was used, you could have two conflicting installations of librealsense on the same computer.

Meiyi6, 2024-02-21:
Hi @MartyG-RealSense , thanks for your help , but I used **Method 2**  , so i don't knoww how to resolve ?

MartyG-RealSense, 2024-02-21:
A 'Method 1' installation may work better for you.  This method installs the librealsense SDK and the ROS1 wrapper at the same time.

First, please uninstall the Method 2 wrapper that you installed earlier by completely deleting the **/catkin_ws/src** catkin workspace folder.

Then install librealsense and the ROS1 wrapper with the instruction below.

`sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`

Meiyi6, 2024-02-21:
Hi @MartyG-RealSense , thanks for your advice agian . 
I tried the method you recommended, but still have the same problem

MartyG-RealSense, 2024-02-21:
Thank you for testing Method 1.

Which Ubuntu version are you using, please?  The most suitable Ubuntu for the ROS1 wrapper is 16.04 or 18.04.

Meiyi6, 2024-02-22:
I am using 18.04.

MartyG-RealSense, 2024-02-22:
The launch seems to be crashing when it reaches the **Initializing nodelet** line of the log.  A healthy launch of the ROS1 wrapper should resemble the lines below at this stage.  As the next line after 'Initializing nodelet' detects the ROS1 wrapper version, this suggests that there could be a problem with the wrapper installation on your computer.

```
process[camera/realsense2_camera_manager-1]: started with pid [13859]
process[camera/realsense2_camera-2]: started with pid [13860]
[ INFO] [1698166980.813586508]: Initializing nodelet with 4 worker threads.
[ INFO] [1698166980.940533004]: RealSense ROS v2.3.2
[ INFO] [1698166980.940576091]: Built with LibRealSense v2.50.0
[ INFO] [1698166980.940598773]: Running with LibRealSense v2.50.0
```

MartyG-RealSense, 2024-03-02:
Hi @Meiyi6  Do you require further assistance with this case, please?  Thanks!

Meiyi6, 2024-03-06:
> Hi @Meiyi6 Do you require further assistance with this case, please? Thanks!
Yes ， i need !

Meiyi6, 2024-03-06:
> The launch seems to be crashing when it reaches the **Initializing nodelet** line of the log. A healthy launch of the ROS1 wrapper should resemble the lines below at this stage. As the next line [...]
Ok, thank you. I'll try to reinstall the wrapper .

MartyG-RealSense, 2024-03-06:
Which RealSense camera model are you using, please?

Does it make a difference if you try a different launch file, such as the launch instruction below?

`roslaunch realsense2_camera rs_aligned_depth.launch`

Meiyi6, 2024-03-07:
> Which RealSense camera model are you using, please?
> 
> Does it make a difference if you try a different launch file, such as the launch instruction below?
> 
> `roslaunch realsense2_camera rs_aligned_depth.launch`

The camera model I am using is realsense D435i  .

MartyG-RealSense, 2024-03-07:
When a RealSense ROS wrapper problem defies all attempts to fix i, sometimes the only fix that works is to wipe the computer clean (including Linux) and do a completely fresh reinstall of everything, and then the problem may disappear due to the reinstall removing a hidden problem that was causing the failure.  I appreciate that uninstalling Ubuntu is not ideal for everyone though if you have files that you might lose.

closed 2024-03-21 by MartyG-RealSense