realsenseai/realsense-ros #3017: D435i rosbag record does not record /camera/imu

Styazoua, 2024-02-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       |  D435i  }                          | 
| Firmware Version                   | (05.13.00.50)                          | 
| Operating System & Version |  Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  5.4.0-150-generic                                                                       | 
| Platform                                 | PC                          |
| Librealsense SDK Version       |  { 2.<?>.<?> }                                                    | 
| ROS Distro			         |  Melodic                                       | 

### Issue Description
Hello I have a problem with my realsense D435i.
I am trying to record a bag file with the 2 infrared topics and the combined imu topic.

Here are the commands I used to record the bagfile:

```
roslaunch realsense2_camera rs_camera.launch unite_imu_method:=copy

rosbag record -O aufnahme4.bag /camera/infra1/image_rect_raw /camera/infra2/image_rect_raw /camera/imu
```
The problem that I am facing is when I run `rosbag info aufnahme4.bag` it shows me that the `/camera/imu` topic hasn't been recorded.

```
path:        aufnahme4.bag
version:     2.0
duration:    9.8s
start:       Feb 22 2024 14:55:19.09 (1708610119.09)
end:         Feb 22 2024 14:55:28.90 (1708610128.90)
size:        173.0 MB
messages:    590
compression: none [197/197 chunks]
types:       sensor_msgs/Image [060021388200f6f0f447d0fcd9c64743]
topics:      /camera/infra1/image_rect_raw   295 msgs    : sensor_msgs/Image
             /camera/infra2/image_rect_raw   295 msgs    : sensor_msgs/Image

```

Then I tried:
```
roslaunch realsense2_camera rs_camera.launch unite_imu_method:=linear_interpolation
```
and the result was the same as before.
Is there any known way I can fix that? I looked into other issues but wasn't able to find a solution for this problem.

MartyG-RealSense, 2024-02-22:
Hi @Styazoua  The Accel and Gyro IMU topics are disabled by default in the RealSense ROS wrapper.  You can enable them in your roslaunch instruction.  The **unite_imu_method** should then combine the two individual topics into a single **/imu** topic.

`roslaunch realsense2_camera rs_camera.launch enable_accel:=true enable_gyro:=true unite_imu_method:=copy`

Styazoua, 2024-02-23:
Thank you! I will close that issue

closed 2024-02-23 by Styazoua

MartyG-RealSense, 2024-02-23:
You are very welcome.  I'm pleased that I could help.  Thanks very much for the update!