realsenseai/realsense-ros #3018: The alignment problem between depth image and color image

HeadReaper-hc, 2024-02-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | {D435i }                          |                         | 
| Operating System & Version |   { Linux (Ubuntu 20/22) | 
| Platform                                 | NVIDIA Jetson   |                                                 | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| ROS Distro			         |  {Humble}                                           |                                                   | 

### Issue Description
Hi, I set 'align_depth:=true' in rs_launch.py and get follow result:
[image]
The top left is the aligned depth image, the top right is the original depth image, and the bottom left is the color image.
The difference between the aligned depth image and the color image is significant. I would like to know if this is normal.

MartyG-RealSense, 2024-02-23:
Hi @HeadReaper-hc  The color texture does not look as though it is appearing at all on the aligned image.

Is there any difference if you enable alignment in the launch instruction instead of in rs_launch.py?

`ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true`

HeadReaper-hc, 2024-02-26:
@MartyG-RealSense Thank you for your reply. I tried this but have no difference

MartyG-RealSense, 2024-02-26:
Which version of the RealSense ROS2 wrapper are you using, please?

HeadReaper-hc, 2024-02-27:
@MartyG-RealSense 4.54.1 and I use it in docker.

MartyG-RealSense, 2024-02-27:
Another RealSense user with a Jetson at https://github.com/IntelRealSense/realsense-ros/issues/3022 using 4.54.1 and Docker has also been having problems when align_depth was enabled.  They found that their problem no longer occurred if they disabled CUDA in their librealsense source code build using the CMake flag **-DBUILD_WITH_CUDA=false**

HeadReaper-hc, 2024-02-28:
@MartyG-RealSense Thankyou for your advice. I will try this method and update the issue.

MartyG-RealSense, 2024-03-13:
Hi @HeadReaper-hc  Do you have an update about this case that you can provide, please?  Thanks!

HeadReaper-hc, 2024-03-13:
@MartyG-RealSense Yes. I recompiled the librealsense with your advice but nothing changed.
I rechecked the depth map using a rainbow display and I have found many places where depth does not exist, such as next to the cup. So I think this may be the reason why the cup looks very wide in the aligned depth map, as it displays a grayscale image.

Uploading depth-vis.mp4…

MartyG-RealSense, 2024-03-13:
Does the Viewer's 'rainbow' image improve if you change the Visual Preset setting at the top of the Viewer's side-panel from 'Default' to **Medium Density**.  The Medium Density preset provides a good balance between depth accuracy and the amount of detail on the depth image.

HeadReaper-hc, 2024-03-14:
@MartyG-RealSense There seems to be no change.
[
[image]
](url)

MartyG-RealSense, 2024-03-14:
It looks as though your office's ceiling lights may be of the **fluorescent ** type.  If they are fluorescent  then this could be a source of noise, as such lights can flicker at a frequency that is difficult to see with the human eye. 

Changing the Power Line Frequency option (under **Stereo Module > Controls** in the Viewer) can help to negate this effect. For North American regions, setting the Power Line Frequency to '60' is advisable, whilst European regions are likely to use '50'.

[image]

MartyG-RealSense, 2024-03-21:
Hi @HeadReaper-hc  Do you require further assistance with this case, please? Thanks!

HeadReaper-hc, 2024-03-21:
@MartyG-RealSense No need, thank you very much for your help during this period.

closed 2024-03-21 by HeadReaper-hc