realsenseai/realsense-ros #3021: Post-Processing filters decimation_filter ,temporal_filter ,spatial_filter and disparity_filter in python file

josh-001, 2024-02-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D455   |                     
| Operating System & Version |   (Ubuntu 22) | 
| Platform                                 | PC   |
| Language                               |  python  |                         
| ROS Distro			         |  Humble  |

### Issue Description
<Describe your issue / question / feature request / etc..>
I want to enable the post processing filters using python ( decimation_filter.enable:=true temporal_filter.enable:=true spatial_filter.enable:=true disparity_filter.enable:=true)
instead of using the realsense package in the terminal.

MartyG-RealSense, 2024-02-26:
Hi @josh-001  I hope that the Python post-processing script at https://github.com/IntelRealSense/librealsense/issues/1672#issuecomment-387438447 will be a helpful reference for you.

josh-001, 2024-02-26:
Thank you,its working fine .
But, the shape of my depth is changing, i want it to be 1280*720 at last but after post-processing by using the filters depth shape is 640*480.
How to keep it as constant?(1280,720)

MartyG-RealSense, 2024-02-26:
The Decimation filter reduces the resolution that has been set in order to reduce the detail in the depth image and so reduce the amount of data bandwidth that the depth stream uses.  The filter is set by default to divide the resolution by 2.  So if you do not use the Decimation filter then the resolution should stay at 1280x720.

josh-001, 2024-02-26:
is there any other way to maintain the shape (1280*720) after using the filter of decimation_filter?

MartyG-RealSense, 2024-02-26:
No, unfortunately.  The only way to maintain the resolution will be not to use decimation.

If you were hoping to use decimation to reduce bandwidth then you could instead reduce bandwdth by using a lower FPS of 15 if you are not doing do already.  

You could also try not using the Spatial filter.  Of all the post-processing filters, it has one of the heaviest processing burdens with only a small benefit in return.

josh-001, 2024-02-26:
I will look into it.
Thankyou

josh-001, 2024-03-03:
while i am runing the realsesne ros package i came across a shape difference between **/camera/depth/image_rect_raw**-->(480, 848) and **/camera/aligned_depth_to_color/image_raw** -->(720, 1280)
[image]
 how the shape is going back to 720,1280?is there any source code to follow up ?so i can test it out in python
 Thank you.

MartyG-RealSense, 2024-03-03:
Are you using the ROS2 wrapper?  If you are then you can set the depth and color resolutions in the launch instruction.  

If you want both depth and colour to be 848x480 and 30 FPS then you can use the ROS2 launch instruction below.

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=848x480x30 rgb_camera.profile:=848x480x30`

If custom resolutions are not defined then the camera's default resolutions will be used.  On the D455 camera model the defaults are 848x480 depth and 1280x720 color.

MartyG-RealSense, 2024-03-17:
Hi @josh-001  Do you require further assistance with this case, please?  Thanks!

josh-001, 2024-03-20:
No, Thank you for giving response.
If i have doubts i will re-open the issue

closed 2024-03-20 by josh-001

MartyG-RealSense, 2024-03-20:
You are very welcome.  Please do feel free to re-open the issue in future if you need to do so.  Thanks very much for the update!