realsenseai/realsense-ros #3023: No points in pointcloud

pragun, 2024-02-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D457| 
| Firmware Version                   | 5.14.0.0 | 
| Operating System & Version |  L4T 35.1 | 
| Kernel Version (Linux Only)    |  5.10  | 
| Platform                                 | nVidia Orin AGX DevKit  |
| Librealsense SDK Version                            |  2.54.1                          | 
| Language                            |  C/C++ | 
| Segment			|  Robot | 
| ROS Distro			         |  Iron                                           | 
| RealSense ROS Wrapper Version   |  4.54.1                                                | 

### Issue Description
I'm running realsense-ros inside a docker container, I'm able to see the color & depth camera streams on rviz, however there are no points in the pointcloud. Here's the output of the ros launch.

```
ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x30 pointcloud.enable:=true
[INFO] [launch]: All log files can be found below /home/rivs/.ros/log/2024-02-23-20-43-25-658979-ubuntu-162
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [169]
[realsense2_camera_node-1] [INFO] [1708721006.843327620] [camera.camera]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO] [1708721006.843596195] [camera.camera]: Built with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1708721006.843633667] [camera.camera]: Running with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1708721007.001112709] [camera.camera]: Device with serial number 234322307122 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1708721007.001346020] [camera.camera]: Device with physical ID /sys/devices/platform/tegra-capture-vi/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1708721007.001415684] [camera.camera]: Device with name Intel RealSense D457 was found.
[realsense2_camera_node-1] [WARN] [1708721007.002113825] [camera.camera]: Error extracting usb port from device with physical ID: /sys/devices/platform/tegra-capture-vi/video4linux/video0
[realsense2_camera_node-1] Please report on github issue at https://github.com/IntelRealSense/realsense-ros
[realsense2_camera_node-1] [INFO] [1708721007.003870618] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1708721007.004657046] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1708721007.004707958] [camera.camera]: Device Name: Intel RealSense D457
[realsense2_camera_node-1] [INFO] [1708721007.004739638] [camera.camera]: Device Serial No: 234322307122
[realsense2_camera_node-1] [INFO] [1708721007.004767766] [camera.camera]: Device physical port: /sys/devices/platform/tegra-capture-vi/video4linux/video0
[realsense2_camera_node-1] [INFO] [1708721007.004795510] [camera.camera]: Device FW version: 5.14.0
[realsense2_camera_node-1] [INFO] [1708721007.004821878] [camera.camera]: Device Product ID: 0xABCD
[realsense2_camera_node-1] [INFO] [1708721007.004849174] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1708721008.854522177] [camera.camera]: Could not set param: rgb_camera.exposure with nan Range: [nan, nan]: parameter 'rgb_camera.exposure' could not be set: Parameter {rgb_camera.exposure} doesn't comply with floating point range.
[realsense2_camera_node-1] [INFO] [1708721008.892202951] [camera.camera]: Set ROS param rgb_camera.profile to default: 640x480x30
[realsense2_camera_node-1] [INFO] [1708721008.894511998] [camera.camera]: No default profile found. Setting the first available profile as the default one.
[realsense2_camera_node-1] [INFO] [1708721008.894766749] [camera.camera]: Set ROS param gyro_fps to default: 50
[realsense2_camera_node-1] [INFO] [1708721008.894934044] [camera.camera]: Set ROS param accel_fps to default: 400
[realsense2_camera_node-1] [INFO] [1708721008.935207704] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1708721009.063653103] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1708721009.344001868] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1708721009.344591466] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1708721009.436288282] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [WARN] [1708721009.436326426] [camera.camera]: 
[realsense2_camera_node-1] [INFO] [1708721009.789663123] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1708721009.811409692] [camera.camera]: RealSense Node Is Up!
```

Here's a list of ros topics
```
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/theora
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/color/points
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/theora
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/imu
/clicked_point
/goal_pose
/initialpose
/parameter_events
/rosout
/tf
/tf_static
```

Here's a screenshot of rviz:
[image]

MartyG-RealSense, 2024-02-26:
Hi @pragun  Have you had the opportunity to test your D457 on a USB connection instead of GMSL to confirm whether your problems are related specifically to GMSL mode, please?

The D457 camera model has a mode switch under a cap on top of its casing that is set to GMSL connection by default but can be moved into the 'USB' position so that the camera is detected as a D455 and can be used on a USB cable without the deserializer component.

https://support.intelrealsense.com/hc/en-us/community/posts/14840675121043/comments/14843232242835

pragun, 2024-02-26:
Closing this issue to continue discussion on  #3022

closed 2024-02-26 by pragun