realsenseai/realsense-ros #3024: D435i camera on topic : compressedDepth get low fps  on ROS1

2991535823, 2024-02-27:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | {  D435i  }                          | 
| Firmware Version                   | (05.13.00.50)                          | 
| Operating System & Version |   { Ubuntu 20.04.6} | 
| Kernel Version (Linux Only)    |  (5.15.0 -97)                                                                       | 
| Platform                                 | PC|                         |
| Librealsense SDK Version       |  { 2.50.0 }                                                    | 
| Language                               |  {C++ }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  noetic                                           | 
| RealSense ROS Wrapper Version   |   by apt intall ros-noetic-realsence*                | 

### Issue Description
I tried using realsense ros on Ubuntu 20.04.6, using ROS1 version
When I run ```roslaunch realsene2_camera rs_camera.launch```; Most of them are working normally, but subscribing to topics (```/camera/depth/image-rect_raw/compressedDepth```) 
only results in low fps when using ```compressedDepth```
There is a very low frame rate(may be 3 fps), and I see in the log that there is
 ```(uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130```
[image]
Also, I would like to ask about the difference between 
```/camera/depth/image_rect_raw/compressed``` and ```/camera/depth/image_rect_raw/compressedDepth```
when i subscribe compressedDepth, i must modify raw data by delete something  as follows:
```c++
void compressed_depth_cb(const sensor_msgs::CompressedImageConstPtr &image)
    {
        cv_bridge::CvImagePtr imgptr;
        // cout<<"depth id : "<<image->header.stamp<<endl;
        sensor_msgs::CompressedImage local_image = *image;
        auto v = local_image.data;
        v.erase(v.begin(), v.size() > 12 ? v.begin() + 12 : v.end());
        local_image.data = v;
        try
        {
            imgptr = cv_bridge::toCvCopy(local_image);
            // imshow("compress image",imgptr->image);
            // waitKey(2);
        }
        catch (const cv::Exception &e)
        {
            std::cerr << e.what() << '\n';
            return;
        }
    }
```

2991535823, 2024-02-27:
I konwn why fps  so low:  change the png_level to 1 will fix the problem under rqt_reconfigure [image]
but compresseDepth still need delete some raw data

MartyG-RealSense, 2024-02-27:
My understanding from the official RealSense ROS2 guide at the link below is that whether you have
`/camera/depth/image_rect_raw/compressed` or `/camera/depth/image_rect_raw/compressedDepth` is simply dependent on which **image-transport** plugin you have installed.  

https://dev.intelrealsense.com/docs/ros2-wrapper#compression-packages

It appears from the image above that you have all three compression plugins installed that were discussed in the guide link.

I am not aware of differences between them and whether one plugin provides better performance than the other.  

I'm pleased to hear that you found a solution to your low FPS problem.  Thanks very much for sharing it!

closed 2024-03-13 by MartyG-RealSense