realsenseai/realsense-ros #3025: Modify parameter enable_auto_exposure 0 in the launch file

Nanmei, 2024-02-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       |  D435i                          | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |  Ubuntu 18 | 
| Kernel Version (Linux Only)    |  (e.g. 5.4)                                                                       | 
| Platform                                 | PC.                           |
| Librealsense SDK Version       |  { 2.<?>.<?> }                                                    | 
| Language                               |  {C++ }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         | ROS1 }                                           | 
| RealSense ROS Wrapper Version   |}                                                  | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I want to increase color_fps to 60HZ,so I set parameter like this:

[image]
But I want to set it in rs_rgbd.launch, how should I do it? Thank you.
this is my launch file:

<launch>
  <arg name="camera"              default="camera_gap"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>

  

  <arg name="serial_no"           default="036522071266"/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="360"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="infra_width"         default="-1"/>
  <arg name="infra_height"        default="-1"/>
  <arg name="enable_infra1"       default="true"/>
  <arg name="enable_infra2"       default="true"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="360"/>
  <arg name="enable_color"        default="true"/>
  <arg name="enable_auto_exposure"        default="0"/>

  <arg name="fisheye_fps"         default="-1"/>
  <arg name="depth_fps"           default="-1"/>
  <arg name="infra_fps"           default="-1"/>
  <arg name="color_fps"           default="60"/>
  <arg name="gyro_fps"            default="-1"/>
  <arg name="accel_fps"           default="-1"/>
  <arg name="enable_gyro"         default="false"/>
[... 100 lines cut ...]
        <remap from="depth_registered/image_rect" to="$(arg depth_registered)/sw_registered/image_rect_raw" />
      </node>

      
      <node pkg="nodelet" type="nodelet" name="points_xyzrgb_sw_registered"
            args="load depth_image_proc/point_cloud_xyzrgb $(arg manager) $(arg bond)" respawn="$(arg respawn)">
        <remap from="rgb/image_rect_color"        to="$(arg rgb)/image_rect_color" />
        <remap from="rgb/camera_info"             to="$(arg rgb)/camera_info" />
        <remap from="depth_registered/image_rect" to="$(arg depth_registered_filtered)/sw_registered/image_rect_raw" />
        <remap from="depth_registered/points"     to="$(arg depth_registered)/points" />
      </node>
    </group>

    <group if="$(eval depth_registered_processing and hw_registered_processing)">
      
      <node pkg="nodelet" type="nodelet" name="points_xyzrgb_hw_registered"
            args="load depth_image_proc/point_cloud_xyzrgb $(arg manager) $(arg bond)" respawn="$(arg respawn)">
        <remap from="rgb/image_rect_color"        to="$(arg rgb)/image_rect_color" />
        <remap from="rgb/camera_info"             to="$(arg rgb)/camera_info" />
        <remap from="depth_registered/image_rect" to="$(arg depth_registered)/image_raw" />
        <remap from="depth_registered/points"     to="$(arg depth_registered_pub)/points" />
      </node>
    </group>
  </group>
</launch>

MartyG-RealSense, 2024-02-27:
Hi @Nanmei  You can set the FPS with the rs_rgbd launch instruction instead of editing the launch file.

`roslaunch realsense2_camera rs_rgbd.launch color_width:=640 color_height:=360 color_fps:=60`

If you also wanted to also set depth to similar settings then you could do it like this:

`roslaunch realsense2_camera rs_rgbd.launch color_width:=640 color_height:=360 color_fps:=60 depth_width:=640 depth_height:=480 depth_fps:=60`

Setting parameters in the launch instruction overrides the values for those parameters that are stored in the launch file.

Nanmei, 2024-02-28:
Thank you very much for your reply. I have tested: roslaunch realsene2_camera rs_rgbd. launch color width:=640 color height:=360 color fps:=60. But different lighting intensities result in different color_fps. I want to set enable_auto_exposure to 0 and expose to 166. How do I set it in rs_rgbd.launch file?

MartyG-RealSense, 2024-02-28:
The status of RGB auto exposure should be settable in the launch file by adding a block of **rosparam** code to it.

```
<rosparam>
 /camera/rgb_camera/enable_auto_exposure: false
</rosparam>
```

You could also try setting a manual exposure value during runtime after launch with a **rosrun** command.

`rosrun dynamic_reconfigure dynparam set /camera/rgb_camera exposure 166`

Nanmei, 2024-02-29:
That can work thank you very much.

closed 2024-02-29 by Nanmei

MartyG-RealSense, 2024-02-29:
You are very welcome, @Nanmei - I'm pleased that I could help.  :)