realsenseai/realsense-ros #3026: hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready  :  on Jetson AGX Xavior

Muhammadjunaidmalik1, 2024-02-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       |  D435i                          | 
| Firmware Version                   | 5.15.1                         | 
| Operating System & Version |  Linux (Ubuntu 20) | 
| Kernel Version (Linux Only)    |                                                                    | 
| Platform                                 | NVIDIA Jetson AGX Xavior.  |
| Librealsense SDK Version       |  { 2.<?>.<?> }   | 
| Language                               |          | 
| Segment			         |  Robot                            | 
| ROS Distro			         |  Noetic                                         | 
| RealSense ROS Wrapper Version   |  {4.51.1, 4.54.1, etc..}                                                  | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hi ! 

I just have installed Librealsense and ROS wrapper. `realsense-viewer` is working perfectly fine and also SDK examples work just fine.
For Jetson I have followed these intsructions for installing [librealsense](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md).

After installation of the ros wrapper , when I run : 
`roslaunch realsense2_camera rs_camera.launch`

I am getting this weird error.  here is the detailed terminal output :

```
roslaunch realsense2_camera rs_camera.launch
... logging to /home/user/.ros/log/2851cc52-d704-11ee-afc1-48b02d2fe384/roslaunch-ubuntu-10237.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:33439/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
[... 97 lines cut ...]
[ INFO] [1709212339.537321300]: Sync Mode: Off
[ INFO] [1709212339.537509502]: Device Sensors: 
[ INFO] [1709212339.687546985]: Stereo Module was found.
[ INFO] [1709212339.792092608]: RGB Camera was found.
[ INFO] [1709212339.793046163]: Motion Module was found.
[ INFO] [1709212339.793397574]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1709212339.793566319]: num_filters: 0
[ INFO] [1709212339.793676148]: Setting Dynamic reconfig parameters.
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ INFO] [1709212344.753410925]: Done Setting Dynamic reconfig parameters.
/opt/ros/noetic/lib/nodelet/nodelet: symbol lookup error: /home/user/intel_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN2cv3MatC1Ev
[camera/realsense2_camera_manager-2] process has died [pid 10262, exit code 127, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/user/.ros/log/2851cc52-d704-11ee-afc1-48b02d2fe384/camera-realsense2_camera_manager-2.log].
log file: /home/user/.ros/log/2851cc52-d704-11ee-afc1-48b02d2fe384/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has finished cleanly
log file: /home/user/.ros/log/2851cc52-d704-11ee-afc1-48b02d2fe384/camera-realsense2_camera-3*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

MartyG-RealSense, 2024-02-29:
Hi @Muhammadjunaidmalik1  Using camera firmware driver **5.15.1.0** with librealsense SDK version **2.50.0** can cause errors.  The recommended firmware for 2.50.0 is **5.13.0.50**.

Muhammadjunaidmalik1, 2024-02-29:
Thanks for your reply. How can I move back to previous version ?

MartyG-RealSense, 2024-02-29:
If you have access to the realsense-viewer tool then you can use it to downgrade to an older firmware with its the **Update Firmware** option.

1.  Visit the RealSense firmware releases page and download a **.bin** file for the 5.13.0.50 firmware.

https://dev.intelrealsense.com/docs/firmware-releases#d400-series-firmware-downloads

2.  Go to the **More** option of realsense-viewer near the top of its options side-panel and select the menu option **Update Firmware**.  
 
3.  When a file selector box pops up, navigate to the location on your computer that you downloaded the 5.13.0.50 firmware .bin file to and select it.  The firmware downgrade process should then commence.

suptimq, 2024-03-01:
Hi I am running into the same error `An error occurred while calling sensor for: depth_module.auto_exposure_limit: hwmon command 0x80 ...`.

I am using SDK version 2.54.2 which recommends firmware 5.15.1.0. I realized if I have to downgrade it to 5.13.0.5 then the SDK version has to be downgraded to 2.51.1. Since I assumed the only way to install SDK 2.51.1 is through `build from source` that could be painful, I wondered if I could use SDK 2.54.2 and only downgrade the firmware to 5.13.0.5.

Muhammadjunaidmalik1, 2024-03-01:
> If you have access to the realsense-viewer tool then you can use it to downgrade to an older firmware with its the **Update Firmware** option.
> 
>     1. Visit the RealSense firmware releases page and download a **.bin** file for the 5.13.0.50 firmware.
> 
> 
> https://dev.intelrealsense.com/docs/firmware-releases#d400-series-firmware-downloads
> 
>     2. Go to the **More** option of realsense-viewer near the top of its options side-panel and select the option **Update Firmware** from its drop-down menu.
> 
>     3. When a file selector box pops up, navigate to the location on your computer that you downloaded the 5.13.0.50 firmware .bin file to and select it.  The firmware downgrade process should then commence.

thanks Marty for the guidance. That solved the major issue and the node is now working also the imu is also working for 435i in realsense viewer. But now there is another issue, I am getting this error :

`01/03 08:53:54,598 WARNING [281472359461232] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`

When i did `dpkg -l | grep "realsense"` , i got :

```
ii  librealsense2:arm64                         2.54.2-0~realsense.2181               arm64        Intel(R) RealSense(tm) Cross Platform API - runtime
ii  librealsense2-dev:arm64                     2.54.2-0~realsense.2181               arm64        Intel(R) RealSense(tm) Camera Capture API - development files
ii  librealsense2-gl:arm64                      2.54.2-0~realsense.2181               arm64        Intel(R) RealSense(tm) - GLSL-enabled extensions
ii  librealsense2-udev-rules:arm64              2.54.2-0~realsense.2181               arm64        Intel(R) RealSense(tm) Camera Capture API - udev rules
ii  librealsense2-utils:arm64                   2.54.2-0~realsense.2181               arm64        Intel(R) RealSense(tm) Camera Capture API - utils and demos
ii  ros-noetic-librealsense2                    2.50.0-1focal.20211115.135230         arm64        Library for capturing data from the Intel(R) RealSense(TM) SR300, D400, L500 Depth cameras and T2xx Tracking devices.
ii  ros-noetic-realsense2-camera                2.3.2-1focal.20240111.191354          arm64        RealSense Camera package allowing access to Intel T265 Tracking module and SR300 and D400 3D cameras
ii  ros-noetic-realsense2-description           2.3.2-1focal.20240125.205152          arm64        RealSense Camera description package for Intel 3D D400 cameras
```

MartyG-RealSense, 2024-03-01:
If the node is working normally and performance does not appear to be negatively impacted by the **control_transfer returned** warning then it can be safely ignored.

Muhammadjunaidmalik1, 2024-03-01:
There is an issue that its not publishing the colored point cloud :

when i run `roslaunch realsense2_camera rs_camera.launch initial_reset:=true enable_pointcloud:=true`

when i try to see that using pointcloud2 in rviz i am geting this :

```
[ INFO] [1709280813.817536825]: publishing unordered pointcloud.
[ WARN] [1709280814.300572867]: No stream match for pointcloud chosen texture Process - Color
```

MartyG-RealSense, 2024-03-01:
The warning **No stream match for pointcloud chosen texture Process - Color** typically means that there are frame drops occurring on the color stream.

Are you able to achieve a textured pointcloud if you add the command below to your roslaunch instruction to texture the cloud with the infrared stream instead of color?

`pointcloud_texture_stream:=RS2_STREAM_INFRARED`

Muhammadjunaidmalik1, 2024-03-01:
Thanks , I first tried [method1](https://github.com/IntelRealSense/realsense-ros/issues/3026#issuecomment-1972733291) it did not work. [method 2 ](https://github.com/IntelRealSense/realsense-ros/issues/3026#issuecomment-1972741737) works perfectly now.

MartyG-RealSense, 2024-03-01:
You are very welcome.  I'm pleased to hear that you were successful!

Muhammadjunaidmalik1, 2024-03-01:
I am sorry for the trouble but the issue is not solved yet. I have done some tests with the camera, i have the same versions of ROS and SDK installed on my laptop and Jetson AGX Xavior. There are several issues that i am facing :)  : 

1. When i enable gyro and acceleromater topics on jetson AGX xavior using `roslaunch realsense2_camera rs_camera.launch initial_reset:=true enable_gyro:=true enable_accel:=true ` 
i am getting a hardware error : 
```
[ WARN] [1709287804.113420968]: 
[ WARN] [1709287804.113695159]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
[ WARN] [1709287804.356530016]: Hardware Notification:Motion Module failure,1.70929e+12,Error,Hardware Error

```
but it works perfectly fine for my laptop with the same versions. 

2. I have tried this method to see the point clouds : 
`roslaunch realsense2_camera rs_camera.launch initial_reset:=true enable_pointcloud:=true`

it also work fine for my laptop ,but when it comes to my Jetson AGX xavior, i have to use the other launch file.

3. third issue i tried running ir cam with 2 methods:
**method 1:**
For the infrared sensor , when I  run: 
 `roslaunch realsense2_camera rs_camera.launch initial_reset:=true enable_infra:=true infra_rgb:=true` 
 
 I am getting an error : 
` [ INFO] [1709292156.226581520]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...`
on both laptop and jetson

**method 2:**
when i run : 
`roslaunch realsense2_camera rs_camera.launch enable_infra:=true enable_infra1:=true enable_infra2:=true`

i am getting an error :

```
"[ WARN] [1709292408.415422023]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 1, Width: 848, Height: 480, FPS: 30, Format: RGB8
[ WARN] [1709292408.416083531]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 2, Width: 848, Height: 480, FPS: 30, Format: RGB8"
```
in this case also infrared camera is not working for both.

MartyG-RealSense, 2024-03-01:
It's no problem at all.  :)

1.  Does the motion module work if you exit out of the launch with Ctrl-C and perform the launch again?

2.  Jetson boards can sometimes have problems with generating a pointcloud with the ROS1 wrapper's rs_camera launch file.  An rs_rgbd launch usually works instead on Jetson.

3.  The confidence message is not something to be concerned about.  The D435i camera model does not support the confidence stream and so the launch cannot provide it.  Most RealSense users receive this message in their log.

4.  Please check in the log whether your D435i is being detected as being on a USB 2.1 or USB 3.2 connection.  The USB type is provided on the **Device USB type** line of the log.  If the camera is being detected as being in USB 2.1 mode then the Infrared 2 stream will be unavailable (as it is only available on USB 3) and 840x480 may not be supported (640x480 should work).

Muhammadjunaidmalik1, 2024-03-01:
**1.**     i have tried various times and still the same issue. It is able to find the motion module , 
`[ INFO] [1709298290.884287470]: Motion Module was found.`

but it fails a bit later , here is a bit detailed version:
```
[ INFO] [1709298290.784937731]: Stereo Module was found.
[ INFO] [1709298290.881337873]: RGB Camera was found.
[ INFO] [1709298290.884287470]: Motion Module was found.
[ INFO] [1709298290.885208415]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1709298290.886039211]: num_filters: 0
[ INFO] [1709298290.886566343]: Setting Dynamic reconfig parameters.
[ INFO] [1709298295.857978944]: Done Setting Dynamic reconfig parameters.
[ INFO] [1709298295.862473039]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1709298295.875854136]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1709298295.877721691]: gyro stream is enabled - fps: 200
[ INFO] [1709298295.879020032]: accel stream is enabled - fps: 100
[ INFO] [1709298295.880304613]: setupPublishers...
[ INFO] [1709298295.899642955]: Expected frequency for depth = 30.00000
[ INFO] [1709298296.130871205]: Expected frequency for color = 30.00000
[ INFO] [1709298296.282082618]: setupStreams...
 01/03 14:04:56,494 WARNING [281472518713712] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ WARN] [1709298296.625555311]: 
[ WARN] [1709298296.626195441]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
[ WARN] [1709298296.870837654]: Hardware Notification:Motion Module failure,1.7093e+12,Error,Hardware Error
 01/03 14:04:57,015 WARNING [281472074240368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1709298297.067127884]: SELECTED BASE:Depth, 0
[ INFO] [1709298297.151696900]: RealSense Node Is Up!
 01/03 14:04:57,151 WARNING [281472074240368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/03 14:04:57,203 WARNING [281472074240368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/03 14:04:57,254 WARNING [281472074240368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/03 14:04:57,306 WARNING [281472074240368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/03 14:05:55,429 WARNING [281472074240368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/03 14:05:55,480 WARNING [281472074240368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
```
**3,4**  As it turns out not enabling infra-rgb to true makes it to work perfectly fine :

`roslaunch realsense2_camera rs_camera.launch enable_infra1:=true enable_infra2:=true`

using `enable_infra:=true` still gives the same error : 
`
[ INFO] [1709298678.294063001]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...`
 but still infrared camera works.

MartyG-RealSense, 2024-03-01:
1.  If the Motion Module is being erratic in a ROS launch then there is not much that can be done other than to cancel and re-try launch, unfortunately.

3,4.  Yes, do not set infra_rgb to true when using the infrared streams.  Also, I suggest using **enable_infra1:=true** instead of enable_infra so that only the Infra1 stream enables, otherwise it may try to enable Infra2 too and fail if the USB connection type is being detected as 2.1.

Muhammadjunaidmalik1, 2024-03-01:
> 1. If the Motion Module is being erratic in a ROS launch then there is not much that can be done other than to cancel and re-try launch, unfortunately.
>
>
> 3,4. Yes, do not set infra_rgb to true when [...]

1. Its doing the same thing again and again even after trying several times. Even after connecting and disconnecting the camera from jetson . while its working for my laptop with ROS noetic.

MartyG-RealSense, 2024-03-02:
Is the Motion Module more reliable on an RGBD launch on your Jetson?

`roslaunch realsense2_camera rs_rgbd.launch enable_accel:=true enable_gyro:=true `

Muhammadjunaidmalik1, 2024-03-05:
Also getting the same error :

`[ WARN] [1709636969.011481285]: Hardware Notification:Motion Module failure,1.70964e+12,Error,Hardware Error
`

MartyG-RealSense, 2024-03-05:
When you first reported the motion module problem at https://github.com/IntelRealSense/realsense-ros/issues/3026#issuecomment-1973030293 the warning `frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.` was displayed.  So does the motion module failure still occur if you set `global_time_enabled` to true?

`roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation global_time_enabled:=true`

Muhammadjunaidmalik1, 2024-03-12:
Hi Marty ! 

sorry for the delay in my response. I did not have access to my camera. Setting it to true doesn't seem to solve the issue.

MartyG-RealSense, 2024-03-12:
Is the motion module failure still present if you have both global time and **sync** set to true?

`roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation global_time_enabled:=true enable_sync:=true`

Muhammadjunaidmalik1, 2024-03-14:
Altough i am getting the same error as before : 
```

[ WARN] [1710429591.474782756]: 
[ WARN] [1710429591.475415054]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
[ WARN] [1710429591.544402176]: Hardware Notification:Motion Module failure,1.71043e+12,Error,Hardware Error
```

its not publishing the accelerometer and gyroscope data but It is publishing a topic `/camera/imu` which gives a synchronized information accelarometer and gyroscope together.

MartyG-RealSense, 2024-03-14:
Another four RealSense ROS users reported this motion module failure with Jetson in particular at https://github.com/IntelRealSense/realsense-ros/issues/2213 and https://github.com/IntelRealSense/librealsense/issues/10749#issuecomment-1206406678 so it is not just an issue with your Jetson alone.

One of those users reported at https://github.com/IntelRealSense/librealsense/issues/10749#issuecomment-1206406678 that an uninstall and reinstall of the ROS wrapper resolved their motion module failure.

Muhammadjunaidmalik1, 2024-03-15:
Got it. Before closing this issue i am facing another problem as well. , I am using the camera for detection of aruco markers. But the problem is that the when I run the camera it starts giving an error: 

`[camera/aruco_detector_node-1] process has died [pid 91790, exit code -11`

and the camera node gives an error : 
` 15/03 09:13:04,002 WARNING [281472745329008] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`

the common thing between both errors is error code number 11. What do you suggest the issue is in this case ?

MartyG-RealSense, 2024-03-15:
The two messages are not related despite both having the number 11.  The `control_transfer returned` message indicates that there is a communication problem between the camera and the computing device, most likely a USB related issue.

The aruco problem sounds like a case of the same exit code -11 with a RealSense camera at https://github.com/UbiquityRobotics/fiducials/issues/144

Muhammadjunaidmalik1, 2024-03-15:
I understand. But the weird thing about this is that the node doesn't give any error when i run it without the camera, but as soon as i run the camera it gives this error. Also, when I run the camera node it keeps on throwing this error when the call the topic related to the aruco marker. 

If the camera node is call seperately, it continuously publishes this error.

Otherwise, There is no issue with the code for aruco maker as I have tested it on my laptop before and didnot get an issues while running.

MartyG-RealSense, 2024-03-15:
I am not familiar with this aruco detection tool.  I note from the case that I linked to that the aruco launch file seems to try to access RealSense topics.  For some tools, the RealSense ROS wrapper has to be launched first before the other tool that you want to uae (such as your aruco detector) is then launched secondly after the RealSense launch has completed and published its topics.

As you are using ROS1 Noetic, the wrapper launch instruction would be:

`roslaunch realsense2_camera rs_camera.launch`

Muhammadjunaidmalik1, 2024-03-15:
Yes, I have tried that as well and that case the Aruco node dies immediately.

MartyG-RealSense, 2024-03-15:
So your main suspect of the problem cause is that the **realsense2_camera** ROS wrapper node is continuously generating the `control_transfer returned` warning?

Muhammadjunaidmalik1, 2024-03-15:
Yea i suppose.

MartyG-RealSense, 2024-03-16:
You mentioned earlier at https://github.com/IntelRealSense/realsense-ros/issues/3026#issuecomment-1972992248 that `roslaunch realsense2_camera rs_rgbd.launch` was working well.  Is the aruco detector able to run if you use rs_rgbd instead of rs_camera?

Muhammadjunaidmalik1, 2024-03-18:
I have tried it but unfortunately, It didn't solve the issue. The error : 

`WARNING [281471730303344] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`

is actually creating the problem. Can you please help me solve this ?

MartyG-RealSense, 2024-03-18:
When **control_transfer returned** warnings occur, it is rare that they can be eliminated.  The best that can be hoped for is that the streams are published successfully in RViz despite the warnings, and that the rate at which the warnings are being generated in the log is reduced.

Does your Xavier have a different USB port that the camera could be tried in if you have not done so already?

MartyG-RealSense, 2024-03-27:
Hi @Muhammadjunaidmalik1  Do you require further assistance with this case, please?  Thanks!

Muhammadjunaidmalik1, 2024-03-28:
Hi ! 

I just need to ask a question, 

Actually I need it for pose estimation of the aruco markers , I found that realsense-viewer does not have the calibration parameters of the RGBD camera. Is there any way as well to self-calibrate it ? I have tried using the intrinsic parameters in the /camera_info node but those parameters are not giving very realistic results.

MartyG-RealSense, 2024-03-28:
The RGB sensor can be calibrated with the RealSense **Dynamic Calibration** software tool.  It can be installed on Ubuntu using the instructions on page 14 onwards of the tool's user guide at the link below.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

MartyG-RealSense, 2024-03-28:
The tool does calibrate both the depth and RGB sensors, but only calibrates extrinsics.  Calibrating intrinsics as well requires a special OEM version of the Dynamic Calibration software supplied with the $1500 USD Calibration Target Board product.

https://store.intelrealsense.com/buy-intel-realsense-d400-cameras-calibration-target.html

Muhammadjunaidmalik1, 2024-03-28:
Got it thanks alot for the guidance.

closed 2024-03-28 by Muhammadjunaidmalik1