realsenseai/realsense-ros #3027: Wrong result in registered depth image on realsense D435if using rs_rgbd.launch (ROS1)

Alessio-Parmeggiani, 2024-02-29:
Hi, I noticed that using the rs_rgbd.launch file from the ROS1 branch and `align_depth=false` the output registered depth image has some problems.
In the following image I show the rgb image, the depth image and the registered depth computed by the launch file.
[image]

The problem occurs because the register node from depth_image_proc needs an additional parameter `fill_upsampling_holes=true`.
Otherwise the RGB image has a higher resolution and the pixels with no depth information will not be filled.

This is the result with the `fill_upsampling_holes=true` parameter set in the launch file.
[image]

Tested using Ubuntu 22 and ROS2 but the parameter is the same in ROS1 Noetic.

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435if                          | 
| Firmware Version                   | 5.15.1                          | 
| Operating System & Version |  Ubuntu 20 |                                                                        | 
| Platform                                 | PC/Jetson Orin Nano                           |
| Librealsense SDK Version       |  { 2.<?>.<?> }                                                    | 
| ROS Distro			         |  Noetic                                          |

MartyG-RealSense, 2024-02-29:
Hi @Alessio-Parmeggiani  Thanks very much for your report about registered depth in the rs_rgbd launch file and the sharing of your method of using `fill_upsampling_holes=true` to resolve it.

When align_depth is true, the librealsense SDK's **align processing block** automatically compensates for differences in resolution between depth and color.

Alessio-Parmeggiani, 2024-02-29:
Thank you for answering @MartyG-RealSense.
I know about using align_depth=true but when I used it on the Jetson Orin Nano I got a really low rate (about 4 Hz). By using instead  the depth_image_proc package on the rs_rgbd.launch I get a much higher rate.

MartyG-RealSense, 2024-02-29:
As the RealSense ROS1 wrapper has ceased development (the ROS2 wrapper continues to be updated), your `fill_upsampling_holes=true` workaround is likely to be the best available solution to your registered depth issue.  Thanks very much again for sharing your find.

closed 2024-03-08 by Alessio-Parmeggiani