realsenseai/realsense-ros #3028: Question: Why does RealSense IMU give negative acceleration values for Earth's gravity when it's just sitting flat on a table?

milan-r-shah, 2024-02-29:
| Required Info                         |                                                     |
|----------------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.12.14.50 | 
| Operating System & Version  | Linux (Ubuntu 18) | 
| Kernel Version (Linux Only)   |  5.4.0 | 
| Platform                                 | Computer  |
| SDK Version                          | 2.44.0 | 
| Language                              | C++ | 
| Segment			        | {Robot/Smartphone/VR/AR/others} | 

### Issue Description:

As per https://github.com/IntelRealSense/realsense-ros/issues/1449#issuecomment-709889034

> In order to align and them with SDK-wide established orientation convention using the depth sensor as CS origin, each IMU sample is multiplied internally by the extrinsic matrix. The resulted orientation angles and acceleration vectors share the coordinate system with the depth sensor. The positive x-axis points to the right. The **positive y-axis points down**. The positive z-axis points forward

So, when the RealSense D435i camera is just sitting flat on the table, the positive y-axis points down i.e. positive y-axis is in the direction of Earth's gravity, correct? Because of that, the acceleration values along the y-axis should be around **+9.8** `m/s^2`, right? However, in my case, I've been getting negative values, around **-9.8** `m/s^2`:

[image]

This happens for all the axes. For example, if I tilt the camera 90&deg; and make a positive z-axis point downward, I get around **-9.8** `m/s^2` acceleration along the z-axis:

[image]

The same holds true in the case of the positive x-axis.

Shouldn't I be getting positive values for Earth's gravity when the corresponding axis is pointing downward i.e. pointing toward the direction of Earth's gravity? Am I missing something?

MartyG-RealSense, 2024-03-01:
It is normal in RealSense for the 9.81 Y-Accel gravity value to be minus.

https://github.com/IntelRealSense/librealsense/blob/master/doc/d435i.md#sensor-origin-and-coordinate-system

In the real world, the 9.81 gravity can be expressed as a negative value when talking specifically about acceleration towards the surface of the Earth due to attraction.

The behavior of IMU Z-axis moment that you describe is what I would expect to occur.

As the IMU component in RealSense cameras- does not have internal hardware calibration, acceleration may be non-zero when the camera is at rest on a flat surface until the IMU has been calibrated with RealSense's Python IMU calibration tool and a calibration saved to the camera hardware.

https://github.com/IntelRealSense/librealsense/tree/master/tools/rs-imu-calibration

https://dev.intelrealsense.com/docs/imu-calibration-tool-for-intel-realsense-depth-camera

milan-r-shah, 2024-03-01:
@MartyG-RealSense thank you for your reply.

> It is normal in RealSense for the 9.81 Y-Accel gravity value to be minus.
>
> https://github.com/IntelRealSense/librealsense/blob/master/doc/d435i.md#sensor-origin-and-coordinate-system
>
> In the real world, the 9.81 gravity can be expressed as a negative value when talking speciIn the real world, the 9.81 gravity can be expressed as a negative value when talking specifically about acceleration towards the surface of the Earth due to attraction.

But why? I'm still confused. The gravity points downward and so does the positive y-axis when the camera is just sitting flat on a table i.e. both of them are in the same direction, correct? So, by that logic, shouldn't I be getting a positive 9.81 `m/s^2`?

Also, what are the approximate measurements that I'll get for the accelerations (along all three axes) when the camera is just free-falling? Assuming, the positive y-axis (of the camera coordinate system) is pointing downward & perpendicular to the Earth's surface and air resistance is negligible.

MartyG-RealSense, 2024-03-02:
There is not further information available about why the IMU functions as it does.  My understanding though is that when the camera is in motion Y-Accel is a minus value, and when the camera is stationary at rest Y-Accel is meant to be zero (not +9.81) if the IMU has been calibrated.

I do not have the gravity physics knowledge to comment on your second question, unfortunately.

MartyG-RealSense, 2024-03-08:
Hi @milan-r-shah  Bearing in mind the response in the comment above, do you require further assistance with this case please?  Thanks!

MartyG-RealSense, 2024-03-16:
Case closed due to no further comments received.

closed 2024-03-16 by MartyG-RealSense

farbod1277, 2025-05-13:
> [@MartyG-RealSense](https://github.com/MartyG-RealSense) thank you for your reply.
>
>
> But why? I'm still confused. The gravity points downward and so does the positive y-axis when the camera is just [...]

@milan-r-shah given the IMU is not calibrated for gravity here is how you can understand why the value for gravity measured is -9.8m/s^2 when the y-axis points downwards. Imagine the IMU is in free-fall (or equivalently in outer space where no gravity exists - this is called Einstein's equivalence principle by the way, give it a google). During this scenario you would expect to see zeros on all axis of the IMU (remember IMUs are basically just microscopic springs, and when no external force acts on them the springs are in the relaxed state and hence the reading is zero). Now imagine the IMU sitting on the surface of the earth with its y-axis pointing down. The only force acting on the IMU at this point in time is the Normal force from the surface that the IMU is sitting on, effectively accelerating the IMU up (again this comes from Einstein's equivalence principle). This is identical to the IMU acceleration up (in the opposite direction of the y-axis) in outer space. This is why you would expect to read a negative -9.8m/s^2 value. Another way to look at it is that the "micro spring" inside the IMU is sagging down (in the positive y-direction) and hence it'll read a negative value. Similarlly if the IMU was for example accelerating forwards (in the positive z-axis direction) the "micro-spring" would sag backwards (in the negative z-direction) and hence it gives you a positive reading in the z-axis. Hope this helps!

MartyG-RealSense, 2025-05-14:
Thanks so much @farbod1277 for sharing such a detailed physics explanation!