realsenseai/realsense-ros #3033: No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY

tulingcheng86, 2024-03-04:
[WARN] [1709533751.643964743] [rosbag2_player]: New subscription discovered on topic '/camera/camera/imu', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY

wait for imu ... 
wait for imu ... 
wait for imu ... 
wait for imu ... 
wait for imu ... 
wait for imu ... 
wait for imu ... 

can not receive imu

MartyG-RealSense, 2024-03-04:
Hi @tulingcheng86  Can you confirm please that you have enabled the IMU's gyro and accel topics in your launch instruction (as the IMU topics are disabled by default) and used **unite_imu_method** to combine the two gyro and accel topics together into a single `/imu` topic please?

ros2 launch realsense2_camera rs_launch.py enable_gyro:=true enable_accel:=true unite_imu_method:=2

tulingcheng86, 2024-03-04:
> Hi @tulingcheng86 It sounds as though you are attempting to play back a rosbag file containing RealSense camera data.. Is that correct, please?
> 
> If you are using a rosbag file, can you confirm please that when the rosbag file was recorded, the IMU's gyro and accel topics were included in the launch instruction and **unite_imu_method** was used to combine the two gyro and accel topics together into a single `/imu` topic.
> 
> `ros2 launch realsense2_camera rs_launch.py enable_gyro:=true enable_accel:=true unite_imu_method:=2`
> 
> Please also confirm which RealSense camera model the rosbag was recorded with.

Yes, I tried running VINS-FUSION with D435i in a ROS humble environment. Ran three commands,：
rviz2; 
ros2 launch realsense2_camera rs_fusion_camera_launch.py；
ros2 run vins vins_node src/VINS-Fusion-ROS2/config/realsense_d435i/realsense_stereo_imu_config.yaml；

The second terminal reports an error：
[WARN] [1709533751.643964743] [rosbag2_player]: New subscription discovered on topic '/camera/camera/imu', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY.....

The third terminal reports an error：
wait for imu ...
wait for imu ...
wait for imu ...
wait for imu ...
wait for imu ...
wait for imu ...
wait for imu ...

tulingcheng86, 2024-03-04:
When I looked up the topic information, I found that:
$ ros2 topic info --verbose /camera/camera/imu
Type: sensor_msgs/msg/Imu
Publisher count: 1
Node name: camera
Node namespace: /camera
Topic type: sensor_msgs/msg/Imu
Endpoint type: PUBLISHER
GID: 01.0f.47.a8.ad.01.70.2f.01.00.00.00.00.00.47.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: BEST_EFFORT
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
Subscription count: 1
Node name: vins_estimator
Node namespace: /
Topic type: sensor_msgs/msg/Imu
Endpoint type: SUBSCRIPTION
GID: 01.0f.47.a8.0e.05.18.c2.01.00.00.00.00.00.1e.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
  
they don't have the same qos policies, so they can't communicate
how can i solve this problem?THINKS!

MartyG-RealSense, 2024-03-04:
Is the second terminal the one that uses `ros2 launch realsense2_camera rs_fusion_camera_launch.py`

Is `rs_fusion_camera_launch.py` a custom launch file that you have created?

If it is a custom launch, could you paste its contents into a comment so that I can view it please?

tulingcheng86, 2024-03-05:
> 第二个终端是使用的终端吗`ros2 launch realsense2_camera rs_fusion_camera_launch.py`
> 
> 是`rs_fusion_camera_launch.py`您创建的自定义启动文件吗？
> 
> 如果是自定义启动，您能否将其内容粘贴到评论中以便我查看？

OK！
```
"""Launch realsense2_camera node."""
import os
import yaml
from launch import LaunchDescription
import launch_ros.actions
from launch.actions import DeclareLaunchArgument, OpaqueFunction
from launch.substitutions import LaunchConfiguration

configurable_parameters = [{'name': 'camera_name',                  'default': 'camera', 'description': 'camera unique name'},
                           {'name': 'camera_namespace',             'default': 'camera', 'description': 'namespace for camera'},
                           {'name': 'serial_no',                    'default': "''", 'description': 'choose device by serial number'},
                           {'name': 'usb_port_id',                  'default': "''", 'description': 'choose device by usb port id'},
                           {'name': 'device_type',                  'default': "''", 'description': 'choose device by type'},
                           {'name': 'config_file',                  'default': "''", 'description': 'yaml config file'},
                           {'name': 'json_file_path',               'default': "''", 'description': 'allows advanced configuration'},
                           {'name': 'initial_reset',                'default': 'false', 'description': "''"},
                           {'name': 'accelerate_gpu_with_glsl',     'default': "false", 'description': 'enable GPU acceleration with GLSL'},
                           {'name': 'rosbag_filename',              'default': "''", 'description': 'A realsense bagfile to run from as a device'},
                           {'name': 'log_level',                    'default': 'info', 'description': 'debug log level [DEBUG|INFO|WARN|ERROR|FATAL]'},
                           {'name': 'output',                       'default': 'screen', 'description': 'pipe node output [screen|log]'},
                           {'name': 'enable_color',                 'default': 'true', 'description': 'enable color stream'},
                           {'name': 'rgb_camera.profile',           'default': '640,480,0', 'description': 'color image width'},
                           {'name': 'depth_module.emitter_enabled', 'default': '0', 'description': 'jiegouguang'},
                           {'name': 'rgb_camera.color_format',      'default': 'RGB8', 'description': 'color stream format'},
                           {'name': 'rgb_camera.enable_auto_exposure', 'default': 'true', 'description': 'enable/disable auto exposure for color image'},
                           {'name': 'enable_depth',                 'default': 'true', 'description': 'enable depth stream'},
                           {'name': 'enable_infra',                 'default': 'true', 'description': 'enable infra0 stream'},
                           {'name': 'enable_infra1',                'default': 'true', 'description': 'enable infra1 stream'},
                           {'name': 'enable_infra2',                'default': 'true', 'description': 'enable infra2 stream'},
                           {'name': 'depth_module.profile',         'default': '0,0,0', 'description': 'depth module profile'},
                           {'name': 'depth_module.depth_format',    'default': 'Z16', 'description': 'depth stream format'},
                           {'name': 'depth_module.infra_format',    'default': 'RGB8', 'description': 'infra0 stream format'},
                           {'name': 'depth_module.infra1_format',   'default': 'Y8', 'description': 'infra1 stream format'},
                           {'name': 'depth_module.infra2_format',   'default': 'Y8', 'description': 'infra2 stream format'},
                           {'name': 'depth_module.exposure',        'default': '8500', 'description': 'Depth module manual exposure value'},
                           {'name': 'depth_module.gain',            'default': '16', 'description': 'Depth module manual gain value'},
                           {'name': 'depth_module.hdr_enabled',     'default': 'false', 'description': 'Depth module hdr enablement flag. Used for hdr_merge filter'},
                           {'name': 'depth_module.enable_auto_exposure', 'default': 'true', 'description': 'enable/disable auto exposure for depth image'},
                           {'name': 'depth_module.exposure.1',      'default': '7500', 'description': 'Depth module first exposure value. Used for hdr_merge filter'},
                           {'name': 'depth_module.gain.1',          'default': '16', 'description': 'Depth module first gain value. Used for hdr_merge filter'},
                           {'name': 'depth_module.exposure.2',      'default': '1', 'description': 'Depth module second exposure value. Used for hdr_merge filter'},
                           {'name': 'depth_module.gain.2',          'default': '16', 'description': 'Depth module second gain value. Used for hdr_merge filter'},
                           {'name': 'enable_sync',                  'default': 'true', 'description': "'enable sync mode'"},
                           {'name': 'enable_rgbd',                  'default': 'true', 'description': "'enable rgbd topic'"},
                           {'name': 'enable_gyro',                  'default': 'true', 'description': "'enable gyro stream'"},
[... cut ...]
            parameters=[params, params_from_file],
            output=LaunchConfiguration('output' + param_name_suffix),
            arguments=['--ros-args', '--log-level', LaunchConfiguration('log_level' + param_name_suffix)],
            emulate_tty=True,
            )
    ]

def generate_launch_description():
    return LaunchDescription(declare_configurable_parameters(configurable_parameters) + [
        OpaqueFunction(function=launch_setup, kwargs = {'params' : set_configurable_parameters(configurable_parameters)})
    ])

```

MartyG-RealSense, 2024-03-05:
My personal level of knowledge about QOS is limited, unfortunately.  I do apologize.

If your problem is that you need to change the type of QOS policy so that it is compatible with VINS-FUSION, the discussion at https://github.com/IntelRealSense/realsense-ros/pull/2936 might provide useful information.

tulingcheng86, 2024-03-05:
> 不幸的是，我个人对 QOS 的了解有限。我很抱歉。
> 
> [如果您的问题是需要更改 QOS 策略的类型以使其与 VINS-FUSION 兼容， #2936](https://github.com/IntelRealSense/realsense-ros/pull/2936)中的讨论可能会提供有用的信息。

OK,THINKS

closed 2024-03-07 by tulingcheng86

MartyG-RealSense, 2024-03-07:
Thanks so much @tulingcheng86 for sharing your solution!

imedpsd, 2025-05-13:
This helped me too thank you both