realsenseai/realsense-ros #3034: Camera Model: D435i

wendwosenbb, 2024-03-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D405 / D415 / D435 / D435i / D455 / D457 }                          | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |   { Win (10/11) / Linux (Ubuntu 18/20/22) / MacOS } | 
| Kernel Version (Linux Only)    |  (e.g. 5.4)                                                                       | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..                           |
| Librealsense SDK Version       |  { 2.<?>.<?> }                                                    | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  {Iron/Humble/Rolling/etc.. }                                           | 
| RealSense ROS Wrapper Version   |  {4.51.1, 4.54.1, etc..}                                                  | 

### Issue Description
<Describe your issue / question / feature request / etc..>

wendwosenbb, 2024-03-05:
rs-435i camera was working until yesterday, when someone unplugged the usbc port form the usb-hub and plugged it to nuc--11 directly, that damage the camera. 

[realsense2_camera_node-1] [INFO] [1709661829.164238045] [stereo_camera.stereo_camera]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO] [1709661829.164372114] [stereo_camera.stereo_camera]: Built with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1709661829.164390854] [stereo_camera.stereo_camera]: Running with LibRealSense v2.54.1
[realsense2_camera_node-1] [WARN] [1709661829.173163914] [stereo_camera.stereo_camera]: No RealSense devices were found!
[realsense2_camera_node-1] [WARN] [1709661835.180983763] [stereo_camera.stereo_camera]: No RealSense devices were found!
[realsense2_camera_node-1] [WARN] [1709661841.190699589] [stereo_camera.stereo_camera]: No RealSense devices were found!

also, rs-enumerate-devices 

**No device detected. Is it plugged in?**

MartyG-RealSense, 2024-03-05:
Hi @wendwosenbb  Thanks very much for the information.  Under normal conditions RealSense cameras should not be damaged at all by unplugging them from one port and inserting them into another.  

I note that you have measured an 8v spike on the USB port of your NUC-11.  I am not aware of past reports of damage to RealSense cameras in this way, though my research of USB power spikes in general indicates that it can damage the USB port on electronic devices.

Have you tested the camera with a different USB cable to establish whether possible damage is In the cable rather than the camera's micro USB port?  Have you also tried having the USB cable inserted into the camera in both directions (USB-C cables are two-way insertion at the micro sized end).

MartyG-RealSense, 2024-03-13:
Hi @wendwosenbb  Do you require further assistance with this case, please?  Thanks!

wendwosenbb, 2024-03-13:
No. thank you for you quick response.

MartyG-RealSense, 2024-03-13:
You are very welcome.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2024-03-13 by MartyG-RealSense