realsenseai/realsense-ros #3035: Camera node do not work properly

RiccardoZuppetti, 2024-03-06:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D405 / D415 / D435 / **D435i** / D455 / D457 }                          | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |   { Win (10/11) / **Linux (Ubuntu 22.04)** / MacOS } | 
| Kernel Version (Linux Only)    |  (e.g. 5.4)                                                                       | 
| Platform                                 | **PC**/Raspberry Pi/ NVIDIA Jetson / etc..                           |
| Librealsense SDK Version       |  { **2.54.1** }                                                    | 
| Language                               |  {C/C#/labview/opencv/pcl/**python**/unity }         | 
| Segment			         |  {**Robot**/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  {Iron/**Humble**/Rolling/etc.. }                                           | 
| RealSense ROS Wrapper Version   |  {4.51.1, **4.54.1**, etc..}                                                  | 

### Issue Description

Running the following command:

```
ros2 launch realsense_camera2 rs_launch.py
```

everything seems work fine (I am able to visualize the camera stream on RViz). Modifying the `rs_launch.py` script with the following parameters:

```
configurable_parameters = [{'name': 'camera_name',                  'default': 'camera', 'description': 'camera unique name'},
                           {'name': 'serial_no',                    'default': "''", 'description': 'choose device by serial number'},
                           {'name': 'usb_port_id',                  'default': "''", 'description': 'choose device by usb port id'},
                           {'name': 'device_type',                  'default': "''", 'description': 'choose device by type'},
                           {'name': 'config_file',                  'default': "''", 'description': 'yaml config file'},
                           {'name': 'unite_imu_method',             'default': "2", 'description': '[0-None, 1-copy, 2-linear_interpolation]'},
                           {'name': 'json_file_path',               'default': "''", 'description': 'allows advanced configuration'},
                           {'name': 'log_level',                    'default': 'info', 'description': 'debug log level [DEBUG|INFO|WARN|ERROR|FATAL]'},
                           {'name': 'output',                       'default': 'screen', 'description': 'pipe node output [screen|log]'},
                           {'name': 'depth_module.profile',         'default': '0,0,0', 'description': 'depth module profile'},                           
                           {'name': 'enable_depth',                 'default': 'true', 'description': 'enable depth stream'},
                           {'name': 'rgb_camera.profile',           'default': '0,0,0', 'description': 'color image width'},
                           {'name': 'enable_color',                 'default': 'true', 'description': 'enable color stream'},
                           {'name': 'enable_infra1',                'default': 'true', 'description': 'enable infra1 stream'},
                           {'name': 'enable_infra2',                'default': 'true', 'description': 'enable infra2 stream'},
                           {'name': 'infra_rgb',                    'default': 'false', 'description': 'enable infra2 stream'},
                           {'name': 'enable_fisheye1',              'default': 'true', 'description': 'enable fisheye1 stream'},
                           {'name': 'enable_fisheye2',              'default': 'true', 'description': 'enable fisheye2 stream'},
                           {'name': 'enable_confidence',            'default': 'true', 'description': 'enable depth stream'},
                           {'name': 'gyro_fps',                     'default': '200', 'description': "''"},
                           {'name': 'accel_fps',                    'default': '63', 'description': "''"},
                           {'name': 'enable_gyro',                  'default': 'true', 'description': "''"},
                           {'name': 'enable_accel',                 'default': 'true', 'description': "''"},
                           {'name': 'enable_pose',                  'default': 'true', 'description': "''"},                           
                           {'name': 'pose_fps',                     'default': '200', 'description': "''"},                           
                           {'name': 'pointcloud.enable',            'default': 'true', 'description': ''},
                           {'name': 'pointcloud.stream_filter',     'default': '2', 'description': 'texture stream for pointcloud'},
                           {'name': 'pointcloud.stream_index_filter','default': '0', 'description': 'texture stream index for pointcloud'},
                           {'name': 'enable_sync',                  'default': 'true', 'description': "''"},
                           {'name': 'align_depth.enable',           'default': 'true', 'description': "''"},
                           {'name': 'colorizer.enable',             'default': 'false', 'description': "''"},
                           {'name': 'clip_distance',                'default': '-2.', 'description': "''"},                           
                           {'name': 'linear_accel_cov',             'default': '0.01', 'description': "''"},                           
                           {'name': 'initial_reset',                'default': 'false', 'description': "''"},
                           {'name': 'allow_no_texture_points',      'default': 'false', 'description': "''"},                           
                           {'name': 'pointcloud.ordered_pc',        'default': 'false', 'description': ''},
                           {'name': 'tf_publish_rate',              'default': '0.0', 'description': 'Rate of publishing static_tf'},
                           {'name': 'diagnostics_period',           'default': '0.0', 'description': 'Rate of publishing diagnostics. 0=Disabled'},
                           {'name': 'decimation_filter.enable',     'default': 'false', 'description': 'Rate of publishing static_tf'},
                           {'name': 'rosbag_filename',              'default': "''", 'description': 'A realsense bagfile to run from as a device'},
                           {'name': 'depth_module.exposure.1',      'default': '7500', 'description': 'Depth module first exposure value. Used for hdr_merge filter'},
                           {'name': 'depth_module.gain.1',          'default': '16', 'description': 'Depth module first gain value. Used for hdr_merge filter'},
                           {'name': 'depth_module.exposure.2',      'default': '1', 'description': 'Depth module second exposure value. Used for hdr_merge filter'},
                           {'name': 'depth_module.gain.2',          'default': '16', 'description': 'Depth module second gain value. Used for hdr_merge filter'},
                           {'name': 'depth_module.hdr_enabled',     'default': 'false', 'description': 'Depth module hdr enablement flag. Used for hdr_merge filter'},
                           {'name': 'hdr_merge.enable',             'default': 'false', 'description': 'hdr_merge filter enablement flag'},
[... 29 lines cut ...]
[realsense2_camera_node-1] [WARN] [1709653677.162254970] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1709653677.164378361] [camera.camera]: Set ROS param rgb_camera.profile to default: 640x480x15
[realsense2_camera_node-1] [ERROR] [1709653677.164859848] [camera.camera]: Given value, 63 is invalid. Set ROS param back to: 100
[realsense2_camera_node-1] [INFO] [1709653677.174241477] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1709653677.201415707] [camera.camera]: Starting Sensor: Depth Module
[... cut ...]
[realsense2_camera_node-1]  05/03 15:48:02,211 WARNING [139738136688192] (backend-v4l2.cpp:1510) Frames didn't arrived within 5 seconds
[realsense2_camera_node-1] [WARN] [1709653682.213564628] [camera.camera]: XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.70965e+12,Warn,Frames Timeout
[realsense2_camera_node-1]  05/03 15:48:02,235 WARNING [139738119284288] (backend-v4l2.cpp:1510) Frames didn't arrived within 5 seconds
```

and consequently it seems that now on the various topics nothing is published (basically in RViz I I can't visualize anything, like the camera stream), apart from the imu infos.

RiccardoZuppetti, 2024-03-06:
**Update:**

Rerunning the following command with the default parameters:

```
ros2 launch realsense_camera2 rs_launch.py
```

leads to the same issue encountered with the updated parameters:

```
[realsense2_camera_node-1]  06/03 09:59:05,679 WARNING [140280535086656] (backend-v4l2.cpp:1510) Frames didn't arrived within 5 seconds
[realsense2_camera_node-1] [WARN] [1709719145.679961686] [camera.camera]: XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.70972e+12,Warn,Frames Timeout
```

RiccardoZuppetti, 2024-03-06:
Referring to this error:

```
[realsense2_camera_node-1] [ERROR] [1709723913.783679926] [camera.camera]: Given value, 63 is invalid. Set ROS param back to: 100
```

what values can be used?

MartyG-RealSense, 2024-03-06:
Hi @RiccardoZuppetti  Your log shows that your camera firmware driver version is **5.13.0.55** and your librealsense version is **2.54.1**.  Firmware 5.13.0.55 was designed for use with librealsense 2.50.0 and 2.51.1 and can cause errors if used with 2.54.1.  The recommended firmware for 2.54.1 is **5.15.0.2**.

RiccardoZuppetti, 2024-03-07:
Hi @MartyG-RealSense, updating the camera firmware and connecting the camera to an USB3 port seems to overcome the encountered issue. By the way there is always the error about the `accel_fps` value that I've reported above.

MartyG-RealSense, 2024-03-07:
I note that in your log, **accel_fps** is set to 63.  Around April 2022, the IMU component in RealSense cameras was changed to a similar component that supports **100** Accel instead of 63.  So please try changing accel_fps from 63 to 100 in your modified launch file.

RiccardoZuppetti, 2024-03-07:
> I note that in your log, **accel_fps** is set to 63. Around April 2022, the IMU component in RealSense cameras was changed to a similar component that supports **100** Accel instead of 63. So please try changing accel_fps from 63 to 100 in your modified launch file.

So the accel_fps values of 63 or 250 are not anymore supported? Cause if I put 250 instead, by default it returns 100 the same. By the way, I'm now visualizing the camera stream inside the simulator.

MartyG-RealSense, 2024-03-07:
The original IMU supported accel values of 63 or 250.  The updated IMU supports 100 or 200.

RiccardoZuppetti, 2024-03-11:
Instead, the gyro values supported are still 200 and 400?

MartyG-RealSense, 2024-03-11:
Yes, gyro is still 200 and 400.

MartyG-RealSense, 2024-03-21:
Hi @RiccardoZuppetti  Do you require further assistance with this case, please?  Thanks!

RiccardoZuppetti, 2024-03-21:
Hi @MartyG-RealSense, no thanks. I'll close the issue right now. Thank you again.

closed 2024-03-21 by RiccardoZuppetti