realsenseai/realsense-ros #3037: RS2_OPTION_AUTO_EXPOSURE_LIMIT for camera D455 not supported

jeferal, 2024-03-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D455                          | 
| Firmware Version                   | 5.15.1                         | 
| Operating System & Version |   Linux (Ubuntu 20) | 
| Kernel Version (Linux Only)    |  5.15.49                                                                       | 
| Platform                                 | PC                           |
| Librealsense SDK Version       |  { 2.54.2 }                                                    | 
| Language                               |  C++         | 
| Segment			         |  Robot                              | 
| ROS Distro			         |  Noetic                                            | 
| RealSense ROS Wrapper Version   |                                                  | 

### Issue Description
Hello,

I am trying to implement my own Realsense ROS driver and I am interested in detecting when an IR image has the IR pattern on or off.
I am also making use of the sensor options RS2_OPTION_INTER_CAM_SYNC_MODE and RS2_OPTION_EMITTER_ON_OFF to control the emitter of different cameras.

A problem that I encountered is that it seems that in the camera D455 (Stereo Module) the RS2_OPTION_AUTO_EXPOSURE_LIMIT is not supported (I can also get this info directly using the program rs-sensor-control). **I wanted to ask if there is any work around to set a lower upper limit when using auto exposure in the depth module of the camera D455**.

Because I want IR images with emitter on and IR images with emitter off, it looks like if the exposure time is too long, then images with IR with metadata indicating that the emitter was off, I can appreciate visually the IR pattern.

Thank you very much in advance for your help.

MartyG-RealSense, 2024-03-07:
Hi @jeferal  Support for AUTO_EXPOSURE_LIMIT on D435i and D455 was added to the RealSense SDK's **development** branch on February 1 2024 - please see https://github.com/IntelRealSense/librealsense/pull/12582  

The development branch becomes the next 'master' release of the SDK when published.

https://github.com/IntelRealSense/librealsense/tree/development

jeferal, 2024-03-08:
Thank you very much for your help and for your prompt reply.

Very good to know that this feature has been added to the development branch. We will test the branch and keep an eye to see when the next tag is released.
I am using 259 mode at the moment. Thank you, I will also check [#2592](https://github.com/IntelRealSense/realsense-ros/issues/2592).

jeferal, 2024-03-11:
I will close because the feature I was asking for has been added to development.

MartyG-RealSense, 2024-03-11:
You are very welcome, @jeferal - I'm pleased that I could help.  Thanks very much for the update!