realsenseai/realsense-ros #3039: Enable ordered pointcloud result in untextured pointcloud published

JXNCTED, 2024-03-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i                        | 
| Firmware Version                   |5.15.1                        | 
| Operating System & Version |   Ubuntu 22| 
| Kernel Version (Linux Only)    | 6.5.0-21-generic                                                              | 
| Platform                                 |Intel NUC x86_64                  |
| Librealsense SDK Version       |     2.54                                             | 
| Language                               | PCL/C++        | 
| Segment			         |  Robot                           | 
| ROS Distro			         |  Humble                                        | 
| RealSense ROS Wrapper Version   |  4.54.1                                                | 

### Issue Description

I want to use my D451i to extract the pointcloud from a detected bounding box in RGB image, and was trying to achieve this through ordered pointcloud, such that the coordinate from RGB detection can be correspond to that in the pointcloud published. However, by setting the paramater `pointcloud.ordered_pc` to `true`, the published pointcloud contains black region with no textured assigned, even when `pointcloud.allow_no_texture_points` is set to `false`. 

I am wondering if this is the expected behavior and whether there are possible workarounds. 

Many thanks.

JXNCTED, 2024-03-08:
https://github.com/IntelRealSense/realsense-ros/blob/c0a674cc0c829b8cf171210e2120e2c7273525bc/realsense2_camera/src/named_filter.cpp#L225-L250

Oh well, at least according to the code, this is what is expected?

MartyG-RealSense, 2024-03-09:
Hi @JXNCTED  The `allow_no_texture_points` setting, when True, fills in areas at the **outer edges** of the image with zeroes.  If the black regions are in the inner areas of the image then this setting would not fill them in.

When depth and RGB color are both enabled and the pointcloud is enabled, the depth points should be mapped onto the color points.  So there should be correspondence between depth and color without having to use the `ordered_pc` setting.

JXNCTED, 2024-03-09:
Thank you for your response. I understand what is going on now. Will try that.

closed 2024-03-09 by JXNCTED