realsenseai/realsense-ros #3040: When the camera is in emitter_on_off mode, RS2_FRAME_METADATA_FRAME_EMITTER_MODE has an inverted logic

jeferal, 2024-03-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D455                           | 
| Firmware Version                   |     5.15.1                   | 
| Operating System & Version |  Linux (Ubuntu 20)  | 
| Kernel Version (Linux Only)    |  5.15.49                                                                      | 
| Platform                                 | PC                           |
| Librealsense SDK Version       |  2.54.2                                                    | 
| Language                               |  C, C++         | 
| Segment			         |  Robot                             | 
| ROS Distro			         | galactic                                         | 
| RealSense ROS Wrapper Version   |  4.54.1                                                  | 

### Issue Description
Hello,

I am interested in using N cameras where one of them works as a master with inter_cam_sync_mode=1 and the other ones as slave.
The camera that is the master, should have emitter_on_off = true. I would like to detect what infrared frame has the pattern IR emitter On and which does not.
To do this, I am storing first whether this camera has the emitter_on_off option and after that, in every frame callback I ask the for the value of the metadata RS2_FRAME_METADATA_FRAME_EMITTER_MODE.
It seems that when emitter_on_off mode is true, then I get a value false in RS2_FRAME_METADATA_FRAME_EMITTER_MODE when the image does have the emitter and a value true when the image was captured with the emitter off (consistent inverted logic).

I am wondering if this is an issue, if this was expected or if I am missing something and it is working correctly.

In these lines you can see the implemented filter:
https://github.com/IntelRealSense/realsense-ros/compare/4.54.1...jeferal:realsense-ros:ros2_sync_params?expand=1#diff-a54ad2538ccd7b6cce4c6c6d8f57af062bded3d802b88e7b938faefb6f02f4ca

Thank you very much in advance.

MartyG-RealSense, 2024-03-11:
Hi @jeferal `emitter_on_off` will turn the emitter on and off on a per-frame basis (one frame on, one frame off, one frame on, etc).  It would be turning on and off independently of other attached cameras (i.e unsynched on-off).

`inter_cam_sync_mode` 259 and 260 produce two frames per trigger with laser ON-OFF (259) and OFF-ON (260).  The difference between 259 and 260 is that with 259, the Laser ON frame is sent first and then the Laser OFF frame secondly.  With 260 it is the opposite - the Laser OFF frame is sent first and then the Laser ON frame secondly.

You should not need to have `emitter_on_off` enabled when using inter_cam_sync_mode 259 or 260 to sync the emitter on-off.  Please see what results you get if you remove the `emitter_on_off` commands from your launch instruction, as the emitter should alternate anyway if mode 259 or 260 is used.

jeferal, 2024-03-11:
Thank you very much for your prompt reply. I apologize, perhaps the description of the issue was confusing.
I am mostly talking about the camera that acts as master, which has `inter_cam_sync_mode = 1` and `emitter_on_off = 1`. When I read the metadata RS2_FRAME_METADATA_FRAME_EMITTER_MODE of this camera in a frame callback, I get **true** when a frame contains an image with the emitter **off** and **false** when the image does have the **pattern**.

The behaviour is the opposite when I do not set emitter_on_off to 1.

MartyG-RealSense, 2024-03-12:
The RealSense API documentation provides the following information about RS2_FRAME_METADATA_FRAME_EMITTER_MODE.  "Emitter mode: 0 - all emitters disabled. 1 - laser enabled".  

In your modified BasRealsenseNode script in line 848, **is_emitter_on** is defined as a **bool** type variable (true / false), it looks as though line 851 is trying to set the is_emitter_on bool to a **numeric** value based on what the value of the RS2_FRAME_METADATA_FRAME_EMITTER_MODE metadata parameter currently is (0 or 1).  Apologies if I am interpreting your code incorrectly.

jeferal, 2024-03-12:
Thank you for your reply.

**is_emitter_on** is assigned to the result of the expression:
f.get_frame_metadata(RS2_FRAME_METADATA_FRAME_EMITTER_MODE) **> 0**;
which should be a boolean.
What is happening is that when the camera is working with emitter mode On Off set to 1, the RS2_FRAME_METADATA_FRAME_EMITTER_MODE seems to work inverted to what the documentation states, as if it was "Emitter mode: 1 - all emitters disabled. 0 - laser enabled".

closed 2024-03-27 by MartyG-RealSense