realsenseai/realsense-ros #3041: creating  synchronised depth and colour images subscriber in ros2

shrutichakraborty, 2024-03-12:
Hi! I am using a D435i with ros2 humble, I am wondering if there is a code example for subscribing to depth and colour image at the same time in a synchronised way? In ros1 I used message_filters which does not seem a possibility in ros2..  What I'd like to do is subscribe to the colour and depth image at the same time and do the processing in one single callback, any solutions for doing this with ros2?

MartyG-RealSense, 2024-03-12:
Hi @shrutichakraborty  If you just wanted depth and color to have their closest frames gathered together and sent with the same timetag then adding `enable_sync:=true` to your ros2 launch instruction may help to achieve your goal.

As you mention callbacks though, it sounds as though you wish to subscribe via code.  The ROS2 wrapper's example Python node script **rs2_listener.py** and its `imageColorCallback` may be a helpful script reference.

https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/scripts/rs2_listener.py#L85-L92

shrutichakraborty, 2024-03-12:
> Hi @shrutichakraborty If you just wanted depth and color to have their closest frames gathered together and sent with the same timetag then adding `enable_sync:=true` to your ros2 launch instruction [...]

Hi yes I wish to create a subscriber on ros2. I have gone through the code on the link, however I'd like to be able to subscribe to the colour and depth image at the same time, that is I'd like that the colour image and depth image correspond (for I have a camera mounted on robot). In ros1 I could achieve this by messag filters which had time synchronisation, how can this be achieved in ros2?

Thanks!

MartyG-RealSense, 2024-03-12:
Would it be an option for you to install a ROS2 package that provides message filter functionality for the purpose of time synchronization, like https://github.com/ros2/message_filters

closed 2024-03-27 by MartyG-RealSense