realsenseai/realsense-ros #3044: I want to solve this problem.

yang6746, 2024-03-14:
My environment:
Ubuntu 18.04
realsense-ros-ros1-legacy
librealsense 2.50.0
Realsense t265

When i run "roslaunch realsense2_camera rs_camera.launch initial_reset:=true",i got this error:

----------------------------------------------------------------------------------------------------
... logging to /home/jadeting/.ros/log/fd529da0-e208-11ee-9df4-e454e84ebef5/roslaunch-gota-Precision-7920-Tower-69172.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://gota-Precision-7920-Tower:33509/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type:
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters:
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: 848
 * /camera/realsense2_camera/initial_reset: True
 * /camera/realsense2_camera/json_file_path:
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/ordered_pc: False
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/reconnect_timeout: 6.0
 * /camera/realsense2_camera/rosbag_filename:
 * /camera/realsense2_camera/serial_no:
 * /camera/realsense2_camera/stereo_module/exposure/1: 7500
 * /camera/realsense2_camera/stereo_module/exposure/2: 1
 * /camera/realsense2_camera/stereo_module/gain/1: 16
 * /camera/realsense2_camera/stereo_module/gain/2: 16
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method:
 * /camera/realsense2_camera/usb_port_id:
 * /camera/realsense2_camera/wait_for_device_timeout: -1.0
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [69182]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to fd529da0-e208-11ee-9df4-e454e84ebef5
process[rosout-1]: started with pid [69201]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [69218]
process[camera/realsense2_camera-3]: started with pid [69219]
[ INFO] [1710423874.030137715]: Initializing nodelet with 72 worker threads.
[ INFO] [1710423877.153541782]: RealSense ROS v2.3.2
[ INFO] [1710423877.153603599]: Built with LibRealSense v2.50.0
[ INFO] [1710423877.153630199]: Running with LibRealSense v2.50.0
 14/03 21:44:38,195 WARNING [140585468450560] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x1, error: Resource temporarily unavailable, err. num: 11
 14/03 21:44:38,195 ERROR [140585468450560] (tm-boot.h:39) Error booting T265
[ WARN] [1710423880.201895147]: No RealSense devices were found!
 14/03 21:44:47,214 WARNING [140585468450560] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x1, error: Resource temporarily unavailable, err. num: 11
 14/03 21:44:47,214 ERROR [140585468450560] (tm-boot.h:39) Error booting T265

MartyG-RealSense, 2024-03-14:
Hi @yang6746  The T265 camera model is no longer supported by Intel, unfortunately.  However, you could try launching with the **rs_t265** launch file instead of rs_camera.

`roslaunch realsense2_camera rs_t265.launch`

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy?tab=readme-ov-file#using-t265

yang6746, 2024-03-14:
Thank you for your reply! But there is still this error when entering all the instructions in this website address.

 14/03 22:39:58,747 WARNING [139706534409984] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x1, error: Resource temporarily unavailable, err. num: 11
 14/03 22:39:58,747 ERROR [139706534409984] (tm-boot.h:39) Error booting T265
[ WARN] [1710427200.754615485]: No RealSense devices were found!

MartyG-RealSense, 2024-03-14:
The **Error booting T265** error has been known to occur if a USB hub is attached to the computer, as referred to in cases https://github.com/IntelRealSense/librealsense/issues/4681 and https://github.com/IntelRealSense/librealsense/issues/9660

MartyG-RealSense, 2024-03-21:
Hi @yang6746  Were you able to resolve your problem, please?  Thanks!

yang6746, 2024-03-21:
> Hi @yang6746 Were you able to resolve your problem, please? Thanks!

I already understand the reason for the problem, realsense t265 should be using USB 3.0, while my computer only has USB 2.0.

yang6746, 2024-03-21:
> Hi @yang6746 Were you able to resolve your problem, please? Thanks!

Thank you very much！

MartyG-RealSense, 2024-03-21:
You are very welcome.  I'm pleased to hear that you worked out the cause of your problem.  Thanks very much for the update!

closed 2024-03-21 by MartyG-RealSense