realsenseai/realsense-ros #3045: Error message and camera D415 wont start up

12003586, 2024-03-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D415                          | 
| Firmware Version                   | 5.15.1                         | 
| Operating System & Version |     Linux (Ubuntu 18/20/22)  | 
| Kernel Version (Linux Only)    |  (e.g. 5.4)                                                                       | 
| Platform                                 | NVIDIA Jetson .                           |
| Librealsense SDK Version       |  { 2.<?>.<?> }                                                    | 
| Language                               |  {python }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  Melodic                                         | 
| RealSense ROS Wrapper Version   |  {4.51.1, 4.54.1, etc..}                                                  | 

### Issue Description
When I run roslaunch realsense2_camera rs_camera.launch I get this output:

/opt/ros/melodic/lib/nodelet/nodelet: symbol lookup error: /home/bo/catkin_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN20ddynamic_reconfigure19DDynamicReconfigureC1ERKN3ros10NodeHandleE
[camera/realsense2_camera_manager-1] process has died [pid 29541, exit code 127, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/bo/.ros/log/804eee58-e20d-11ee-a93d-48b02dd3e855/camera-realsense2_camera_manager-1.log].
log file: /home/bo/.ros/log/804eee58-e20d-11ee-a93d-48b02dd3e855/camera-realsense2_camera_manager-1*.log
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/bo/.ros/log/804eee58-e20d-11ee-a93d-48b02dd3e855/camera-realsense2_camera-2*.log

all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

When I open the logfile it says this:

[0m[ INFO] [1710430089.548469055]: Loading nodelet /camera/realsense2_camera of type realsense2_camera/RealSenseNodeFactory to manager realsense2_camera_manager with the following remappings:[0m
[0m[ INFO] [1710430089.567961247]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] could not connect to host [bo-desktop:58755], waiting...[0m
[0m[ INFO] [1710430089.635986809]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] is now available.[0m
[0m[ INFO] [1710430100.089433209]: Bond broken, exiting[0m

I don't know how to fix this.

12003586, 2024-03-14:
I installed it but it still gives me the same message

MartyG-RealSense, 2024-03-14:
Which librealsense SDK version and RealSense ROS1 wrapper version are you using, please?

12003586, 2024-03-14:
ROS1 wrapper  2.3.2 -> https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.2
librealsense I downloaded tar.gz from  https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

MartyG-RealSense, 2024-03-14:
Did you install librealsense 2.50.0 from source code first and then install wrapper 2.3.2 from source code using the Method 2 section of the ROS1 wrapper instructions, please?

Or did you install the wrapper with the Method 1 instructions using the command `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`

12003586, 2024-03-15:
I used method 2 to install ROS1 wrapper

MartyG-RealSense, 2024-03-15:
Okay, so if you installed both librealsense and the ROS1 wrapper from source then you should not have more than one installation of librealsense on your computer, unless you have installed it previously (more than one version of librealsense on a computer can potentially cause conflicts).

When the **ZN20ddynamic_reconfigure** error occurs, installing ddynamic_reconfigure usually resolves that problem.  Not in this case though.

As you are using librealsense 2.50.0, the camera firmware driver that should be used with that SDK version is **5.13.0.50**.  Using the 5.15.1.0 firmware that you currently have in your camera with older librealsense versions than 2.54.2 ,could cause errors to occur.  I recommend a firmware downgrade to 5.13.0.50.

12003586, 2024-03-15:
I downgraded but it still doesn't work.
I'm sorry but i'm new to this but I think i did something wrong with the librealsense, because there is a folder with a lock on it called librealsense2

12003586, 2024-03-15:
No the folder with the lock on it only has one folder in it called presets and in it is a file called sample.preset

MartyG-RealSense, 2024-03-15:
Ok, it's likely nothing to worry about then.  However, there should be a librealsense2 source code folder somewhere on your computer if you extracted tar.gz and installed the wrapper from source using the extracted folder.  If you cannot find this folder then it may mean that you need to reinstall librealsense 2.50.0.

12003586, 2024-03-15:
I extracted librealsense 2.50.0 again, is there maybe an installation guide to make sure im not doing anything wrong?

12003586, 2024-03-15:
I deleted the /catkin_ws/src and then used the command sudo apt-get install ros-melodic-realsense2-camera
This output came
bo@bo-desktop:~/catkin_ws/src$ sudo apt-get install ros-melodic-realsense2-camera
[sudo] password for bo: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ros-melodic-realsense2-camera is already the newest version (2.3.2-1bionic.20221025.211354).
The following packages were automatically installed and are no longer required:
  apt-clone archdetect-deb bogl-bterm busybox-static cryptsetup-bin dc
  dpkg-repack festival festlex-cmu festlex-poslex festvox-kallpc16k flite
  gir1.2-timezonemap-1.0 gir1.2-xkl-1.0 grub-common kde-window-manager kinit
  kio kpackagetool5 kwayland-data kwin-common kwin-data kwin-x11
  libdebian-installer4 libestools2.5 libkdecorations2-5v5
  libkdecorations2private5v5 libkf5activities5 libkf5attica5
  libkf5completion-data libkf5completion5 libkf5declarative-data
  libkf5declarative5 libkf5doctools5 libkf5globalaccel-data libkf5globalaccel5
  libkf5globalaccelprivate5 libkf5idletime5 libkf5jobwidgets-data
  libkf5jobwidgets5 libkf5kcmutils-data libkf5kcmutils5 libkf5kiocore5
  libkf5kiontlm5 libkf5kiowidgets5 libkf5newstuff-data libkf5newstuff5
  libkf5newstuffcore5 libkf5package-data libkf5package5 libkf5plasma5
  libkf5quickaddons5 libkf5solid5 libkf5solid5-data libkf5sonnet5-data
  libkf5sonnetcore5 libkf5sonnetui5 libkf5textwidgets-data libkf5textwidgets5
  libkf5waylandclient5 libkf5waylandserver5 libkf5xmlgui-bin libkf5xmlgui-data
  libkf5xmlgui5 libkscreenlocker5 libkwin4-effect-builtins1 libkwineffects11
  libkwinglutils11 libkwinxrenderutils11 libqgsttools-p1 libqt5multimedia5
  libqt5multimedia5-plugins libqt5multimediaquick-p5 libqt5multimediawidgets5
  libxcb-composite0 libxcb-cursor0 libxcb-damage0 os-prober
  python3-dbus.mainloop.pyqt5 python3-icu python3-pam python3-pyqt5
  python3-pyqt5.qtsvg python3-pyqt5.qtwebkit
  qml-module-org-kde-kquickcontrolsaddons qml-module-qtmultimedia
  qml-module-qtquick2 rdate ros-melodic-audio-capture
  ros-melodic-audio-common-msgs ros-melodic-audio-play ros-melodic-sound-play
  tasksel tasksel-data
Use 'sudo apt autoremove' to remove them.
0 to upgrade, 0 to newly install, 0 to remove and 296 not to upgrade.

MartyG-RealSense, 2024-03-15:
So if you perform `rs_camera.launch` you still have the same problem?

MartyG-RealSense, 2024-03-15:
Please check whether the streams can be displayed correctly in RViz.  Sometimes they will work normally even when the control_transfer returned warning is generating continuously in the log.  In that case, you can ignore the warnings.

12003586, 2024-03-15:
Just started up RViz, i can see the camera streams. Thank you very much for the help!

MartyG-RealSense, 2024-03-15:
That's great to hear.  I'm pleased that I could help.  :)

closed 2024-03-15 by 12003586