realsenseai/realsense-ros #3046: Are there any demo of D534i+SLAM with ROS2?

PANQingrui, 2024-03-14:
I have checked related issues and it seems that "https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i" only support "Kinetic" and "Melodic". Since 5 years passed, I would like to know if there are any alternative applications, to build a SLAM with D435i on Unbuntu 22.04. 

Thank you for your help with this matter!

MartyG-RealSense, 2024-03-15:
Hi @PANQingrui  SLAM on ROS2 can be difficult to implement.  RealSense ROS2 users who have attempted it have typically used a combination of **slam_toolbox** and **depthimage_to_laserscan**.

The link for slam_toolbox is below.  You can select a branch of the tool for a specific ROS2 version.  As you are using 22.04, Humble or Iron would likely be the best choice.

https://github.com/SteveMacenski/slam_toolbox

The ROS2 branch of depthimage_to_laserscan is here:

https://github.com/ros-perception/depthimage_to_laserscan/tree/ros2

********

Alternatively, you could consider using **ORB-SLAM** with ROS2 and running it in Stereo mode with your D435i stereo depth camera.  It has a branch for Humble.

https://github.com/zang09/ORB_SLAM3_ROS2/tree/humble

Forsyth-Creations, 2024-03-16:
I'm running into this same issue. For traceability, I was able to get your D435i example (from 2019) running in a Docker container. Needs a little clean-up, but here's the container code:

```
# SLAM with ROS

FROM ros:kinetic-ros-core

# install ROS
# https://docs.ros.org/en/kinetic/Installation/Ubuntu-Install-Debians.html

RUN apt-get update && apt-get install -y \
    software-properties-common \
    curl \
    && add-apt-repository universe

# Add the ROS 2 apt repository to your system
RUN curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg && \
    echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(lsb_release -sc) main" | tee /etc/apt/sources.list.d/ros2-latest.list > /dev/null && \
    apt-get update -y && apt-get upgrade -y && apt-get install -y \
    ros-kinetic-ros-base \
    ros-kinetic-rtabmap-ros \
    ros-kinetic-robot-localization \
    ros-kinetic-imu-filter-madgwick \
    ros-kinetic-realsense2-*

# Install rviz
RUN apt-get install -y ros-kinetic-rviz

# install x11 so it can interact with the host
RUN apt-get install -y x11-apps

# install git
RUN apt-get install -y git

# Download the https://github.com/IntelRealSense/realsense-ros
RUN curl -L0 https://github.com/IntelRealSense/librealsense/archive/refs/tags/v2.18.1.zip -o v2.18.1.zip

# unzip the file
RUN apt install -y unzip
RUN unzip v2.18.1.zip

# install python
RUN apt-get install -y python3-pip nano

RUN rosdep init
```

And the accompanying compose file:

```
services:
  ros_slam:
    build: 
      context: .
      dockerfile: Dockerfile
    network_mode: "host"
    privileged: true
    environment:
      - DISPLAY=$DISPLAY
    volumes:
      - /tmp/.X11-unix:/tmp/.X11-unix
    command : roslaunch realsense2_camera opensource_tracking.launch
```

When I switched a few things to use Humble, I ran into a lot of missing topics (they would show in ros2 topic list, but a ros2 topic hz would indicate that they weren't actually publishing).

Having a pre-baked solution would be great to have for long-term support of this product. Is it a hardware or driver limitation for getting it to ROS2?

MartyG-RealSense, 2024-03-17:
Thanks so much @Forsyth-Creations for sharing your Docker tracking solution!

There is an official ROS2 RealSense SLAM guide that uses **depthimage_to_laserscan** at the link below, though it is old now and was designed for the obsolete `ros2-legacy` wrapper rather than the current `ros2-development `ROS2 wrapper. 

https://intel.github.io/robot_devkit_doc/pages/rs_slam.html

https://github.com/IntelRealSense/realsense-ros/tree/ros2-legacy

The guide uses both a 400 Series camera and the RealSense T265 Tracking Camera, so if you do not have a T265 then you would also have to skip the T265 parts of the instructions and use the **rs_launch.py** launch file instead of the guide's suggested `rs_t265_and_d400.launch.py`

******

Another option is the ROS2 wrapper for the **Kimera** 'semantic SLAM' tool.

https://github.com/MIT-SPARK/Kimera-VIO-ROS2

https://www.youtube.com/watch?v=Zjevg5wQTdI

******

For a more modern ROS2 SLAM solution that is compatible with RealSense and is an out of the box solution, you could look at **SLAMcore**.
 
https://www.imveurope.com/press-releases/slamcore-teams-realsense-provide-out-box-slam-algorithms

MartyG-RealSense, 2024-03-23:
Hi @PANQingrui and @Forsyth-Creations   Do either of you require further assistance with this case, please?  Thanks!

PANQingrui, 2024-03-27:
> Hi @PANQingrui and @Forsyth-Creations Do either of you require further assistance with this case, please? Thanks!

Hi Marty, thank you for your suggestions. I haven't tried your suggestions yet, but I just checked this issue "https://github.com/IntelRealSense/realsense-ros/issues/2992", and it seems to be similar to my needs.

I will try ORB-SLAM  first, and thank you for your help again! I will update here if further assistance is needed.

MartyG-RealSense, 2024-03-27:
You are very welcome, @PANQingrui - good luck!

MartyG-RealSense, 2024-04-19:
Case closed due to no further comments received.

closed 2024-04-19 by MartyG-RealSense

u06d93, 2025-10-20:
> I'm running into this same issue. For traceability, I was able to get your D435i example (from 2019) running in a Docker container. Needs a little clean-up, but here's the container code:
> 
> ```
> # SLAM with ROS
> 
> FROM ros:kinetic-ros-core
> 
> # install ROS
> # https://docs.ros.org/en/kinetic/Installation/Ubuntu-Install-Debians.html
> 
> RUN apt-get update && apt-get install -y \
>     software-properties-common \
>     curl \
>     && add-apt-repository universe
> 
> 
> # Add the ROS 2 apt repository to your system
> RUN curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg && \
>     echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(lsb_release -sc) main" | tee /etc/apt/sources.list.d/ros2-latest.list > /dev/null && \
>     apt-get update -y && apt-get upgrade -y && apt-get install -y \
>     ros-kinetic-ros-base \
>     ros-kinetic-rtabmap-ros \
>     ros-kinetic-robot-localization \
>     ros-kinetic-imu-filter-madgwick \
>     ros-kinetic-realsense2-*
> 
> # Install rviz
> RUN apt-get install -y ros-kinetic-rviz
> 
> # install x11 so it can interact with the host
> RUN apt-get install -y x11-apps
> 
> # install git
> RUN apt-get install -y git
> 
> # Download the https://github.com/IntelRealSense/realsense-ros
> RUN curl -L0 https://github.com/IntelRealSense/librealsense/archive/refs/tags/v2.18.1.zip -o v2.18.1.zip
> 
> # unzip the file
> RUN apt install -y unzip
> RUN unzip v2.18.1.zip
> 
> # install python
> RUN apt-get install -y python3-pip nano
> 
> RUN rosdep init
> ```
> 
> And the accompanying compose file:
> 
> ```
> services:
>   ros_slam:
>     build: 
>       context: .
>       dockerfile: Dockerfile
>     network_mode: "host"
>     privileged: true
>     environment:
>       - DISPLAY=$DISPLAY
>     volumes:
>       - /tmp/.X11-unix:/tmp/.X11-unix
>     command : roslaunch realsense2_camera opensource_tracking.launch
> ```
> 
> When I switched a few things to use Humble, I ran into a lot of missing topics (they would show in ros2 topic list, but a ros2 topic hz would indicate that they weren't actually publishing).
> 
> Having a pre-baked solution would be great to have for long-term support of this product. Is it a hardware or driver limitation for getting it to ROS2? I know TF was clearly missing when I was fiddling around in rviz2.