realsenseai/realsense-ros #3047: WARNING [139770164401920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

Avinaashmani, 2024-03-15:
I am using an Intel Realsense 455, with SDK 5.0 and FW 2.13. I am getting this error message on my terminal when i use with real-sense viewer and with ROS (Noetic). 

Could any one assist me sorting this issue.

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                      D455                        | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version | Ubuntu 20.04 (Focal) along with ROS noetic.                                                                   | 
| Platform                                 | PC                     |

Avinaashmani, 2024-03-15:
I did, and I made sure they all match.

MartyG-RealSense, 2024-03-15:
Can you view the ROS depth and color streams in RViz even when `control_transfer returned` messages are generating in the log?  Or is there no data published?

closed 2024-03-29 by MartyG-RealSense