realsenseai/realsense-ros #3051: ros wrapper installation is failing on ubuntu 22

alonsnir, 2024-03-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435                          | 
| Firmware Version                   | 5.14.0                         | 
| Operating System & Version |  Linux (Ubuntu 22)  | 
| Kernel Version (Linux Only)    |  5.15.0-43-generic                                                                    | 
| Platform                                 | PC                         |
| Librealsense SDK Version       |  modinfo uvcvideo grep "version:" version:        1.1.1-realsense-1.3.24 srcversion:     762B0408ACF11F959293A92  | 
| Language                               |  python        | 
| Segment			         |  Robot                            | 
| ROS Distro			         |  Humble                                         | 
| RealSense ROS Wrapper Version   |  cannot get it installed with sudo apt install ros-humble-realsense2-*                                                  | 

### Issue Description
cannot get the ROS2 wrapper to be installed on a new system

Installed [22.04.1](https://old-releases.ubuntu.com/releases/22.04.1/ubuntu-22.04.1-desktop-amd64.iso) and then changed kernel to 5.15.0-43-generic due to the Reasense SDK requirements

```
cat /etc/apt/sources.list.d/ros-latest.list
deb http://packages.ros.org/ros/ubuntu jammy main
```

```
sudo apt install ros-humble-realsense2-*
.........
Get:179 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-xacro amd64 2.0.8-1jammy.20240217.032951 [36.8 kB]
Ign:180 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-realsense2-description amd64 4.54.1-1jammy.20240217.081213
Ign:177 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-realsense2-camera-dbgsym amd64 4.54.1-1jammy.20240217.085916
Ign:180 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-realsense2-description amd64 4.54.1-1jammy.20240217.081213
Ign:177 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-realsense2-camera-dbgsym amd64 4.54.1-1jammy.20240217.085916
Ign:180 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-realsense2-description amd64 4.54.1-1jammy.20240217.081213
Err:177 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-realsense2-camera-dbgsym amd64 4.54.1-1jammy.20240217.085916
  Connection failed [IP: 64.50.236.52 80]
Err:180 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-realsense2-description amd64 4.54.1-1jammy.20240217.081213
  Connection failed [IP: 64.50.236.52 80]
Fetched 143 MB in 8min 28s (281 kB/s)
E: Failed to fetch http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-humble-realsense2-camera/ros-humble-realsense2-camera-dbgsym_4.54.1-1jammy.20240217.085916_amd64.deb  Connection failed [IP: 64.50.236.52 80]
E: Failed to fetch http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-humble-realsense2-description/ros-humble-realsense2-description_4.54.1-1jammy.20240217.081213_amd64.deb  Connection failed [IP: 64.50.236.52 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

I'm stuck on it a few days already, any help would be much appreciated

MartyG-RealSense, 2024-03-19:
Hi @alonsnir  Before installing the RealSense ROS2 wrapper with `sudo apt install ros-humble-realsense2-*` did you first install the librealsense SDK using the instruction below, please (which uses 'librealsense2' instead of the wrapper's 'realsense2').

`sudo apt install ros-humble-librealsense2*`

alonsnir, 2024-03-19:
Hi @MartyG-RealSense I have followed the installation instructions step by step
1. Install the ROS2 distribution - humble
2. Install latest Intel® RealSense™ SDK 2.0 - install from [Linux Debian Installation Guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages)
3. Install Intel® RealSense™ ROS2 wrapper
Option 1: Install debian package from ROS servers

MartyG-RealSense, 2024-03-19:
Thanks very much for the information.  If you installed the SDK with the distribution_linux instructions then your SDK version is likely to be **2.54.2**.  If your SDK is 2.54.2 then the firmware version that should be used with it is **5.15.1.0** rather than the 5.14.0.0 version (which is designed for use with 2.53.1).  This is unlikely to be the cause of your wrapper installation problem though.

Can you confirm if you configured your Ubuntu repositories using the instructions at https://wiki.ros.org/Installation/Ubuntu/Sources before you used the `sudo apt install ros-humble-realsense2-* installation` instruction, please?

MartyG-RealSense, 2024-03-19:
And you are installing on a PC such as a laptop or desktop and not a computer board such as Nvidia Jetson?

alonsnir, 2024-03-19:
It's Intel SoC. Details here [https://www.aaeon.com/en/p/subcompact-boards-gene-adp6](url)

MartyG-RealSense, 2024-03-19:
SDK installation problems are more likely to occur on computing devices with an 'exotic' hardware / software configuration, such as industrial boards.  In this situation, building librealsense from source code with the **libuvc backend** or **RSUSB backend** installation methods can often resolve these problems.  This is because these installation types are not dependent on  Liux versions or kernel versions and do not require kernel patching.

A quick and easy way to perform a libuvc installation is by using a build script with an installation procedure at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

alonsnir, 2024-03-19:
I will give it a try and get back with results.
Regarding this, does it mean I can ignore the realsense sdk requirement for a specific kernel and go with the default one delivered by the regular Ubuntu 22.04 install?

MartyG-RealSense, 2024-03-19:
Yes, if you use the libuvc or RSUSB backend then you can use the default kernel as librealsense will not need to rely on the kernel.

alonsnir, 2024-03-19:
@MartyG-RealSense before I go with the libuvc method, just to clarify that the librealsense install went well. Could you please explain how this will help with the ros wrapper installation?

MartyG-RealSense, 2024-03-19:
If your current librealsense installation is working fine then stick with that instead of using libuvc.

Considering that you reported **Connection failed** errors when trying to install the wrapper from Debian packages on the ROS server, you could consider whether there is an internet connectivity problem with downloading the packages from IP address 64.50.236.52 80.  This could be due to a firewall or proxy on your computer or a network that the computer is on.

alonsnir, 2024-03-21:
@MartyG-RealSense I have a related question please. Why are node names are differ between what I have installed from the ubuntu packages and the one build from sources?

built from sources, note the version
```
~$ ros2 run realsense2_camera realsense2_camera_node
[INFO] [1711007775.084144083] [camera.camera]: RealSense ROS v4.54.1
[INFO] [1711007775.084229028] [camera.camera]: Built with LibRealSense v2.54.2
[INFO] [1711007775.084235330] [camera.camera]: Running with LibRealSense v2.54.2
```

```
$ ros2 node list
/camera/camera

ros2 topic list
/camera/camera/color/camera_info
/camera/camera/color/image_raw
..............
```

Installed from packages
```
$ ros2 run realsense2_camera realsense2_camera_node
[INFO] [1711007453.627304683] [camera]: RealSense ROS v4.54.1
[INFO] [1711007453.627558324] [camera]: Built with LibRealSense v2.54.1
[INFO] [1711007453.627610465] [camera]: Running with LibRealSense v2.54.1
```

```
$ ros2 node list
/camera

ros2 topic list
/color/camera_info
/color/image_raw
...........
```

MartyG-RealSense, 2024-03-21:
The packages installed from the ROS server will be based on what the best librealsense and ROS wrapper version matches were at the time that the packages were created.  Whilst librealsense 2.54.2 works with wrapper 4.54.1, this SDK version did not exist at the time that the packages were created.  So officially the recommended match for wrapper 4.54.1 is still librealsense 2.54.1.

However, since 2.54.2 was mostly bug-fixes rather than new features, it will still work fine with wrapper 4.54.1 if you install librealsense from source code or non-ROS packages.

MartyG-RealSense, 2024-03-27:
Hi @alonsnir  Do you require further assistance with this case, please?  Thanks!

alonsnir, 2024-03-27:
> @MartyG-RealSense I have a related question please. Why are node names are differ between what I have installed from the ubuntu packages and the one build from sources?
>
> built from sources, note the [...]

I have resolved this by remapping, so nodes run on dev machine (deb packages) vs target machine (build from sources) will refer same topics (e.g. /camera/camera/color/image_raw) and still the question is puzzling

MartyG-RealSense, 2024-03-27:
I believe that the discrepancy may be because a change to camera naming - https://github.com/IntelRealSense/realsense-ros/pull/2857 - was merged into the ROS2 wrapper's source code on August 29 2023.  But that change would not be in the package version of the 4.54.1 wrapper, as the package version was released on June 27 2023.  

The package version of the 4.54.1 will date from its original publishing date of 27 June 2023, but numerous changes have been added to the source code version since then.  The package version will be updated when the next ROS wrapper version is published.

I believe that the topic discrepancy that you highlighted is likely due to a **camera_name** system that was introduced into the source code of the wrapper on August 29 2023 in Pull Request https://github.com/IntelRealSense/realsense-ros/pull/2857

The camera_name feature is described in the wrapper documentation links below.

https://github.com/IntelRealSense/realsense-ros?tab=readme-ov-file#camera-name-and-camera-namespace

https://dev.intelrealsense.com/docs/ros2-wrapper#camera-namespace-and-camera-name

closed 2024-03-27 by alonsnir