realsenseai/realsense-ros #3059: Camera instability issue with multi camera usage on Jetson Orin Nano

kevinstoesser, 2024-03-26:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D455 / D456 / D457                      | 
| Firmware Version                   | 5.15.1                      | 
| Operating System & Version |   20 | 
| Kernel Version (Linux Only)    |  5.10.192-tegra with Kernel patch                                                                      | 
| Platform                                 | NVIDIA Jetson Orin Nano                       |
| Librealsense SDK Version       |   v2.54.2                                             | 
| Language                               |     C++     | 
| Segment			         |  Robot                           | 
| ROS Distro			         | Noetic                                     | 
| RealSense ROS Wrapper Version   |   v2.3.2                                            | 

### Issue Description
I am having major issues with the stability of the Realsense ROS wrapper or SDK in a setup with two cameras running simultaneously. The setup and attempted solutions are as follows:

- Two cameras are connected to the NVIDIA Jetson Orin Nano via a USB3 hub
- Jetpack v5.1.3
- Build the Realsense SDK and the ROS wrapper from the sources with the native Linux USB backend and the Jetson kernel patch
- Required resolution: 1280x720; required fps: 5/15
- Use of rs_multiple_devices.launch (slightly modified)
- Resetting the cameras on startup leads to the problem that one of the cameras does not start properly. If I do it with a timed roslaunch it works better, but not always and really unreliable (https://github.com/IntelRealSense/realsense-ros/issues/774#issuecomment-493933596)
- I tried to launch the cameras using the Python wrapper functions of the Realsense SDK and the same problem occurs

The following error occurs a few times, and after a while it occurs more frequently and even shuts down the cameras completely:

`[ERROR] [1711380104.418901754]: An exception has been thrown: xioctl(VIDIOC_S_FMT) failed, errno=16 Last Error: Device or resource busy
[ERROR] [1711380104.419646152]: Exception: xioctl(VIDIOC_S_FMT) failed, errno=16 Last Error: Device or resource busy
`

Is there any way to solve this problem? At the moment I have no idea what else I could try to get it running reliably.

MartyG-RealSense, 2024-03-27:
Hi @kevinstoesser  Is the USB3 hub that you are using a mains electricity powered hub, please?

kevinstoesser, 2024-03-27:
Hi, no it's a "normal" hub. It's one of those https://www.amazon.de/ICZI-%F0%9D%9F%B0-%F0%9D%97%A3%F0%9D%97%BC%F0%9D%97%BF%F0%9D%98%81%F0%9D%98%80-Verteiler-kompatibel-Desktop-Computer/dp/B01MTL4UVH

MartyG-RealSense, 2024-03-27:
Thanks very much for the information.  Mains powered hubs are typically a better choice for RealSense multiple camera setups on Jetson than a non-powered 'passive' hub, as a passive hub is drawing power for all attached cameras from the Jetson's power supply in the same way that plugging the camera directly into a port on the Jetson does.  

With a mains powered hub, power delivery is more stable because it is provided by the mains and so is independent of the Jetson's power supply.  I appreciate that not every project is suited to a mains hub though, especially mobile robots.

kevinstoesser, 2024-03-27:
Another issue I have is: If I set initial_reset as true, one of my cameras is not connecting at all, even with a time delay in the launch of the nodelets. Do you know how I could solve this issue?

MartyG-RealSense, 2024-03-27:
I would recommend only using initial_reset = true if you are having problems connecting to the camera.  If the camera is able to connect without setting initial_reset to true then you should not need to use a reset.

kevinstoesser, 2024-03-27:
Okay, sometimes I have issues with the auto exposure if I don't reset the cameras. 

For the other issue: is multi camera supported by the RSUSB or still not recommended?

MartyG-RealSense, 2024-03-27:
RSUSB can work with multiple cameras but it is not ideal as it was intended for single camera use and a kernel-based build of librealsense is recommended for multicam.  

For a comparison of RSUSB with a patch-based librealsense build, please visit https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604 and scroll down through the linked-to comment to the section headed **What are the advantages and disadvantages of using libuvc vs patched kernel modules?**

closed 2024-04-10 by MartyG-RealSense