realsenseai/realsense-ros #3060: ROS Noetic Intel RealSense D435 - 26/03 18:30:10,150 WARNING [138995691710208] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b

monajalal, 2024-03-26:
```
az acr login --name ardcr1

docker pull ardcr1.azurecr.io/ros_noetic:latest

(base) mona@ada:~$ docker rm $(docker ps -a -f status=exited -q)

(base) mona@ada:~$ docker run -it --name ros_noetic --entrypoint "/bin/bash" -v /dev:/dev -v /home/mona:/workspace -v /tmp/.X11-unix:/tmp/.X11-unix -e DISPLAY=$DISPLAY --device /dev/bus/usb ardcr1.azurecr.io/ros_noetic:latest

roslaunch realsense2_camera rs_camera.launch align_depth:=true
```

error

```
root@1c28ee0e8fc6:/workspace# roslaunch realsense2_camera rs_camera.launch align_depth:=true
... logging to /root/.ros/log/de991fa2-eb9e-11ee-8933-0242ac110002/roslaunch-1c28ee0e8fc6-27.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://1c28ee0e8fc6:43609/
SUMMARY

PARAMETERS

    /camera/realsense2_camera/accel_fps: -1
    /camera/realsense2_camera/accel_frame_id: camera_accel_frame
    /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
    /camera/realsense2_camera/align_depth: True
    /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
    /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
    /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
    /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
    /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
    /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
    /camera/realsense2_camera/allow_no_texture_points: False
    /camera/realsense2_camera/base_frame_id: camera_link
    /camera/realsense2_camera/calib_odom_file:
    /camera/realsense2_camera/clip_distance: -2.0
    /camera/realsense2_camera/color_fps: -1
    /camera/realsense2_camera/color_frame_id: camera_color_frame
    /camera/realsense2_camera/color_height: -1
    /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
    /camera/realsense2_camera/color_width: -1
    /camera/realsense2_camera/confidence_fps: -1
    /camera/realsense2_camera/confidence_height: -1
    /camera/realsense2_camera/confidence_width: -1
    /camera/realsense2_camera/depth_fps: -1
    /camera/realsense2_camera/depth_frame_id: camera_depth_frame
    /camera/realsense2_camera/depth_height: -1
    /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
    /camera/realsense2_camera/depth_width: -1
    /camera/realsense2_camera/device_type:
    /camera/realsense2_camera/enable_accel: False
    /camera/realsense2_camera/enable_color: True
    /camera/realsense2_camera/enable_confidence: True
    /camera/realsense2_camera/enable_depth: True
    /camera/realsense2_camera/enable_fisheye1: False
    /camera/realsense2_camera/enable_fisheye2: False
    /camera/realsense2_camera/enable_fisheye: False
    /camera/realsense2_camera/enable_gyro: False
    /camera/realsense2_camera/enable_infra1: False
    /camera/realsense2_camera/enable_infra2: False
    /camera/realsense2_camera/enable_infra: False
    /camera/realsense2_camera/enable_pointcloud: False
    /camera/realsense2_camera/enable_pose: False
    /camera/realsense2_camera/enable_sync: False
    /camera/realsense2_camera/filters:
    /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
[... 141 lines cut ...]
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D405 / D415 / D435 / D435i / D455 / D457 }                          | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |   { Win (10/11) / Linux (Ubuntu 18/20/22) / MacOS } | 
| Kernel Version (Linux Only)    |  (e.g. 5.4)                                                                       | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..                           |
| Librealsense SDK Version       |  { 2.<?>.<?> }                                                    | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  {Iron/Humble/Rolling/etc.. }                                           | 
| RealSense ROS Wrapper Version   |  {4.51.1, 4.54.1, etc..}                                                  | 

### Issue Description
<Describe your issue / question / feature request / etc..>

monajalal, 2024-03-26:
problem is with usb port of camera not always working -- cross-checked with `(base) mona@ada:~$ realsense-viewer` and I get the image there however, in the capture by ros I get this message

```
root@1c28ee0e8fc6:/workspace# roslaunch realsense2_camera rs_camera.launch align_depth:=true
... logging to /root/.ros/log/2e081ee8-eba0-11ee-85d9-0242ac110002/roslaunch-1c28ee0e8fc6-245.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://1c28ee0e8fc6:37243/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
[... 74 lines cut ...]
 26/03 18:39:48,289 ERROR [124544614356736] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_IO
 26/03 18:39:48,289 ERROR [124544639534848] (sensor.cpp:572) acquire_power failed: failed to set power state
 26/03 18:39:48,291 WARNING [124544639534848] (rs.cpp:310) null pointer passed for argument "device"
[ INFO] [1711478394.304288023]:  
[ WARN] [1711478394.311977426]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1711478394.312026635]: The requested device with  is NOT found. Will Try again.
 26/03 18:39:54,310 ERROR [124544622749440] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_IO
 26/03 18:39:54,310 ERROR [124544639534848] (sensor.cpp:572) acquire_power failed: failed to set power state
 26/03 18:39:54,311 WARNING [124544639534848] (rs.cpp:310) null pointer passed for argument "device"
[ INFO] [1711478400.322378592]:  
 26/03 18:40:00,330 ERROR [124544605964032] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_IO
[ WARN] [1711478400.332223021]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1711478400.332265968]: The requested device with  is NOT found. Will Try again.
 26/03 18:40:00,331 ERROR [124544639534848] (sensor.cpp:572) acquire_power failed: failed to set power state
 26/03 18:40:00,332 WARNING [124544639534848] (rs.cpp:310) null pointer passed for argument "device"
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
root@1c28ee0e8fc6:/workspace# 

```

MartyG-RealSense, 2024-03-27:
Hi @monajalal  Do you get better performance if you use the parameter **usb_port_id** to tell the launch which port the camera is plugged into?  Your log states that the ID of the port that the camera is in is '1-7'.

`roslaunch realsense2_camera rs_camera.launch align_depth:=true usb_port_id:=1-7`

monajalal, 2024-03-27:
@MartyG-RealSense 

Unfortunately it didn't work

```
root@7ff097760308:/workspace# roslaunch realsense2_camera rs_camera.launch align_depth:=true usb_port_id:=1-7
... logging to /root/.ros/log/9598245e-ec40-11ee-a36d-0242ac110002/roslaunch-7ff097760308-244.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://7ff097760308:35637/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
[... 51 lines cut ...]
[ INFO] [1711547263.339911597]: Initializing nodelet with 32 worker threads.
[ INFO] [1711547263.441660321]: RealSense ROS v2.3.2
[ INFO] [1711547263.441710519]: Built with LibRealSense v2.50.0
[ INFO] [1711547263.441724940]: Running with LibRealSense v2.50.0
[ INFO] [1711547263.473148603]:  
[ INFO] [1711547263.716142293]: Device with serial number 819112070232 was found.

[ INFO] [1711547263.716193517]: Device with physical ID 1-8.1-19 was found.
[ INFO] [1711547263.716205558]: Device with name Intel RealSense D435 was found.
[ INFO] [1711547263.716616422]: Device with port number 1-8.1 was found.
[ERROR] [1711547263.718916159]: The requested device with usb port id 1-7 is NOT found. Will Try again.
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
root@7ff097760308:/workspace# 

```

I am able to get images from realsense-viewer however. Our docker used to work so I don't know what has changed. Do you have any suggestions for debugging this problem?

[image]

monajalal, 2024-03-27:
if I run it with 

`root@cae581563276:/workspace# roslaunch realsense2_camera rs_camera.launch align_depth:=true usb_port_id:=1-8.1`

I get this error:

```
root@cae581563276:/workspace# roslaunch realsense2_camera rs_camera.launch align_depth:=true usb_port_id:=1-8.1
... logging to /root/.ros/log/1257806a-ec42-11ee-af3d-0242ac110002/roslaunch-cae581563276-355.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cae581563276:38065/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
[... 130 lines cut ...]
 27/03 13:58:25,657 WARNING [135299640317696] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/03 13:58:25,708 WARNING [135299640317696] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 27/03 13:58:25,759 WARNING [135299640317696] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

I found that info after I stopped `realsense-viewer`

```
t presets: could not open directory: /home/mona/Documents/librealsense2/presets/
 27/03 09:53:00,775 INFO [130711285309696] (backend-v4l2.cpp:657) Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.1/1-8.1:1.0/video4linux/video0
 27/03 09:53:00,776 INFO [130711285309696] (backend-v4l2.cpp:657) Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.1/1-8.1:1.0/video4linux/video1
 27/03 09:53:00,776 INFO [130711285309696] (backend-v4l2.cpp:657) Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.1/1-8.1:1.0/video4linux/video2
 27/03 09:53:00,776 INFO [130711285309696] (backend-v4l2.cpp:657) Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.1/1-8.1:1.0/video4linux/video3
 27/03 09:53:00,776 INFO [130711285309696] (backend-v4l2.cpp:657) Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.1/1-8.1:1.3/video4linux/video4
 27/03 09:53:00,776 INFO [130711285309696] (backend-v4l2.cpp:657) Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.1/1-8.1:1.3/video4linux/video5
 27/03 09:53:00,776 INFO [130711285309696] (context.cpp:336) Found 1 RealSense devices (mask 0xfe)
^C

```

MartyG-RealSense, 2024-03-27:
The log says that your camera firmware driver version is 5.15.1.0.  This firmware is for use with librealsense SDK version 2.54.2, and using it with older librealsense SDK versions such as the 2.50.0 one that you have got installed can cause errors.

The recommended firmware driver version for 2.50.0 is **5.13.0.50**.

monajalal, 2024-03-28:
Correct. I switched to my laptop with that version of firmware and issue is resolved. Thanks a lot.

closed 2024-03-28 by monajalal

MartyG-RealSense, 2024-03-28:
You are very welcome.  I'm pleased to hear that your issue is solved.  Thanks very much for the update!