realsenseai/realsense-ros #3065: RPi5 + ROS2 + realsense2_camera = No RealSense devices were found

chrismatthieu, 2024-03-29:
I'm trying to get either a D435 or D455 RealSense Camera running on Raspberry Pi 5 (latest Bookwork OS release) and ROS2. Everything installs correctly but "No RealSense devices were found" errors displays using D435 or D455.

Here's my setup:

Raspberry Pi 5 with Debian Bookworm OS
https://docs.ros.org/en/iron/How-To-Guides/Installing-on-Raspberry-Pi.html 

https://docs.ros.org/en/iron/How-To-Guides/Using-Python-Packages.html
docker pull ros 
sudo docker run -it --rm ros

ROS RealSense: https://github.com/IntelRealSense/realsense-ros?tab=readme-ov-file 
sudo apt update 
sudo apt install ros-humble-realsense2-camera
apt pkg list | grep “real”

^ returns:
realsense2_camera
realsense2_camera_msgs

Next I run:
ros2 launch realsense2_camera rs_launch.py

^ returns:
https://gist.github.com/chrismatthieu/ce404568c474af2368b651205f7a0b18

MartyG-RealSense, 2024-03-30:
Hi @chrismatthieu  Does it make a difference if you add **--privileged** to your Docker instruction, please?

`sudo docker run -it --privileged --rm ros`

chrismatthieu, 2024-03-30:
That worked!  Thank you :)

closed 2024-03-30 by chrismatthieu

MartyG-RealSense, 2024-03-30:
It's great to hear that the --priviliged setting worked for you.  Thanks very much for the update!

chrismatthieu, 2024-04-04:
> Hi everyone, I tried the same but with the D435i and the ' --privileged ' flag doesn't do the trick, anyone have any idea ?
> My Docker file --
> FROM ros:humble-ros-base
>
> RUN sudo apt-get update \
>     && apt-get install -y \
>     nano \
>     vim \
>     && rm -rf /var/lib/apt/lists/*
>
> # Installing rviz2
> RUN sudo apt-get update \
>     && apt-get install -y \
>     ros-humble-rviz2 ros-humble-rqt \
>     && rm -rf /var/lib/apt/lists/*
>
> # Install Nav2 packages
> RUN apt-get update && apt-get install -y \
>    ros-$ROS_DISTRO-navigation2 \
>    ros-$ROS_DISTRO-nav2-bringup \
>    ros-$ROS_DISTRO-robot-localization \
>    ros-$ROS_DISTRO-slam-toolbox \
>    && rm -rf /var/lib/apt/lists/*
>
> # Install realsense2 joint-state-publisher xacro packages
> [... 43 lines cut ...]
> --device /dev/ttyACM0  \
> --device /dev/ttyUSB0  \
> --privileged  -it --rm antonio-ros
>
> Bash output -- Screenshot.2024-04-04.155437.png (view on web)

Here’s my working docker container: https://hub.docker.com/r/chrismatthieu/rpi5-ros2-realsense2

chrismatthieu, 2024-04-04:
https://hub.docker.com/r/chrismatthieu/rpi5-ros2-realsense2

tedlin0913, 2025-01-10:
@chrismatthieu Brother, could you please share your Dockerfile? thank you.

chrismatthieu, 2025-01-10:
Here you go :)

https://hub.docker.com/repository/docker/chrismatthieu/rpi5-ros2-realsense2/general￼
hub.docker.com

tedlin0913, 2025-01-10:
Thank you so much for the response, however, can't open the link, got 404.

chrismatthieu, 2025-01-10:
Try this link - https://hub.docker.com/r/chrismatthieu/rpi5-ros2-realsense2

matbeedotcom, 2025-10-03:
can you share the dockerfile contents, I cant view the source dockerfile