realsenseai/realsense-ros #3068: Raspberry Pi Connection timed out

16holson, 2024-04-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i                          | 
| Firmware Version                   | 5.15.1                          | 
| Operating System & Version |  Linux (Ubuntu 22)| 
| Kernel Version (Linux Only)    | 5.15.0-1049-raspi                                                                    | 
| Platform                                 | Raspberry Pi                           |
| Librealsense SDK Version       | 2.54.2                                                  | 
| Language                               | C++         | 
| Segment			         | Robot                          | 
| ROS Distro			         | Humble                                           | 
| RealSense ROS Wrapper Version   | 4.54.1                                               | 

### Issue Description
I just installed the realsense-ros according to the installation instructions listed. When running the launch command (ros2 launch realsense2_camera rs_launch.py) I receive this error:  

`[INFO] [1712252166.503030456] [camera]: Device with physical ID /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[INFO] [1712252166.503094228] [camera]: Device with name Intel RealSense D435I was found.
[INFO] [1712252166.504415108] [camera]: Device with port number 2-1 was found.
[INFO] [1712252166.504537227] [camera]: Device USB type: 3.2
[INFO] [1712252166.511213786] [camera]: getParameters...
[INFO] [1712252166.513545834] [camera]: JSON file is not provided
[INFO] [1712252166.513697487] [camera]: Device Name: Intel RealSense D435I
[INFO] [1712252166.513759981] [camera]: Device Serial No: 047322071069
[INFO] [1712252166.513810717] [camera]: Device physical port: /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/video4linux/video0
[INFO] [1712252166.513857731] [camera]: Device FW version: 5.15.1
[INFO] [1712252166.513904209] [camera]: Device Product ID: 0x0B3A
[INFO] [1712252166.513955148] [camera]: Sync Mode: Off
 04/04 17:36:11,737 ERROR [281472642902240] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
Failed to start device: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
 04/04 17:36:16,866 ERROR [281472642902240] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[ERROR] [1712252176.867428885] [camera]: Error starting device: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out`

16holson, 2024-04-05:
Thank you for your quick response. I tried both commands and receive the same connection timed out error.I will mention that I am able to get the device working on my linux laptop.

MartyG-RealSense, 2024-04-05:
Which method did you use to install the librealsense SDK on your Raspberry Pi, please?  The **xioctl(UVCIOC_CTRL_QUERY)** error typically suggests that there is a conflict between librealsense and the kernel.  Typically on Pi boards, the **libuvc backend** installation method works well.  This method bypasses the kernel and does not require a kernel patch script to be applied, and so it helps to avoid kernel conflicts.

The link below has a build script for easily installing the SDK on Ubuntu by this method.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

You would then have to build the ROS2 wrapper separately from source code after the librealsense SDK installation has been successfully completed.

16holson, 2024-04-09:
So I reimaged my pi with Ubuntu 22 and installed ros2 humble along with the SDK with the instructions you provided. I am running the ros2 wrapper from source and I am able to get video feed for about a minute then it freezes. I have tried different usb cables, different commands(ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=640x480x15 and ros2 launch realsense2_camera rs_launch.py). I've seen that it can be an issue with the SDK and Wrapper versions not being compatible, however, I'm not sure where to check that. Here is my log: 

ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=640x480x15 
[INFO] [launch]: All log files can be found below /home/pper/.ros/log/2024-04-09-08-20-42-626904-pper-desktop-2237
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [2238]
[realsense2_camera_node-1] [INFO] [1712672444.849395292] [camera.camera]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO] [1712672444.849744624] [camera.camera]: Built with LibRealSense v2.54.2
[realsense2_camera_node-1] [INFO] [1712672444.849870487] [camera.camera]: Running with LibRealSense v2.54.2
[realsense2_camera_node-1] [INFO] [1712672445.256901105] [camera.camera]: Device with serial number 047322071069 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1712672445.257225217] [camera.camera]: Device with physical ID 2-1-2 was found.
[realsense2_camera_node-1] [INFO] [1712672445.257309823] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1712672445.258243808] [camera.camera]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1712672445.258461240] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1712672445.267468513] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1712672445.277944166] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1712672445.278120508] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1712672445.278187541] [camera.camera]: Device Serial No: 047322071069
[realsense2_camera_node-1] [INFO] [1712672445.278274333] [camera.camera]: Device physical port: 2-1-2
[realsense2_camera_node-1] [INFO] [1712672445.278328515] [camera.camera]: Device FW version: 5.15.1
[realsense2_camera_node-1] [INFO] [1712672445.278380678] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1712672445.278428657] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1]  09/04 08:20:47,144 WARNING [281472476768480] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
[realsense2_camera_node-1]  09/04 08:20:47,663 WARNING [281472476768480] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
[realsense2_camera_node-1] [WARN] [1712672447.936184832] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1712672447.939806484] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1712672448.339834918] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1712672448.344267565] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1712672448.345252122] [camera.camera]: Set ROS param accel_fps to default: 63
[realsense2_camera_node-1] [INFO] [1712672448.415808302] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1712672448.416177189] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1712672448.536604328] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1]  09/04 08:20:48,663 WARNING [281472476768480] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  09/04 08:20:48,674 WARNING [281472476768480] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1712672448.685393305] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1] [INFO] [1712672448.695052906] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1]  09/04 08:20:48,787 WARNING [281472476768480] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1712672448.808768004] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1712672448.820760978] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1]  09/04 08:20:49,103 WARNING [281472476768480] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  09/04 08:20:49,115 WARNING [281472476768480] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
[realsense2_camera_node-1]  09/04 08:30:21,788 WARNING [281472325642464] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  09/04 08:30:21,899 WARNING [281472325642464] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  09/04 08:30:21,910 ERROR [281472216787168] (error-handling.cpp:96) Error during polling error handler: get_xu(id=7) failed! Last Error: Success

MartyG-RealSense, 2024-04-09:
Your librealsense and ROS wrapper version match-up is fine.

Continuously generating **control_transfer returned** warnings indicate a communication problem between the camera and computing device, typically because of the USB cable or the USB port of the computing device (a Raspberry Pi in this case).  I note that you have already tried changing the USB cable.

It is sometimes possible for the streams to work normally even when control_transfer returned messages are continuously generating, which could explain why it is able to work for a minute before the freeze occurs.

If the USB cable is eliminated as a cause of the issue then the USB port on the Pi remains a possibility.  Which model of Pi board are you using, please? (Pi 3, Pi 4, Pi Zero, etc).

16holson, 2024-04-09:
I am using a Raspberry Pi 4 Model B

MartyG-RealSense, 2024-04-10:
How does it behave at 6 FPS instead of 15?

`depth_module.depth_profile:=640x480x6`

MartyG-RealSense, 2024-04-10:
Another course of action that you could try is to use **compressed** depth and color topics, which can be published by installing support for data compression for Humble with the instruction below.

`sudo apt install ros-humble-image-transport`

https://dev.intelrealsense.com/docs/ros2-wrapper#compression-packages

MartyG-RealSense, 2024-04-17:
Hi @16holson  Do you require further assistance with this case, please?  Thanks!

RedstoneGithub, 2024-04-17:
I am having the same issue on Pi 4 Ubuntu 22.04 with ROS Humble, tried building it with libusb script and with RSUSB backend, and am having the same issues. It only seems like very low resolutions at 6FPS work intermittently, but only for depth and no point cloud. I also get this error:
`WARNING (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`

16holson, 2024-04-17:
@RedstoneGithub I also was able to get it working at low resolutions (640x480). I found better performance if I set both the depth and color camera to the lower resolution. As for the point cloud I can get it displaying for half a second then it goes away. I'm wondering if we disable the cameras, if possible, it will free up some resources so the point cloud can be published.

MartyG-RealSense, 2024-04-18:
Hi @16holson  Raspberry Pi is best suited for streaming the individual depth and color streams of RealSense cameras.  If it is asked to do more, such as pointcloud generation and depth-color alignment, then problems occur.  The IMU component of RealSense camera models equipped with an IMU also typically cannot be accessed on PI.

babbatem, 2024-04-22:
Hi @MartyG-RealSense, is that always true re: the IMU data? It seems like there are some successes in perusing the issues? 

We were hoping to use D435i and a RPi 4 for an upcoming project, but maybe we need to reconsider...

MartyG-RealSense, 2024-04-22:
@babbatem  My understanding is that the best chance of success with the IMU on Pi 4 will be if you build the librealsense SDK from source code with CMake on the Ubuntu OS and include in the CMake build instruction the flag ** -DFORCE_LIBUVC=TRUE**

Using this flag builds librealsense in a mode that bypasses the kernel.  A past investigation of the Raspberry Pi IMU problem by my Intel RealSense colleagues indicated that the cause may have been a conflict with the kernel on Pi, which a libuvc build of the SDK should bypass.

https://github.com/IntelRealSense/librealsense/issues/8994#issuecomment-850581163 is an example of a Pi case involving an IMU-equipped D435i camera where a libuvc source code build enabled the RealSense user in that case to achieve a working SDK build.

babbatem, 2024-04-22:
Thanks, @MartyG-RealSense. I will try this.

MartyG-RealSense, 2024-04-28:
Hi @babbatem  Do you have an update about this case that you can provide, please?  Thanks!

babbatem, 2024-04-30:
Hey @MartyG-RealSense, after compiling with uvc I am able to read from the IMU topics but I'm observing the frame drops and "permanent blackouts" described here: 

https://github.com/IntelRealSense/realsense-ros/issues/1469#issuecomment-727844189. 

Sounds like the Pi isn't well suited for IMU applications.

MartyG-RealSense, 2024-05-01:
Hi @babbatem  Yes, there are known problems on Pi boards with reading the RealSense IMU or doing more than simply accessing the depth and RGB streams with Pi.

MartyG-RealSense, 2024-05-08:
Hi @16holson and @babbatem  Do either of you require further assistance with this case, please?  Thanks!

babbatem, 2024-05-08:
@MartyG-RealSense I'm all set. Thanks.

MartyG-RealSense, 2024-05-08:
Thanks very much @babbatem   :)

closed 2024-05-17 by MartyG-RealSense