realsenseai/realsense-ros #3069: camera/color/image_raw topic issue

kks1467TW, 2024-04-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D455 }                          | 
| Firmware Version                   | (05.13.00.50)                          | 
| Operating System & Version |   {  Linux (Ubuntu 18.04)  } | 
| Kernel Version (Linux Only)    |  (Linux 5.4.0-150-generic)                                                                       | 
| Platform                                 | PC                           |
| Librealsense SDK Version       |  { 2.54.2 }                                                    | 
| Language                               |  {C++ }         | 
| Segment			         |  {Robot }                             | 
| ROS Distro			         |  {melodic }                                           | 
| RealSense ROS Wrapper Version   |  {2.3.2}                                                  | 

### Issue Description
<Describe your issue / question / feature request / etc..>

                      **Issue content**
• When running the Realsense D455 camera through the Pc's boot script, hepatologically @ camera/color/image_raw
Does not come up at this time, depth/image_rect_raw is normal

• According to Rostopic echo/diagonostics, the actual frequency is also 0 Hz for color(RGB camera)
But! for depth, the maximum & minimum frequency is satisfied at 29.89 Hz

•Modified the code to add ROS_INFO ("Stream Type: %d", stream_type) within the frame_callback() function to check the output of stream_type, it was confirmed that the output value of depth_rect_raw's output 1/image_raw was 2

If it is running normally, stream_type repeatedly outputs 1,2, while hepatologically outputs stream_type:1 repeatedly

• At this time, if we deactivate the camera with rosservice call /camera/enable "data:false" and activate it with data:true
color/image_raw the output was normally
– Therefore, if color/image_raw data was not received, temporary action was taken through activation after deactivation

As a side note, base_realsense_node.cpp was modified and tested by changing the order of execution 
(Color in RGB Camera > Depth in Stereo Module           to                   Depth in Stereo    > Color in RGB Camera)

I attached a log file and the log is  when I ran Device Name: Intel RealSense D455 after booting. The two logs are the difference in the order of execution.

As a side note, the result of the execution was that the topic did not break even after 40 repetitions of the problem that occurred as hepatic blood (1 time per 10 times)
This may have been due to a luck factor, so please keep it in mind.

[Depth in Stereo first.txt](https://github.com/IntelRealSense/realsense-ros/files/14881131/Depth.in.Stereo.first.txt)

MartyG-RealSense, 2024-04-05:
Hi @kks1467TW  Whilst you have the correct firmware driver installed for the 2.3.2 ROS1 wrapper, the librealsense SDK version that should be used with this wrapper is 2.50.0 or 2.51.1.  Development of the ROS1 wrapper has ceased, so the wrapper has not been updated for use with newer SDK versions.

kks1467TW, 2024-04-08:
@MartyG-RealSense
 If the SDK version is 2.50.0 or 2.51.1
Can I expect the topic to be uninterrupted regardless of the order of execution of RGB cameras?

MartyG-RealSense, 2024-04-08:
It should not usually be necessary to change the order of execution, and if I were enabling the sensors manually then I would normally enable depth first and color second.  There may be something unique about the librealsense and wrapper installation on your particular computer though where enabling color first and depth second works better.  

I have not seen other RealSense users perform the edits that you have implemented and it is not something that I would suggest doing.  But if doing so provides you with better stability then it is not wrong to do it.

kks1467TW, 2024-04-08:
I have an additional question. @MartyG-RealSense  I got a reply that the update to ros1 was interrupted at 2.51.1 and if I run it at 2.54.2 i.e. the latest version, can I cause an error?

MartyG-RealSense, 2024-04-08:
The ROS1 wrapper version 2.3.2 is not designed for use with librealsense 2.54.2 and using it with 2.54.2 can cause errors.  The newest SDK that should be used with it is 2.51.1.  This is because development of the ROS1 wrapper has ended and only the ROS2 wrapper is now updated, so the ROS1 wrapper has not been updated to work with the latest librealsense versions.

kks1467TW, 2024-04-09:
@MartyG-RealSense  

Thank you for your response.

But it's a bit different from warper version 2.3.2
The same problem occurred when using SDK 2.51.1 / 2.5.0.

I don't think there's a logic to determine if you're getting data for color/image_raw even if you're looking for a device

So is there any chance of this happening when I ran the device as soon as I booted the camera from my PC hepathetically?

MartyG-RealSense, 2024-04-09:
It does not seem likely that it would make a difference if you booted the camera immediately. 

Does it make any difference if you reset the camera at launch by placing the **initial_reset:=true** command in your roslaunch instruction?

`roslaunch realsense2_camera rs_camera.launch initial_reset:=true`

closed 2024-04-24 by MartyG-RealSense