realsenseai/realsense-ros #3070: When outputting RGBD Topic, only 0 is echoed in depth.

ryosei1250, 2024-04-07:
I use translation software
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D455                          | 
| Firmware Version                   | (2.54.2)                          | 
| Operating System & Version |   Linux (Ubuntu 22) | 
| Kernel Version (Linux Only)    |  (6.2.0-39-generic)                                                                       | 
| Platform                                 | PC                          |
| Librealsense SDK Version       |  { 2.54.2 }                                                    | 
| Language                               |  C++         | 
| Segment			         |  Robot                             | 
| ROS Distro			         |  Humble                                          | 
| RealSense ROS Wrapper Version   |  I don't know how to look it up.                                        | 

### Issue Description
I output /camera/cemera/rgbd according to the RGBD TOPIC in the README, but when I do ros2 topic echo /camera/camera/rgbd, color is output, but 0 is output for depth.
[image]
[image]

MartyG-RealSense, 2024-04-07:
Hi @ryosei1250  The wrapper version can be found in the log produced by the launch and will look like this.

```
[realsense2_camera_node-1] [INFO]  [camera.camera]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO]  [camera.camera]: Built with LibRealSense v2.54.2
[realsense2_camera_node-1] [INFO]  [camera.camera]: Running with LibRealSense v2.54.2
[realsense2_camera_node-1] [INFO]  [camera.camera]: Device with serial number xxxxxxxxxxxx was found.
```

Is depth produced if you reset the camera at launch by adding initial_reset to your ros2 launch instruction please?

`ros2 launch realsense2_camera rs_launch.py initial_reset:=true`

ryosei1250, 2024-04-07:
Thanks for the reply.
RealSense ROS Wrapper Version is v4.54.1
Follow the commands in the README.
ros2 launch realsense2_camera rs_launch.py enable_rgbd:=true enable_sync:=true align_depth.enable:=true enable_color:=true enable_depth:= true initial_reset:=true
but it did not work

MartyG-RealSense, 2024-04-07:
Depth and color frames are also published on the topics `/camera/camera/depth/image_rect_raw` and `/camera/camera/color/image_raw` so please try echo of these topics for color and depth data.

ryosei1250, 2024-04-07:
When I echoed /camera/camera/depth/image_rect_raw, it was echoed properly (not 0). 
The same goes for /camera/camera/color/image_raw.

MartyG-RealSense, 2024-04-07:
The RGBD topic publishes depth and color in the same topic when align_depth.enable = true.  So you may be able to achieve the same outcome as the RGBD topic if you echo the aligned depth-color topic `/camera/camera/aligned_depth_to_color/image_raw`

ryosei1250, 2024-04-07:
Does it need to change any options?

MartyG-RealSense, 2024-04-07:
For the aligned depth topic to be published, the launch instruction should have `align_depth.enable:=true` and depth and color must also be true.

ryosei1250, 2024-04-07:
I echoed the output of /camera/camera/aligned_depth_to_color/image_raw and it was 0
[image]

MartyG-RealSense, 2024-04-07:
So there is no data published on the aligned topic if you enable alignment, depth and color with the launch instruction below?

`ros2 launch realsense2_camera rs_launch.py enable_depth:=true enable_color:=true depth_module.depth_profile:=1280x720x30 rgb_camera.color_profile:=1280x720x30 align_depth.enable:=true`

MartyG-RealSense, 2024-04-07:
And there is also no depth from the aligned topic if you launch with ros2 run instead of ros2 launch?

ros2 run realsense2_camera realsense2_camera_node --ros-args -p enable_color:=true -p enable_depth:=true -p align_depth.enable:=true

ryosei1250, 2024-04-07:
Yes, that command did not work either.

MartyG-RealSense, 2024-04-07:
What happens when you reduce resolution from 1280x720 to 640x480?

`ros2 launch realsense2_camera rs_launch.py enable_depth:=true enable_color:=true depth_module.depth_profile:=640x480x30 rgb_camera.color_profile:=640x480x30 align_depth.enable:=true`

ryosei1250, 2024-04-07:
Yes, that command didn't work either, it prints 0

MartyG-RealSense, 2024-04-07:
So non-zero data can be obtained with the individual depth topic **/camera/camera/depth/image_rect_raw** but not on an aligned topic.

It does not sound as though there is anything wrong with your librealsense SDK or ROS wrapper installation since you are able to obtain depth and color individually.  It is only when align_depth.enable = true that a problem occurs.

Is there any change if you also publish a pointcloud in addition to having alignment enabled?

`ros2 launch realsense2_camera rs_launch.py enable_depth:=true enable_color:=true depth_module.depth_profile:=1280x720x30 rgb_camera.color_profile:=1280x720x30 align_depth.enable:=true pointcloud.enable:=true`

ryosei1250, 2024-04-08:
That command does not work either.

ryosei1250, 2024-04-08:
I did it from step2 of the README installation on Ubuntu of realsense-ros.

MartyG-RealSense, 2024-04-08:
Okay, that looks okay for installing librealsense as kernel 6.2 is supported by the DKMS package installation.

After you had installed librealsense, how did you install the ROS2 wrapper afterwards?

ryosei1250, 2024-04-08:
(Creating a Workspace)
mkdir -p ~/realsense_ws/src
cd ~/realsense_ws/src/

(Install realsense-ros and colcon build)
git clone https://github.com/IntelRealSense/realsense-ros.git
cd ~/realsense_ws
colcon build --symlink-install

(The default branch is ros2-development, so use it)
cd src/realsense-ros/
git branch
* ros2-development

I followed this procedure.

MartyG-RealSense, 2024-04-08:
That is fine too.  So you have installed the librealsense SDK and ROS wrapper correctly.  And depth and color work individually but the aligned depth topic does not work.

I think that this is a problem where everything has been tried and none of it has worked, so the only action left to try is to wipe everything on the computer and do a fresh installation of it all (Ubuntu, librealsense, ROS wrapper) and hope that there is a hidden problem that a complete reinstall of the system removes.  The other option is to leave everything as it currently is and not use aligned topics.

ryosei1250, 2024-04-08:
As a test, I followed the same procedure to build an environment on another PC and typed the same command, but the result was the same: 0 was output.
I will try to test with another Realsense455.

Current measures include
1. get timestamps of depth and rgb
2. use message_filter package

If rgbd or assigned_depth_to_color is not available, I am thinking of trying these methods. However, I am wondering if the angle of view is synchronized.

MartyG-RealSense, 2024-04-08:
As you have installed the librealsense **librealsense2-utils** package, you should have the SDK's example programs installed on your computer.  So you could try launching the **rs-align** example program to see whether the camera is able to produce an aligned image in that program even if the ROS wrapper is not providing one.

ryosei1250, 2024-04-10:
What kind of command does it enter?

MartyG-RealSense, 2024-04-10:
Are you asking how the rs-align example is launched, please?  If you are then the examples should be installed at the location **usr/local/bin** on your Linux computer, so you could navigate to that folder and try inputting **rs-align** to run that program.

ryosei1250, 2024-04-10:
This is the result[
[image]
](url)

ryosei1250, 2024-04-10:
This is the result
[image]

MartyG-RealSense, 2024-04-11:
Thanks very much for the image of rs-align.  It looks like it is mostly depth information in the image and the RGB data is present in the background but barely visible.

If you have access to the RealSense Viewer tool, please try resetting the camera to its factory-new default calibration values in the Viewer using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487 to see whether it makes any difference to alignment in the ROS wrapper.

MartyG-RealSense, 2024-04-12:
That rs-align image seems like a significant improvement over the previous one.  You can clearly see the RGB in the aligned image now.

But you still do not have aligned data in the ROS wrapper after the calibration was reset in the Viewer?

ryosei1250, 2024-04-12:
Yes, that's right.

MartyG-RealSense, 2024-04-13:
There still does not appear to be a clear reason why alignment is not functioning for you in the ROS wrapper when the individual depth and color topics can work.  We have not had reports like this from other RealSense ROS users recently, which suggests that there is not a bug in the wrapper and the problem is local to your computer.  However, I note your mention at https://github.com/IntelRealSense/realsense-ros/issues/3070#issuecomment-2042510831 that you experienced the same problem when you installed on another PC.

Because you are able to echo the  **/camera/camera/aligned_depth_to_color/image_raw** topic, this also suggests that the topic is being published, even if the data in it is all zero.

Are the PCs that you tested on  a normal desktop / laptop computer or a single board computer like Raspberry Pi or Nvidia Jetson, please?

ryosei1250, 2024-04-14:
The PC I used is a laptop computer

MartyG-RealSense, 2024-04-14:
Are the two different Windows PCs that you have used the camera on the same make / model of computer?

Computers have **USB controllers** that handle the data passing through the USB ports.  The manufacturer of the computer chooses the brand-name of USB controller that is used, and some controllers work better with RealSense cameras than others.  So if your two Windows PCs were the same make / model then they should have the same brand of USB controller. If your Linux PC is a different make / model then it may use a different USB controller to the Windows PCs.

A way to test whether the USB controller is the source of the problem is to plug the camera into a **USB hub** device, as the USB hub will have a different USB controller chosen by the hub's manufacturer and so may handle the camera better than the controller used on the computer's built-in ports.  Do you have a USB hub that you could try, please?

ryosei1250, 2024-04-15:
One is a Dell and the other is a mouse.

I tested with anker USB hub but it did not work.

MartyG-RealSense, 2024-04-15:
Thank you very much for your patience during this case.  I have examined the case again from the beginning and considered all elements of it.  I note that the problem is not only with alignment as you were also unable to produce a depth-color aligned pointcloud using the pointcloud filter.

Let's next test for further evidence regarding whether the problem is with the RGB sensor by not using the color stream for pointcloud texturing and texturing the pointcloud with the **infrared** stream instead.

`ros2 launch realsense2_camera rs_launch.py enable_infra1:=true pointcloud.enable:=true pointcloud.stream_filter:=1`

Does the pointcloud publish if the above command is used to texture it with infrared instead of the RGB sensor, please?

ryosei1250, 2024-04-16:
This is the result
/camera/camera/infra1/image_rect_raw
[image]

MartyG-RealSense, 2024-04-16:
Can the infrared-textured pointcloud be displayed in RViz, please?

ryosei1250, 2024-04-17:
I tried this command and got the following warning
[WARN] [1713353780.137373198] [camera.camera]: No stream match for pointcloud chosen texture Process - Depth

I tried to output with Rviv2, but it did not work
[image]
I don't know if my method is correct.
Changed fixed frame to camera_link.
added PointCloud2 and changed Topic to /camera/camera/depth/color/points

MartyG-RealSense, 2024-04-17:
I believe that your message type (Pointcloud2), Fixed Frame (camera_link) and Topic (/camera/camera/depth/color/points) are correct.

When not using the official cable supplied with the camera, the choice of your own cable can have an impact on whether a camera works correctly, because RealSense cameras need a high quality cable due to the amount of data that they can transmit through the cable.   This means that cables designed for data transfer should be used and not inexpensive ones intended only for device charging.  

Cables that are longer than 2 meters can have performance reduction or disconnection problems if they are not high quality.  If the cable is longer than 5 meters then it is recommendable to use a cable that contains 'active repeater' components to boost the signal as it moves through the cable.

ryosei1250, 2024-04-17:
Using anker's USB-C to Gigabit Ethernet Adapter did not change the results.
What do you think about this as a cable choice?

ryosei1250, 2024-04-17:
Using anker's USB-C to USB-C Thunderbolt 3.0 100W cable (0.7m) did not change the results.
What do you think about this as a cable choice?

ryosei1250, 2024-04-18:
The results were the same for other D455s.

MartyG-RealSense, 2024-04-18:
There have been past cases where there were problems with using Thunderbolt cables with RealSense.  A characteristic of some of those past cases was that the camera would work with the cable up until a certain point, but when greater volumes of data bandwidth needed to be transmitted through the cable (such as when increasing resolution) then the problems occurred.

ryosei1250, 2024-04-18:
You have had such a case. 

The realsense D455 firmware uses Version-5_15_1_0, but the Intel® RealSense™ Depth Module D455 is not supported. Should I install a different firmware?

MartyG-RealSense, 2024-04-18:
The D450 Depth Module and D450 Depth Camera are compatible with firmware driver 5.15.1.0, as shown in the image below.

[image]

ryosei1250, 2024-04-19:
No D455 is written at the URL there.
[image]

MartyG-RealSense, 2024-04-19:
For previous firmware files, the download launched immediately when the file link was clicked on.  5.15.1.0 is the first firmware version where you are taken to a download page instead.

5.15.1.0 is definitely compatible with D455, so it appears that the information on the download page is outdated.

MartyG-RealSense, 2024-04-19:
enable_infra1:=true enables the left-side infrared stream and pointcloud.stream_filter:=1 uses the enabled infrared stream as the texture for the pointcloud instead of the color stream.

ryosei1250, 2024-04-19:
Understood.
Again
ros2 launch realsense2_camera rs_launch.py enable_infra1:=true pointcloud.enable:=true pointcloud.stream_filter:=1
but it gives the error below
[WARN] [1713522593.229900794] [camera.camera]: No stream match for pointcloud chosen texture Process - Depth

MartyG-RealSense, 2024-04-19:
It sounds as though there are infrared frames being dropped so that some of the infrared frames are missing.  However, the pointcloud still seems able to successfully add the infrared image to the pointcloud despite this warning.

ryosei1250, 2024-04-19:
Should the problem of missing infrared frames be addressed?

MartyG-RealSense, 2024-04-19:
If it appears that the pointcloud is performing normally then warnings can be ignored.

ryosei1250, 2024-04-19:
ros2 launch realsense2_camera rs_launch.py enable_rgbd:=true enable_sync:=true align_depth.enable:=true enable_color:=true enable_depth:= true 
When I ran this command, rviz2 seemed to output
[image]

[image]

However, when I echo the topic, it outputs zero.

MartyG-RealSense, 2024-04-19:
If you are using `ros2 topic echo /camera/camera/rgbd` and there is an apparently correctly working image in RViz I am inclined to think that it is working okay despite the values of 0 for the frame_id in the topic echo.

ryosei1250, 2024-04-19:
ros2 launch realsense2_camera rs_launch.py enable_rgbd:=true align_depth.enable:=true enable_color:=true enable_depth:=true initial_reset :=true
When this command is executed

outputs the following error
[ERROR] [1713527046.297134347] [camera.camera]: To enable the rgbd topic, you must enable the color stream, depth stream, sync_mode and align_depth.

MartyG-RealSense, 2024-04-19:
It seems as though depth-color alignment is not going to work on your computer under any circumstances for unknown reasons, unfortunately, and an infrared-textured pointcloud is the only kind of mapping together of two streams that will work.

MartyG-RealSense, 2024-04-24:
Hi @ryosei1250  Bearing in mind the advice above about the non-likelihood of alignment working on your computer, do you require further assistance with this case please? Thanks!

closed 2024-05-08 by MartyG-RealSense