realsenseai/realsense-ros #3072: Cannot subscribe to color and depth streams at the same time over the network

benoit-robotics, 2024-04-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i                         | 
| Firmware Version                   | 5.15.1                         | 
| Operating System & Version |  Ubuntu 22.04 | 
| Kernel Version (Linux Only)    |  5.15.0-94-generic                                                                      | 
| Platform                                 | PC                         |
| Librealsense SDK Version       |  2.54.2                                                    | 
| Language                               |          | 
| Segment			         |  Robot                             | 
| ROS Distro			         |  Humble                                         | 
| RealSense ROS Wrapper Version   |  4.54.1                                                 | 

### Issue Description

I run the camera driver with **aligned depth enabled** with the following command `ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true camera_namespace:='/'`. The realsense is connected to a computer ( that I will call PC A) and I can subscribe to the aligned depth and color topics, _/camera/aligned_depth_to_color/image_raw_  and _/camera/color/image_raw_ at the same time with no problem. But I cannot subscribe to the same two topics at the same time from another computer ( PC B, A and B are on the same wired network ), if I do it, _/camera/aligned_depth_to_color/image_raw_ won't output anything. I also noticed that the CPU usage of the driver drops significantly when I do that. This problem does **not** happen if I subscribe to the normal depth topic instead ( so without enabling the depth alignment ). I initially thought that there was an issue with DDS because the problem is happening when data is transmitted over the network, but if I record the two topics into a rosbag on PC A and playback the rosbag on PC A ( so without running the realsense driver ), then I can subscribe to the 2 topics on PC B. So I think that there really is a problem with the realsense driver. For the record, I am using cyclonedds. A way to test the double subscriptions is simply to run "ros2 topic echo". 
Could you please let me know how I could solve this problem please ?
Thanks.

MartyG-RealSense, 2024-04-08:
Hi @benoit-robotics  Depth-color alignment is a processing-intensive function.  So my initial suspicion would be that performing alignment of the depth and color topics in real-time over a network connection is too much for it to handle.

You could first try defining a lower resolution for depth and color to see whether it makes a difference to whether the aligned depth topic is able to be published.

Another possibility would be to use **compressed** depth and color topics, which can be enabled by using the command below to install the image-transport-plugin package.

`sudo apt install ros-humble-image-transport`

https://dev.intelrealsense.com/docs/ros2-wrapper#compression-packages

benoit-robotics, 2024-04-08:
@MartyG-RealSense Changing the resolution of the rgb module to `640x360`  "fixes" the issue, but it's not a viable solution for me, this resolution is too low. I also need the raw images ideally.

MartyG-RealSense, 2024-04-09:
Does 1280x720 RGB at 6 FPS work?

MartyG-RealSense, 2024-04-15:
Hi @benoit-robotics  Do you require further assistance with this case, please?  Thanks!

benoit-robotics, 2024-04-17:
Hi. Sorry for the late reply. After having a closer look I think the problem is on my side. It took me some time to realise that the aligned depth images have a much larger bandwidth than the 'regular' depth images and it seems that the network that I use cannot cope with this additional bandwidth. Thank you for your time, I will close the issue.

closed 2024-04-17 by benoit-robotics

MartyG-RealSense, 2024-04-17:
It's no problem at all, @benoit-robotics - thanks very much for the update!