realsenseai/realsense-ros #3075: No RealSense devices were found!

polmagri, 2024-04-10:
| Camera Model                       |  D435i                                                
| Operating System & Version |   Linux Ubuntu 22.04 | 
| Kernel Version (Linux Only)   |  5.15.122                                                                
| Platform                                 |  NVIDIA Jetson orin nano jetpack 6                                                                                                        
| ROS                                         2  Humble 
Dscription:

After intstalled 
-sudo apt install ros-humble-realsense2-*
-sudo apt install ros-humble-librealsense2*
I have this problem:

Problem:
Distro			
[realsense2_camera_node-1]  01/01 10:45:33,664 WARNING [281472493676768] (d400-factory.cpp:1195) DS5 group_devices is empty.
[realsense2_camera_node-1] [WARN] [0000035133.664734880] [camera.camera]: No RealSense devices were found!

Someone know how i can solve it??

Thanks

MartyG-RealSense, 2024-04-11:
Hi @polmagri  The librealsense SDK does not yet have support for JetPack 6 at the time of writing this, with JetPack 5.0.2 being the latest currently supported version.  Installing librealsense on Jetson typically works best if instead of installing it from packages, it is built from source code with the **RSUSB backend** installation method.

An RSUSB build of the SDK bypasses the kernel and so having an unsupported JetPack version is no longer an issue..  

However, the **JetsonHacks** website's 'installLibrealsense.sh' build script may be the easiest option for an SDK build that does not require kernel patching.

https://github.com/jetsonhacks/installRealSenseSDK/tree/master?tab=readme-ov-file#installlibrealsensesh

polmagri, 2024-04-12:
Thanks.
I followed the instructions, and now the realsense-viewer command works. The issue is that I couldn't find which packages to install for ROS Humble. When I run the code:

`ros2 launch realsense2_camera rs_camera.launch`

it returns this error:

`Package 'realsense2_camera' not found: "package 'realsense2_camera' not found, searching: ['/home/jetson/ros2_ws/install/yolov8_bringup', '/home/jetson/ros2_ws/install/yolov8_ros', '/home/jetson/ros2_ws/install/yolov8_msgs', '/home/jetson/ros2_ws/install/ros2_aruco_interfaces', '/home/jetson/ros2_ws/install/ros2_aruco', '/home/jetson/ros2_ws/install/naoqi_driver', '/opt/ros/humble']"`

MartyG-RealSense, 2024-04-13:
The ROS2 wrapper for Humble can be installed in one of two ways:

1.  From packages with the command below.

`sudo apt install ros-humble-realsense2-*`

2.  Or from source code  Visit the link below and click on the small arrow beside **Step 3: Install Intel RealSense ROS2 wrapper** to expand open the list of available installation methods and then follow the instructions under **Option 2: install from source**.
3. 
https://github.com/IntelRealSense/realsense-ros?tab=readme-ov-file#installation-on-ubuntu

[image]

polmagri, 2024-04-15:
After following the instructions to install the SDK from source, this is the error I encounter:

`jetson@orin-desktop:~/ros2_ws$ ros2 run realsense2_camera realsense2_camera_node
[INFO] [0000001483.879095552] [camera.camera]: RealSense ROS v4.54.1
[INFO] [0000001483.879312064] [camera.camera]: Built with LibRealSense v2.54.1
[INFO] [0000001483.879344608] [camera.camera]: Running with LibRealSense v2.54.1
[INFO] [0000001484.495367552] [camera.camera]: Device with serial number 239722070764 was found.

[INFO] [0000001484.495838592] [camera.camera]: Device with physical ID 2-1.3-3 was found.
[INFO] [0000001484.495898144] [camera.camera]: Device with name Intel RealSense D435I was found.
[INFO] [0000001484.496482688] [camera.camera]: Device with port number 2-1.3 was found.
[INFO] [0000001484.496528832] [camera.camera]: Device USB type: 3.2
[INFO] [0000001484.498468416] [camera.camera]: getParameters...
[INFO] [0000001484.499292288] [camera.camera]: JSON file is not provided
[INFO] [0000001484.499342304] [camera.camera]: Device Name: Intel RealSense D435I
[INFO] [0000001484.499369056] [camera.camera]: Device Serial No: 239722070764
[INFO] [0000001484.499392896] [camera.camera]: Device physical port: 2-1.3-3
[INFO] [0000001484.499409248] [camera.camera]: Device FW version: 5.15.1
[INFO] [0000001484.499426592] [camera.camera]: Device Product ID: 0x0B3A
[INFO] [0000001484.499441568] [camera.camera]: Sync Mode: Off
[INFO] [0000001488.652829664] [camera.camera]: Stopping Sensor: Depth Module
[INFO] [0000001488.653280960] [camera.camera]: Stopping Sensor: RGB Camera
[INFO] [0000001488.653417184] [camera.camera]: Stopping Sensor: Motion Module
[INFO] [0000001488.867815360] [camera.camera]: Starting Sensor: Depth Module
 01/01 01:24:48,988 WARNING [281472493676768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/01 01:24:48,999 WARNING [281472493676768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[INFO] [0000001489.010556736] [camera.camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [0000001489.010721856] [camera.camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [0000001489.010764704] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
 01/01 01:24:49,011 WARNING [281472493676768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[INFO] [0000001489.013079872] [camera.camera]: Starting Sensor: RGB Camera
[INFO] [0000001489.043027328] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
 01/01 01:24:49,046 WARNING [281472728688864] (ds-calib-parsers.cpp:41) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[INFO] [0000001489.049210816] [camera.camera]: Starting Sensor: Motion Module
[INFO] [0000001489.061338080] [camera.camera]: Open profile: stream_type: Accel(0)Format: MOTION_XYZ32F, FPS: 100
[INFO] [0000001489.061495008] [camera.camera]: Open profile: stream_type: Gyro(0)Format: MOTION_XYZ32F, FPS: 200
[INFO] [0000001489.065550752] [camera.camera]: RealSense Node Is Up!
 01/01 01:24:49,226 WARNING [281472493676768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/01 01:24:49,237 WARNING [281472493676768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/01 01:24:49,350 WARNING [281472493676768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/01 01:24:49,361 WARNING [281472493676768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/01 01:24:49,373 WARNING [281472493676768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/01 01:24:49,995 WARNING [281472493676768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/01 01:24:50,006 WARNING [281472493676768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/01 01:24:50,017 WARNING [281472493676768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 01/01 01:24:50,027 WARNING [281472493676768] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`

If I launch the RealSense Viewer, I can see the image (there's also a firmware update suggestion).
If I open rViz, I can see that the camera is being read by the topic, but it's not receiving any image.
What can I do?

MartyG-RealSense, 2024-04-15:
Are frames published if you use **ros2 launch** instead of ros2 run and set a custom stream configuration with a low resolution of 640x480, please?

`ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=640x480x30 rgb_camera.color_profile:=640x480x30`

polmagri, 2024-04-15:
Yes, there is the same error

MartyG-RealSense, 2024-04-15:
**control_transfer returned** warnings indicate that there is a communication problem between the camera and the computer / computing device.  Sometimes though topics can still be successfully displayed in RViz even when the warning is generating continuously in the ROS terminal.

You mentioned that you installed the SDK from source.  Which method did you use to install the ROS wrapper afterwards, please?

polmagri, 2024-04-18:
After trying different ways I decided to install Jetpack 5.1.2.
Can you confirm me that  Realsense and libRealsense works for Ubuntu 20.04 with Ros 2 Humble?

polmagri, 2024-04-19:
Thanks for your help.
I confirm that on Jetpack 5.1.2. and ubuntu 20.04  with ros2 Humble following the  RSUSB backend installation method + Wrapper from source it works.

closed 2024-04-19 by polmagri

MartyG-RealSense, 2024-04-19:
It's great to hear that you were successful with your installation, @polmagri - thanks very much for the update!

szymonzmyslony, 2024-04-29:
Hi, I am facing similar problem on Jetson Orin Nano with Ubuntu 22.04 (Jetpack 6), I installed real-sense sdk from source (and it works, i can see camera images in the GUI), I then installing ros2 wrapper using both prebuilt packages and from source but am getting 
```shell
 ros2 run realsense2_camera realsense2_camera_node

[INFO] [1714398566.527810662] [camera]: RealSense ROS v4.54.1
[INFO] [1714398566.528561857] [camera]: Built with LibRealSense v2.54.1
[INFO] [1714398566.528650980] [camera]: Running with LibRealSense v2.54.1
 29/04 14:49:26,544 WARNING [281473000859872] (d400-factory.cpp:1195) DS5 group_devices is empty.
[WARN] [1714398566.544266254] [camera]: No RealSense devices were found!

``` 

Any ideas on how to make ROS2 wrapper work?

MartyG-RealSense, 2024-04-29:
Hi @szymonzmyslony  Does it make a difference if you reset the camera at launch, please?

`ros2 run realsense2_camera realsense2_camera_node --ros-args -p initial_reset:=true`

If that does not work, does using **ros2 launch** instead of ros2 run enable the camera to be detected?

`ros2 launch realsense2_camera rs_launch.py`

szymonzmyslony, 2024-04-29:
I fixed the installation now, I have RS D435I with 15.16.0.1, and realsenseviewer 2.55.1.0. Realsense installed from source, and the ROS wrapper too. I am able to see the camera in the GUI but for the ROS wrapper, I can see all topics, but only static properties are not null, such info on the camera, raw_image and other streams are empty. Im getting: 

```shell
 ros2 run realsense2_camera realsense2_camera_node --ros-args -p initial_reset:=true
[INFO] [1714419918.530653221] [camera.camera]: RealSense ROS v4.54.1
[INFO] [1714419918.530879913] [camera.camera]: Built with LibRealSense v2.55.1
[INFO] [1714419918.530928426] [camera.camera]: Running with LibRealSense v2.55.1
[INFO] [1714419919.050103639] [camera.camera]: Device with serial number 843112073908 was found.

[INFO] [1714419919.050329916] [camera.camera]: Device with physical ID 1-1-4 was found.
[INFO] [1714419919.050370940] [camera.camera]: Device with name Intel RealSense D435I was found.
[INFO] [1714419919.050805957] [camera.camera]: Device with port number 1-1 was found.
[INFO] [1714419919.050854822] [camera.camera]: Device USB type: 2.1
[WARN] [1714419919.050887015] [camera.camera]: Device 843112073908 is connected using a 2.1 port. Reduced performance is expected.
[INFO] [1714419919.050915815] [camera.camera]: Resetting device...
[INFO] [1714419925.675024296] [camera.camera]: Device with serial number 843112073908 was found.

[INFO] [1714419925.675212459] [camera.camera]: Device with physical ID 1-1-5 was found.
[INFO] [1714419925.675254028] [camera.camera]: Device with name Intel RealSense D435I was found.
[INFO] [1714419925.675723830] [camera.camera]: Device with port number 1-1 was found.
[INFO] [1714419925.675787607] [camera.camera]: Device USB type: 2.1
[WARN] [1714419925.675827608] [camera.camera]: Device 843112073908 is connected using a 2.1 port. Reduced performance is expected.
[INFO] [1714419925.676090717] [camera.camera]: getParameters...
[INFO] [1714419925.676965775] [camera.camera]: JSON file is not provided
[INFO] [1714419925.677035152] [camera.camera]: Device Name: Intel RealSense D435I
[INFO] [1714419925.677140114] [camera.camera]: Device Serial No: 843112073908
[INFO] [1714419925.677179027] [camera.camera]: Device physical port: 1-1-5
[INFO] [1714419925.677213524] [camera.camera]: Device FW version: 5.16.0.1
[INFO] [1714419925.677246676] [camera.camera]: Device Product ID: 0x0B3A
[INFO] [1714419925.677278101] [camera.camera]: Sync Mode: Off
[INFO] [1714419929.610842610] [camera.camera]: Stopping Sensor: Depth Module
[INFO] [1714419929.611211705] [camera.camera]: Stopping Sensor: RGB Camera
[INFO] [1714419929.611309211] [camera.camera]: Stopping Sensor: Motion Module
[INFO] [1714419929.847389060] [camera.camera]: Starting Sensor: Depth Module
[INFO] [1714419929.973987413] [camera.camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 640, Height: 480, FPS: 15
[INFO] [1714419929.974226906] [camera.camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 640, Height: 480, FPS: 15
[INFO] [1714419929.974272955] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 15
 29/04 20:45:29,977 WARNING [281471646099680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[INFO] [1714419929.977680255] [camera.camera]: Starting Sensor: RGB Camera
[INFO] [1714419929.995668681] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 15
 29/04 20:45:29,999 WARNING [281471646099680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/04 20:45:30,011 WARNING [281471881242848] (ds-calib-parsers.cpp:35) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[INFO] [1714419930.015408757] [camera.camera]: Starting Sensor: Motion Module
[INFO] [1714419930.026601814] [camera.camera]: Open profile: stream_type: Accel(0)Format: MOTION_XYZ32F, FPS: 63
[INFO] [1714419930.026766073] [camera.camera]: Open profile: stream_type: Gyro(0)Format: MOTION_XYZ32F, FPS: 200
[INFO] [1714419930.032457420] [camera.camera]: RealSense Node Is Up!
 29/04 20:45:30,295 WARNING [281471646099680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/04 20:45:30,306 WARNING [281471646099680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/04 20:45:30,316 WARNING [281471646099680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/04 20:45:30,328 WARNING [281471646099680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/04 20:45:30,545 WARNING [281471646099680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

```

MartyG-RealSense, 2024-04-30:
The **control_transfer returned** warning indicates that there is a communication problem between the camera and the computer.  Typically the problem will be with the USB port.  You are using an Orin Nano which has USB 3.2 ports but your ROS2 launch is detecting the USB connection as being the slower USB 2.1.  

Is the camera attached directly to a USB 3.2 port on the Orin Nano or to a USB hub?  If you were using a hub and it was a USB 2 hub then the camera could be detected as USB 2.1 even if the hub was attached to a USB 3.2 port on the Orin Nano.

szymonzmyslony, 2024-04-30:
Yes this is definitely part of the problem as I'm connecting directly via USB (not the hub) and it still thinks it uses 2.1. It also says Im using usb 2.1 when im doing ucb c <-> usb c

szymonzmyslony, 2024-04-30:
Yes, I was using the original one, the one that came with the camera itself

MartyG-RealSense, 2024-04-30:
As the Orin Nano has four USB 3.2 ports, I assume that you have tried the camera in all of those ports?

szymonzmyslony, 2024-04-30:
yup

MartyG-RealSense, 2024-04-30:
Is the camera detected as 3.2 if you insert it into the USB port with a firm, quick motion and not a slow insertion?

szymonzmyslony, 2024-04-30:
When I inserted in using a quick motion, it started working (able to see raw data in /camera/camera/color/image_raw topic) ! So thanks a lot ! I am still getting some errors though: 
```shell
jetson@jetson:[~]$ ros2 run realsense2_camera realsense2_camera_node
[INFO] [1714484184.088735911] [camera.camera]: RealSense ROS v4.54.1
[INFO] [1714484184.088945325] [camera.camera]: Built with LibRealSense v2.54.1
[INFO] [1714484184.088995310] [camera.camera]: Running with LibRealSense v2.54.1
[INFO] [1714484184.604534604] [camera.camera]: Device with serial number 843112073908 was found.

[INFO] [1714484184.604749554] [camera.camera]: Device with physical ID 2-1.4-3 was found.
[INFO] [1714484184.604789108] [camera.camera]: Device with name Intel RealSense D435I was found.
[INFO] [1714484184.605209471] [camera.camera]: Device with port number 2-1.4 was found.
[INFO] [1714484184.605253889] [camera.camera]: Device USB type: 3.2
[INFO] [1714484184.606426658] [camera.camera]: getParameters...
[INFO] [1714484184.607222232] [camera.camera]: JSON file is not provided
[INFO] [1714484184.607282682] [camera.camera]: Device Name: Intel RealSense D435I
[INFO] [1714484184.607318171] [camera.camera]: Device Serial No: 843112073908
[INFO] [1714484184.607350460] [camera.camera]: Device physical port: 2-1.4-3
[INFO] [1714484184.607377789] [camera.camera]: Device FW version: 5.15.0.2
[INFO] [1714484184.607405629] [camera.camera]: Device Product ID: 0x0B3A
[INFO] [1714484184.607432126] [camera.camera]: Sync Mode: Off
[INFO] [1714484188.370247913] [camera.camera]: Stopping Sensor: Depth Module
[INFO] [1714484188.370477583] [camera.camera]: Stopping Sensor: RGB Camera
[INFO] [1714484188.370544720] [camera.camera]: Stopping Sensor: Motion Module
[INFO] [1714484188.542999433] [camera.camera]: Starting Sensor: Depth Module
 30/04 14:36:28,629 WARNING [281472776202464] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 30/04 14:36:28,640 WARNING [281472776202464] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[INFO] [1714484188.651849287] [camera.camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1714484188.652044812] [camera.camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1714484188.652091021] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[INFO] [1714484188.655054137] [camera.camera]: Starting Sensor: RGB Camera
[INFO] [1714484188.679836819] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
 30/04 14:36:28,690 WARNING [281472801564896] (ds-calib-parsers.cpp:41) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[INFO] [1714484188.694781745] [camera.camera]: Starting Sensor: Motion Module
[INFO] [1714484188.710076184] [camera.camera]: Open profile: stream_type: Accel(0)Format: MOTION_XYZ32F, FPS: 63
[INFO] [1714484188.710257180] [camera.camera]: Open profile: stream_type: Gyro(0)Format: MOTION_XYZ32F, FPS: 200
[INFO] [1714484188.725786569] [camera.camera]: RealSense Node Is Up!
 30/04 14:36:28,961 WARNING [281472776202464] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 30/04 14:36:28,972 WARNING [281472776202464] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 30/04 14:36:28,983 WARNING [281472776202464] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 30/04 14:36:28,994 WARNING [281472776202464] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 30/04 14:36:29,005 WARNING [281472776202464] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

``` 

Is that sth I should care about / debug ? Also, could you explain why inserting quickly suddently works ?

MartyG-RealSense, 2024-04-30:
Yes, the log shows that the camera was detected as full speed USB 3.2 after a quick insertion motion.

If the streams are being published normally and work correctly in RViz then the **control_transfer returned** warnings generating in the log can be safely ignored.

If the camera's USB connector is inserted slowly into the USB port then when it reaches around three-quarters of the way inserted then it can be detected as USB 2.1 if the insertion is not fast enough.  A quick insertion ensures that the connector becomes fully inserted in the port before the port has an opportunity to mis-identity the connection type as USB 2.1.

szymonzmyslony, 2024-05-01:
Thank you, this answers the question !
I now have another problem with ```rs-imu-calibration.py```, as mentioned earlier I installed librealsense-2.54.1 (from source with USB flag), I have installed corresponding ```pyrealsense2```, and when try to run the script by ```python /home/jetson/realsense/librealsense-2.54.1/tools/rs-imu-calibration/rs-imu-calibration.py```
Im getting ``` /home/jetson/realsense/librealsense-2.54.1/tools/rs-imu-calibration/rs-imu-calibration.py
waiting for realsense device...
Error: module 'pyrealsense2' has no attribute 'context' ```
Is that a problem with package versioning ?

szymonzmyslony, 2024-05-01:
I have now cloned the script and run it separately from preinstalled library, and getting an error, this is same error as running it with newest release pyrealsense2:
```
 python script.py
waiting for realsense device...

Error: No RealSense device.
Traceback (most recent call last):
  File "/home/jetson/temp/script.py", line 732, in <module>
    main()
  File "/home/jetson/temp/script.py", line 721, in main
    if os.name == 'posix' and old_settings is not None:
UnboundLocalError: local variable 'old_settings' referenced before assignment
```
Sorry for the stream of problems but quite hard to figure this all out

MartyG-RealSense, 2024-05-01:
Does it make a difference if you change line 10 of rs-imu-calibration.py from **import pyrealsense2 as rs** to this:

`import pyrealsense2.pyrealsense2 as rs`

szymonzmyslony, 2024-05-01:
```  
File "/home/jetson/temp/script.py", line 10, in <module>
    import pyrealsense2.pyrealsense2 as rs
ModuleNotFoundError: No module named 'pyrealsense2.pyrealsense2'
```

MartyG-RealSense, 2024-05-01:
Okay, thanks, so the problem likely is not with the import line (sometimes the usual import line does not work on some Python setups).

I cannot see an obvious problem with your Python and pyrealsense2 installation.  It looks as though you used **pip install pyrealsense2** to install the wrapper on your Jetson and installation completed successfully.  Python 3.9 is compatible with this installation method on Arm devices such as Jetson.

After the **waiting for realsense device...** line, the output of rs-imu-calibration.py is meant to look like the example below, outputting the camera's information.

```
Device PID: 0B64
 Device name: Intel RealSense L515
 Serial number: 00000000f9340895
 Product Line: L500
 Firmware version: 01.04.01.00
checking minimum firmware requirement ...
 firmware 01.04.01.00 passed check. 
```

So this suggests that the script is not able to access the camera's information.  

You mentioned earlier that you were able to use the RealSense Viewer tool.  Could you try resetting the camera to its factory-new default calibration in the Viewer using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487 please to see whether the camera then works with rs-imu-calibration.py

ssharma0704, 2025-04-04:
Hey @szymonzmyslony  were you able to run it with jetpack 6.2? i am getting error that zero devices found.