realsenseai/realsense-ros #3078: d405 can not set  rgb_camera/ enable_auto_exposure 0

Nanmei, 2024-04-17:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D405  }                          | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |   { Ubuntu 18 } | 
| Kernel Version (Linux Only)    |  (e.g. 5.4)                                                                       | 
| Platform                                 | PC                         |
| Librealsense SDK Version       |  { 2.<?>.<?> }                                                    | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  {ROS1}                                           | 
| RealSense ROS Wrapper Version   |  {4.51.1, 4.54.1, etc..}                                                  | 

### Issue Description

**when i use d435i ,the following two cimmands can work:
rosrun dynamic_reconfigure dynparam set /camera_gap/rgb_camera/ enable_auto_exposure 0
rosrun dynamic_reconfigure dynparam set /camera_gap/rgb_camera/ exposure 166

but when i use d405 :**

[ INFO] [1713341247.746495672]: Initializing nodelet with 8 worker threads.
[ INFO] [1713341248.241251931]: RealSense ROS v2.3.2
[ INFO] [1713341248.241283427]: Built with LibRealSense v2.50.0
[ INFO] [1713341248.241329913]: Running with LibRealSense v2.50.0
[ INFO] [1713341248.269416078]:  
[ INFO] [1713341248.629359692]: Device with serial number 323622272730 was found.

[ INFO] [1713341248.629438774]: Device with physical ID 2-1.1-3 was found.
[ INFO] [1713341248.629485550]: Device with name Intel RealSense D405 was found.
[ INFO] [1713341248.633014510]: Device with port number 2-1.1 was found.
[ INFO] [1713341248.633104612]: Device USB type: 3.2
[ INFO] [1713341248.636497335]: getParameters...
[ INFO] [1713341248.685030628]: setupDevice...
[ INFO] [1713341248.685060044]: JSON file is not provided
[ INFO] [1713341248.685085762]: ROS Node Namespace: camera_gap
[ INFO] [1713341248.685104074]: Device Name: Intel RealSense D405
[ INFO] [1713341248.685117294]: Device Serial No: 323622272730
[ INFO] [1713341248.685132914]: Device physical port: 2-1.1-3
[ INFO] [1713341248.685149317]: Device FW version: 05.15.01.00
[ INFO] [1713341248.685163127]: Device Product ID: 0x0B5B
[ INFO] [1713341248.685178741]: Enable PointCloud: On
[ INFO] [1713341248.685205372]: Align Depth: Off
[ INFO] [1713341248.685224892]: Sync Mode: On
[ INFO] [1713341248.685305992]: Device Sensors: 
[ INFO] [1713341248.894875943]: Stereo Module was found.
[ INFO] [1713341248.894944079]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1713341248.894986983]: Add Filter: pointcloud
[ INFO] [1713341248.895821178]: num_filters: 1
[ INFO] [1713341248.895875342]: Setting Dynamic reconfig parameters.
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ INFO] [1713341258.560429310]: Done Setting Dynamic reconfig parameters.
[ INFO] [1713341258.566850820]: depth stream is enabled - width: 640, height: 480, fps: 90, Format: Z16
[ INFO] [1713341258.571760384]: color stream is enabled - width: 640, height: 480, fps: 60, Format: RGB8
[ INFO] [1713341258.572013042]: setupPublishers...
[ INFO] [1713341258.583846983]: Expected frequency for depth = 90.00000
[ INFO] [1713341258.646202047]: Expected frequency for color = 60.00000
[ INFO] [1713341258.718037157]: setupStreams...
 17/04 16:07:39,187 WARNING [140402044679936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1713341259.237877877]: SELECTED BASE:Depth, 0
[ INFO] [1713341259.245762947]: RealSense Node Is Up!
[ WARN] [1713341259.607734841]: 
 17/04 16:07:39,609 WARNING [140402044679936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

**rosrun dynamic_reconfigure dynparam set /camera_gap/rgb_camera/ enable_auto_exposure 0
the terminal always display:**
Waiting for service /camera_gap/rgb_camera/set_parameters...

[image]

**so how can set d405  enable_auto_exposure 0 and exposure 166? 
thanks.**

MartyG-RealSense, 2024-04-17:
Hi @Nanmei  The D405 camera model does not have a separate RGB sensor and instead its RGB is provided by the left infrared sensor and then passed through an Image Signal Processor (ISP) chip inside the camera to generate the final RGB image.  This means that the D405's RGB exposure is changed by altering the **depth** exposure value as D405 does not have a separate RGB exposure control due to the lack of a RGB sensor.

This also means that the RGB image's exposure cannot be set to 166 as the depth exposure range must be used (the default depth exposure value is 33000).

Nanmei, 2024-04-17:
> 你好@Nanmei D405 相机型号没有单独的 RGB 传感器，而是由门口红外传感器提供 RGB，然后通过相机内部的图像信号处理器（ISP）芯片最终生成的 RGB 图像。这意味着 D405 的 RGB 曝光是通过改变**深度**曝光来的实现的值来改变的，因为D405缺少RGB传感器而没有单独的RGB曝光控制。
> 
> 这也意味着RGB图像的曝光不能设置为166，必须因为使用深度曝光范围（默认深度曝光为33000），并且自动曝光由颜色和深度共享。

how to set deep exposure? Does setting the depth exposure affect the measurement of the depth value?

MartyG-RealSense, 2024-04-17:
Depth auto_exposure in the ROS1 wrapper can be disabled using the command below.

`rosrun dynamic_reconfigure dynparam set /camera/stereo_module enable_auto_exposure 0`

For setting a manual depth exposure value, please try this command and change '33000' to the value that you want to use.

`rosrun dynamic_reconfigure dynparam set /camera/stereo_module exposure 33000`

I ran manual exposure value tests with a D405 and found that on this camera model, changing the value did not have a noticeable effect on the depth quality or the depth value and both remained consistent.

Nanmei, 2024-04-19:
> Depth auto_exposure in the ROS1 wrapper can be disabled using the command below.
>
>
>
> `rosrun dynamic_reconfigure dynparam set /camera/stereo_module enable_auto_exposure 0`
>
>
>
> For setting a manual [...]

When the camera is moving fast, the image is blurred. How can I reduce the effect of this blur? I tried to reduce the exposure to 7500, but it didn't work。

MartyG-RealSense, 2024-04-19:
On a D405 you could try using 848x480 resolution and 90 FPS to reduce fast motion blur.

Nanmei, 2024-04-19:
> On a D405 you could try using 33000 exposure, 848x480 resolution and 90 FPS to reduce fast motion blur.

Ok thanks

closed 2024-05-04 by MartyG-RealSense