realsenseai/realsense-ros #3082: Getting depth stream information with 0 depth information at some points

20221869, 2024-04-20:
Subscribe to topics/camera/depth/image_raw to get depth coordinates of certain points，Depth coordinate is 0 in some frames of the camera, some frames are normal, open realsense-viewer and it's normal again!

MartyG-RealSense, 2024-04-20:
Hello, it sounds as though the camera is not having problems with reading the observed surface for depth information, at least when using the RealSense Viewer tool.  

The RealSense ROS wrapper operates as a layer on top of the RealSense SDK (librealsense) whilst the RealSense Viewer uses librealsense directly.  So there may sometimes be issues with streaming through the ROS wrapper that do not occur in librealsense applications, such as the Viewer.  

Are you using the ROS1 or ROS2 wrapper, please?

Does the occurrence of zero depth values decrease if you reduce the resolution and / or FPS in the ROS wrapper?

20221869, 2024-04-20:
ros2 ,Can you tell me how to reduce the resolution and/or FPS in the ROS wrapper?

MartyG-RealSense, 2024-04-20:
The example launch instruction below sets depth and color to 640x480 and 15 FPS.

`ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=640x480x15 rgb_camera.color_profile:=640x480x15`

closed 2024-05-04 by MartyG-RealSense