realsenseai/realsense-ros #3084: D435i, static/unavailable IMU data, ros melodic

z1V-Chan, 2024-04-22:
## Info
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.61.0.1         | 
| Operating System & Version | Ubuntu 18.04 LTS | 
| Kernel Version (Linux Only)    |  5.4.0-150-generic  | 
| Platform                                 | PC  |
| Librealsense SDK Version       |  librealsense2:amd64                          2.55.1-0~realsense.12427             | 
| Language                               |  C/python         | 
| Segment			         |  Robot  | 
| ROS Distro			         |  Melodic                                         | 
| RealSense ROS Wrapper Version   |  ros-melodic-librealsense2                    2.50.0-1bionic.20211115.141014</br>ros-melodic-realsense2-camera                2.3.2-1bionic.20221025.193216</br>ros-melodic-realsense2-description           2.3.2-1bionic.20230524.155943                                                 | 

## Issue Description

### `.lauch` file

```bash
/opt/ros/melodic/share/realsense2_camera$ sudo gedit ./launch/rs_camera.launch
```

```launch
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="false"/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="-1"/>
  <arg name="depth_height"        default="-1"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="true"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="848"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="-1"/>
  <arg name="color_height"        default="-1"/>
  <arg name="enable_color"        default="false"/>

  <arg name="fisheye_fps"         default="15"/>
  <arg name="depth_fps"           default="15"/>
  <arg name="infra_fps"           default="15"/>
  <arg name="color_fps"           default="15"/>
  <arg name="gyro_fps"            default="200"/>
  <arg name="accel_fps"           default="100"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
[... 373 lines cut ...]
[ 18.38355255   6.39139557 -14.91383839]
[0.44129923 1.706357   9.58109665]
[ 18.38355255   6.39139557 -14.91383839]
[0.44129923 1.706357   9.58109665]
[ 17.31890297   6.39139557 -20.24058342]
[0.4314926  1.73577702 9.62032318]
[ 17.31890297   6.39139557 -20.24058342]
[0.4314926  1.73577702 9.62032318]
[ 13.0585537    7.45604658 -21.30523491]
[0.46091253 1.77500355 9.60070992]
[ 13.0585537    7.45604658 -21.30523491]
[0.46091253 1.77500355 9.60070992]
[ -8.24842644   9.58709335 -28.76128006]
[0.47071919 1.79461694 9.56148338]
[ -8.24842644   9.58709335 -28.76128006]
[0.47071919 1.79461694 9.56148338]
[-21.03121758   5.32499933 -17.04314041]
^CTraceback (most recent call last):
  File "/home/wfj/d435i_calibration/demo.py", line 11, in <module>
    frames = pipeline.wait_for_frames()      
KeyboardInterrupt
```
The driver looked good. Moreover, the topic `/camera/depth/image_rect_raw` could be checked by *rviz* successfully.

Looking forward to your reply Orz...

MartyG-RealSense, 2024-04-22:
Hi @z1V-Chan  Because your camera is being detected as being on a slow USB 2.1 connection, the number of supported resolution / FPS combinations is more limited than on a USB 3.2 connection.  For example, the Infra2 stream is not supported on USB 2.1 and so **infra2_enable:=true** should be left out of the roslaunch instruction.

In regard to the **control_transfer returned** warnings, if they continuously generate in the log then this can indicate that there is a USB communcation problem between the camera annd your computer.  It is worth checking whether the streams are being published normally in RViz despite the warnings.

z1V-Chan, 2024-04-22:
> Hi @z1V-Chan Because your camera is being detected as being on a slow USB 2.1 connection, the number of supported resolution / FPS combinations is more limited than on a USB 3.2 connection. For example, the Infra2 stream is not supported on USB 2.1 and so **infra2_enable:=true** should be left out of the roslaunch instruction.
> 
> In regard to the **control_transfer returned** warnings, if they continuously generate in the log then this can indicate that there is a USB communcation problem between the camera and your computer. It is worth checking whether the streams are being published normally in RViz despite the warnings.

Thanks for your immediate reply. I have noticed **USB 2.1** and **control_transfer returned** warning. When I used **USB 2.1**, this kind of warnings appeared with a much less frequency than the time when i used **USB 3.2**. What's more, when using **USB 3.2**, everything is OK on *realsense-viewer* but nothing, such as `/camera/depth/image_rect_raw`, which could be checked when using **USB 2.1**, could be checked in *RViz*. I would quite appreciate it if u could give me more advice, please.

The following is the log when using **USB 3.2**
```log
wfj@EIP-laptop:~$ roslaunch realsense2_camera rs_camera.launch enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation enable_infra1:=true enable_infra2:=true
... logging to /home/wfj/.ros/log/e21c37a6-00a5-11ef-8ce4-e00af6a198d7/roslaunch-EIP-laptop-10469.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://EIP-laptop:40653/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 100
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 15
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: False
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 15
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
[... 131 lines cut ...]
 22/04 20:43:19,571 WARNING [140516533581568] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 22/04 20:43:19,596 WARNING [140516533581568] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 22/04 20:43:19,630 WARNING [140516533581568] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 22/04 20:43:19,655 WARNING [140516533581568] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 22/04 20:43:19,689 WARNING [140516533581568] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
^C 22/04 20:43:19,714 WARNING [140516533581568] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
[... cut ...]
 22/04 20:43:20,102 WARNING [140516533581568] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 22/04 20:43:20,136 WARNING [140516533581568] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
 22/04 20:43:20,161 WARNING [140516533581568] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
^C
```

What's worse, I had to unplug usb to kill the process instead of simply press `^C`.

MartyG-RealSense, 2024-04-22:
I note that you have updated your camera's firmware to version 5.16.0.1 because you are using the new 2.55.1 librealsense SDK release.  This may cause problems when using the camera with SDK 2.50.0 and wrapper 2.3.2 though, because the recommended firmware version for that older SDK is 5.13.0.50.  Using newer firmwares than 5.13.0.50 with 2.50.0 could result in instability and errors.

This is a problem because RealSense cameras only permit a maximum of 20 firmware downgrades before the ability to downgrade locks permanently and you can only update to newer firmwares than the currently installed one from that point onwards.  So it would be undesirable to repeatedly upgrade / downgrade the firmware depending on whether you were using the camera with the Viewer or the ROS wrapper.

z1V-Chan, 2024-04-23:
> I note that you have updated your camera's firmware to version 5.16.0.1 because you are using the new 2.55.1 librealsense SDK release. This may cause problems when using the camera with SDK 2.50.0 [...]

Thanks to @MartyG-RealSense . It finally worked!

## Downgrade SDK
```bash
sudo apt install librealsense2=2.50.0-0~realsense0.6127 librealsense2-udev-rules=2.50.0-0~realsense0.6127
sudo apt install librealsense2-dkms
sudo apt install librealsense2-gl=2.50.0-0~realsense0.6127 librealsense2-net=2.50.0-0~realsense0.6127
sudo apt install librealsense2-utils=2.50.0-0~realsense0.6127
sudo apt install librealsense2-dev=2.50.0-0~realsense0.6127
```

## Downgrade firmware
```bash
rs-fw-update -s 244222073051 -f ./D400_Series_FW_5_13_0_55/D400_Series_FW_5_13_0_55/Signed_Image_UVC_5_13_0_55.bin
```
Firmware could be downloaded from [https://dev.intelrealsense.com/docs/firmware-update-tool](https://dev.intelrealsense.com/docs/firmware-update-tool).

With much thx to @MartyG-RealSense again!

MartyG-RealSense, 2024-04-23:
You are very welcome.  I am pleased that I was able to help.

There is not a mechanism for unlocking a downgrade-locked camera, unfortunately.  The lock becomes permanent for that particular camera.  That is why it is important to avoid reaching the 20 downgrade limit if possible.

closed 2024-04-23 by z1V-Chan