realsenseai/realsense-ros #3087: Framerate changes to 15 fps when I disable auto exposure

Chenhui-Guo, 2024-04-22:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       |  D435i                    | 
| Firmware Version                   | 5.15.1                 | 
| Operating System & Version |   Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  5.15.0-105-generic                                                                      | 
| Platform                                 | PC                          |
| Librealsense SDK Version       |  2.54.2                                                     | 
| Language                               |  C++         | 
| Segment			         |  Robot                           | 
| ROS Distro			         |              noetic                             | 
| RealSense ROS Wrapper Version   | 2.3.2                                              | 

### Issue Description
When I turn off Auto Exposure of the rgb camera, the fps parameter will be invalid and the fps drops to 15. Whether I set the Auto Exposure parameter in realsense-viewer or in ros launch file, it looks like this:

- Disable Auto Exposure

[image]

- Enable Auto Exposure
 
[image]

Is it true that if I turn off Auto Exposure I can't set the frame rate, it only stays at 15 fps?

MartyG-RealSense, 2024-04-22:
Hi @Chenhui-Guo  I was not able to replicate your FPS drop when disabling auto-exposure.  I found though that if FPS was set to 60 like in the above image and I artifically forced an FPS drop by increasing the manual RGB exposure value then it returned to 60 if RGB exposure was set to a value of 78.

Chenhui-Guo, 2024-04-23:
Thanks for the reply! @MartyG-RealSense I tried it the way you said, after disabling auto-exposure, setting the RGB exposure value affects the frame rate. When the exposure value was set above 160, the frame rate started dropping from 60. 

- Exposure value is 160, frame rate is 59.53

[image]

- Exposure value is 700, frame rate is 14.25

[image]

I need to set a high exposure value for dark scenes, e.g. 600. Does that mean the RGB frame rate won't reach the 60 fps I set? Is low frame rate due to high exposure a hardware bottleneck for the D435i?

I still want to ask if there is any other way to get the frame rate up? Looking forward to your reply!

Chenhui-Guo, 2024-04-24:
Thanks for the advice, you've solved my doubts. I will try to increase RGB Brightness.

closed 2024-04-24 by Chenhui-Guo

MartyG-RealSense, 2024-04-24:
You are very welcome.  I'm pleased that I could help.  :)