realsenseai/realsense-ros #3088: decimation_filter invalid

davidlmljll, 2024-04-28:
### decimation_filter parameter setting is invalid on ros2

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D435i}                          | 
| Firmware Version                   | (V2.54.2)                          | 
| Operating System & Version |   { Linux (Ubuntu 22) } | 
| Kernel Version (Linux Only)    |  (5.15.122-tegra)                                                                       | 
| Platform                                 | NVIDIA Jetson orin nano                           |
| Librealsense SDK Version       |  { 2.54.2 }                                                    | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  {Robot }                             | 
| ROS Distro			         |  {Humble}                                           | 
| RealSense ROS Wrapper Version   |  {4.54.1}                                                  | 

### Issue Description
When I use realsense-viewer and set the decimation_filter.filter_magnitude parameter, I can see changes in the 3D point cloud. But when I set the same parameters under the ros2 driver, there is no obvious change in the pointcloud2 visualized using rviz2. What is the reason?

MartyG-RealSense, 2024-04-28:
Hi @davidlmljll  A ROS2 user at https://github.com/IntelRealSense/realsense-ros/issues/2396#issuecomment-1244290055 who set decimation_filter.filter_magnitude in their launch instruction said that the default magnitude of 2 "kinda worked" and they used '3' as their value.  This suggests that using '3' as the magnitude may provide a more visible difference than 2 in the ROS2 wrapper.

davidlmljll, 2024-04-30:
> Hi @davidlmljll A ROS2 user at [#2396 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/2396#issuecomment-1244290055) who set `decimation_filter.filter_magnitude` in their launch instruction said that the default magnitude of 2 "kinda worked" and they used '3' as their value. This suggests that using '3' as the magnitude may provide a more visible difference than 2 in the ROS2 wrapper.

Thanks for your reply, I tried setting decimation_filter.filter_magnitude=6 (other filtering parameters are the same) but I still found that there is a difference between ros2 and realsense-viewer, as shown in the screenshot below, what is the reason?
[image]
[image]

MartyG-RealSense, 2024-04-30:
Because the ROS2 image has less holes in it than the Viewer image, this may give the impression that the ROS2 decimation filter is not working.  This may not be the case though.  The decimation filter does not remove detail from the depth image but instead reduces the resolution of the image's pixels so that it looks more blurry and pixellated.  

The value set for the Magnitude of the decimation filter determines what the original resolution will be divided by when 'downsampling' it (3 = divided by 3, 6 = divided by 6, etc).  

So my opinion is that the difference between the two images is not because the ROS2 wrapper's decimation filter is failing but because the Viewer's pointcloud is poor quality with large areas that have no depth.

davidlmljll, 2024-04-30:
> Because the ROS2 image has less holes in it than the Viewer image, this may give the impression that the ROS2 decimation filter is not working. This may not be the case though. The decimation filter [...]

Ok, thanks for the reply！

closed 2024-04-30 by davidlmljll