realsenseai/realsense-ros #3090: D405 Unable to set color_profile parameter

hello-fazil, 2024-04-30:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       |  D405                           | 
| Firmware Version                   | 5.16.0.1                          | 
| Operating System & Version |   Ubuntu 22.04  | 
| Kernel Version (Linux Only)    |  6.5.0-28-generic                                                                       | 
| Platform                                 | NUC12                           |
| Librealsense SDK Version       |  v2.55.1                                                    | 
| Language                               |  None        | 
| Segment			         |  None                             | 
| ROS Distro			         | Humble                                           | 
| RealSense ROS Wrapper Version   |  v4.54.1                                                  | 

### Issue Description
Hi, we are trying to set the parameter `depth_module.color_profile` to value `424x240x15` which does not seem to work with the D405 camera instead the camera always streams at `848x480x10` profile. I have tried setting this parameter through the Launch file as well as using the command `ros2 param set /camera depth_module.color_profile 424x240x15` when the camera node is already running which does not do anything.  
Thanks in advance!

MartyG-RealSense, 2024-04-30:
Hi @hello-fazil  As the D405 is not equipped with an RGB sensor and instead obtains RGB from the depth sensor, color uses the same resolution that depth is set to.  What happens if you just set the depth profile?

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=424x240x15 enable_depth:=true enable_color:=true`

hello-fazil, 2024-05-02:
Hi @MartyG-RealSense, Thanks for reply. Here is the output of the command you mentioned
```bash
$ ros2 launch realsense2_camera rs_launch.py depth_module.profile:=424x240x15 enable_depth:=true enable_color:=true
[INFO] [launch]: All log files can be found below /home/hello-robot/.ros/log/2024-05-01-19-53-17-423626-stretch-se3-3002-25030
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [25031]
[realsense2_camera_node-1] [INFO] [1714618397.783325520] [camera.camera]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO] [1714618397.783432822] [camera.camera]: Built with LibRealSense v2.54.2
[realsense2_camera_node-1] [INFO] [1714618397.783438139] [camera.camera]: Running with LibRealSense v2.54.2
[realsense2_camera_node-1] [INFO] [1714618397.798572347] [camera.camera]: Device with serial number 130322271074 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1714618397.798604735] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.3/3-3.3.3/3-3.3.3.1/3-3.3.3.1:1.0/video4linux/video8 was found.
[realsense2_camera_node-1] [INFO] [1714618397.798610460] [camera.camera]: Device with name Intel RealSense D405 was found.
[realsense2_camera_node-1] [INFO] [1714618397.798725359] [camera.camera]: Device with port number 3-3.3.3.1 was found.
[realsense2_camera_node-1] [INFO] [1714618397.798731648] [camera.camera]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [1714618397.798736169] [camera.camera]: Device 130322271074 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1] [INFO] [1714618397.799886324] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1714618397.800083173] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1714618397.800090628] [camera.camera]: Device Name: Intel RealSense D405
[realsense2_camera_node-1] [INFO] [1714618397.800095171] [camera.camera]: Device Serial No: 130322271074
[realsense2_camera_node-1] [INFO] [1714618397.800099343] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.3/3-3.3.3/3-3.3.3.1/3-3.3.3.1:1.0/video4linux/video8
[realsense2_camera_node-1] [INFO] [1714618397.800104146] [camera.camera]: Device FW version: 5.15.1
[realsense2_camera_node-1] [INFO] [1714618397.800108302] [camera.camera]: Device Product ID: 0x0B5B
[realsense2_camera_node-1] [INFO] [1714618397.800111920] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1714618397.908838939] [camera.camera]: Set ROS param depth_module.depth_profile to default: 848x480x10
[realsense2_camera_node-1] [INFO] [1714618397.909139590] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x10
[realsense2_camera_node-1] [INFO] [1714618397.913301879] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1714618397.930595493] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1714618397.957643774] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 848, Height: 480, FPS: 10
[realsense2_camera_node-1] [INFO] [1714618397.957687046] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 10
[realsense2_camera_node-1] [INFO] [1714618397.959528912] [camera.camera]: RealSense Node Is Up!
```
Still the color image seems to stream at 848X480X10 this time.

MartyG-RealSense, 2024-05-02:
It makes sense that color would be 848x480 as it has to use the resolution that depth has been set to.  848x480 is the default resolution for depth on most RealSense 400 Series camera models except D415.

Because your USB connection is being detected in the **Device USB type** line of the log as slow USB 2.1 instead of full speed USB 3.2, the FPS of the streams has been set as 10 FPS instead of 30.

On a USB 2.1 connection, the only FPS available on a D405 at the higher 1280x720 resolution is 5 FPS.  If you need 30 FPS then you would have to set the resolution to 640x80.

It seems unusual that your D405 is being detected as USB 2.1 when it is being used with a NUC 12 which should have USB 3.2 ports.  

Is the camera plugged directly into the NUC's USB ports or is it plugged into a USB hub?

If the camera is plugged directly into a USB port on the NUC, does Device USB type show as 3.2 in the log if you insert the camera into the port with a quick, firm insertion motion before you perform the ros2 launch?

anisotropicity, 2024-05-05:
For me this stopped working with firmware version 15.16.0.1, it seems that depth_module.profile was replaced with depth_module.depth_profile and depth_module.infra_profile.
However, when I set depth_module.infra_profile and depth_module.depth_profile the output on the console reflected these changes but I received no frames.
Rolling back the firmware to version 5.15.1 resolved this issue for me

MartyG-RealSense, 2024-05-05:
The RealSense ROS wrapper 4.54.1 was originally designed for librealsense 2.54.1.  2.54.2 was mostly bug fixes rather than new features, so it works fine with 2.54.2.

However, 2.55.1 introduces a lot of feature and API changes.  In that situation, you ideally want to be using a ROS wrapper designed for that specific librealsense version.  There is not a wrapper version for 2.55.1 at the time of writing this though.

The official version recommendations for the 2.54.1 wrapper are librealsense 2.54.1 and firmware 5.15.0.2.

anisotropicity, 2024-05-11:
Found the solution for me. The realsense-ros/realsense2_camera/launch/rs_launch.py is missing the `depth_module.color_profile` parameter from the `configurable_parameters` list.
Adding `depth_module.color_profile` to the `configurable_parameters` list. allowed me to set this parameter when using the rs_launch.py launch file and configure the color stream resolution and frame rate.

MartyG-RealSense, 2024-05-11:
Thanks so much for sharing your solution and creating a PR  :)

MartyG-RealSense, 2024-05-17:
Hi again @anisotropicity  As you created a PR related to this issue, I have added an **Enhancement** label to the case to indicate that it should be kept open whilst your PR is active.  Thanks again!

MartyG-RealSense, 2026-10-31:
Case closed due to the PR https://github.com/realsenseai/realsense-ros/pull/3096 linked to this issue being merged in May 2024.

closed 2026-10-31 by MartyG-RealSense