realsenseai/realsense-ros #3092: ros launching d435 stuck

danielhertz, 2024-05-05:
laptop msi i7-11, 16gb, 3070
ros2 iron
RealSense ROS v4.54.1
LibRealSense v2.54.1
realsense d435
kernel 6.5
ubuntu 22.04

we talked before about an bug i had in the sdk and i resulved it by disabling the builtin webcam in the pc
the launching proccess is being stuck and not savining the nodes, also i tried to run the rviz2 whle the process is stuck and some topics where running(but not the 3d map), after i doing ctrl+c on that process the topics and headers arent being saved and the rviz2 is stop working, i dont know on what process its being stuck and how to resulve this problem

[image]

[image]

daniel@daniel-Pulse-GL66-11UGKV:~$ rviz2
[INFO] [1714906003.051504953] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1714906003.051547282] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [1714906003.063206896] [rviz2]: Stereo is NOT SUPPORTED

daniel@daniel-Pulse-GL66-11UGKV:~$ ros2 node list
ros2 topic list
/camera/camera
/camera/color/camera_info
/camera/color/image_raw
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/imu
/parameter_events
/rosout
/tf_static
daniel@daniel-Pulse-GL66-11UGKV:~$ 
----------------------------------------------------------------
after i interupting the launching

daniel@daniel-Pulse-GL66-11UGKV:~$ ros2 launch realsense2_camera rs_launch.py
[INFO] [launch]: All log files can be found below /home/daniel/.ros/log/2024-05-05-13-46-25-989884-daniel-Pulse-GL66-11UGKV-8556
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [8559]
[realsense2_camera_node-1] [INFO] [1714905986.087658587] [camera.camera]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO] [1714905986.087705761] [camera.camera]: Built with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1714905986.087711927] [camera.camera]: Running with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1714905986.092983984] [camera.camera]: Device with serial number 831612074153 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1714905986.093004404] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb4/4-4/4-4:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1714905986.093009742] [camera.camera]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1714905986.093114431] [camera.camera]: Device with port number 4-4 was found.
[realsense2_camera_node-1] [INFO] [1714905986.093120342] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1714905986.093544307] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1714905986.093637361] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1714905986.093644202] [camera.camera]: Device Name: Intel RealSense D435
[realsense2_camera_node-1] [INFO] [1714905986.093648293] [camera.camera]: Device Serial No: 831612074153
[realsense2_camera_node-1] [INFO] [1714905986.093652214] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb4/4-4/4-4:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1714905986.093656349] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1714905986.093659988] [camera.camera]: Device Product ID: 0x0B07
[realsense2_camera_node-1] [INFO] [1714905986.093663367] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1714905986.102367799] [camera.camera]: Set ROS param depth_module.profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1714905986.106591361] [camera.camera]: Set ROS param rgb_camera.profile to default: 640x480x30
[realsense2_camera_node-1] [INFO] [1714905986.111443364] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1714905986.113192914] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1714905986.121835205] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1714905986.121977712] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1714905986.132450115] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1714905986.133906667] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1714905986.134979139] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1714905986.251677718] [camera.camera]: 
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[realsense2_camera_node-1] [INFO] [1714906124.468520258] [rclcpp]: signal_handler(signum=2)
[realsense2_camera_node-1] [INFO] [1714906124.536052882] [camera.camera]: Stop Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1714906124.536084047] [camera.camera]: Close Sensor. 
[realsense2_camera_node-1] [INFO] [1714906124.538387619] [camera.camera]: Close Sensor - Done. 
[realsense2_camera_node-1] [INFO] [1714906124.538410903] [camera.camera]: Stop Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1714906124.538417474] [camera.camera]: Close Sensor. 
[realsense2_camera_node-1] [INFO] [1714906124.650097993] [camera.camera]: Close Sensor - Done. 
[INFO] [realsense2_camera_node-1]: process has finished cleanly [pid 8559]
daniel@daniel-Pulse-GL66-11UGKV:~$ 

daniel@daniel-Pulse-GL66-11UGKV:~$ ros2 node list
ros2 topic list
/parameter_events
/rosout
daniel@daniel-Pulse-GL66-11UGKV:~$ 

ty so much

MartyG-RealSense, 2024-05-05:
Hi @danielhertz  Firmware 5.16.0.1 should only be used with the current latest librealsense version 2.55.1.  For SDK 2.54.1, firmware **5.15.0.2** should be used.

danielhertz, 2024-05-07:
still the same problem, even i updated the sdk and deletet the remains of the old one
[image]

also i found this problem, maybe it connected to this ?

daniel@daniel-Pulse-GL66-11UGKV:~$ sudo dmesg | grep uvcvideo
[ 1421.555528] usbcore: registered new interface driver uvcvideo
[ 1614.185648] uvcvideo 4-8:1.1: Non-zero status (-71) in video completion handler.
[ 1614.206139] uvcvideo 4-8:1.4: Non-zero status (-71) in video completion handler.
[ 3803.407762] uvcvideo 4-8:1.1: Non-zero status (-71) in video completion handler.
[ 3803.427012] uvcvideo 4-8:1.4: Non-zero status (-71) in video completion handler.
[ 6391.137861] usbcore: deregistering interface driver uvcvideo
[ 6391.586155] usbcore: registered new interface driver uvcvideo
[ 6599.273901] usbcore: deregistering interface driver uvcvideo
[ 6599.516651] usbcore: registered new interface driver uvcvideo
[ 9260.463885] uvcvideo 4-8:1.1: Non-zero status (-71) in video completion handler.
[ 9260.483135] uvcvideo 4-8:1.4: Non-zero status (-71) in video completion handler.
[ 9675.228885] uvcvideo 4-8:1.1: Non-zero status (-71) in video completion handler.
[ 9675.259134] uvcvideo 4-8:1.4: Non-zero status (-71) in video completion handler.
[ 9712.596634] uvcvideo 4-8:1.1: Non-zero status (-71) in video completion handler.
[ 9712.603633] uvcvideo 4-8:1.2: Non-zero status (-71) in video completion handler.
daniel@daniel-Pulse-GL66-11UGKV:~$ 
daniel@daniel-Pulse-GL66-11UGKV:~$ 
daniel@daniel-Pulse-GL66-11

MartyG-RealSense, 2024-05-07:
The error `Non-zero status (-71) in video completion handler` has not been previously reported in relation to RealSense.

There have been a couple of cases where a launch that failed on ROS2 Iron worked on Humble.  Do you have the ability to change your ROS version to Humble, please?

danielhertz, 2024-05-09:
i changed to humble and its the same 

[image]

MartyG-RealSense, 2024-05-10:
After further research I was able to find some very old past RealSense references to `Non-zero status (-71) in video completion handler` dating back to 2018-2019.  On the few occasions that it was reported, it was mostly in relation to the librealsense SDK rather than the ROS wrapper.   

It is a **uvcvideo** type error, which usually indicates that there is a conflict between the librealsense SDK and the kernel.  

The notes at the top of this case list 6.5 as your kernel version and 2.54.1 as your librealsense version.  Support for kernel 6.5 was not added until the current latest 2.55.1 version of librealsense.  I note that you already tried installing 2.55.1 above though.  Can you confirm that you rebuilt the 4.54.1 ROS wrapper after updating to 2.55.1?  It is necessary to build the wrapper again each time that the librealsense version is changed.

The kernel can be bypassed to help avoid kernel conflicts if librealsense is built from source code with the 'libuvc backend' or 'RSUSB backend' installation methods.  Which method did you use to install librealsense on your computer?

danielhertz, 2024-05-11:
so thats why its always downgrading my sdk version at the launching?
coz at first it was SDK 2.55.1and after a day it was 2.54.1 
[image]

```
daniel@daniel-Pulse-GL66-11UGKV:~$ ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true
[INFO] [launch]: All log files can be found below /home/daniel/.ros/log/2024-05-11-13-07-57-986347-daniel-Pulse-GL66-11UGKV-5250
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [5251]
[realsense2_camera_node-1] [INFO] [1715422078.346648536] [camera.camera]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO] [1715422078.346755717] [camera.camera]: Built with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1715422078.346785276] [camera.camera]: Running with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO] [1715422078.356668973] [camera.camera]: Device with serial number 831612074153 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1715422078.356698847] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb4/4-4/4-4:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1715422078.356708469] [camera.camera]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1715422078.356951137] [camera.camera]: Device with port number 4-4 was found.
[realsense2_camera_node-1] [INFO] [1715422078.356961953] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1715422078.358256993] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1715422078.358387827] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1715422078.358397492] [camera.camera]: Device Name: Intel RealSense D435
[realsense2_camera_node-1] [INFO] [1715422078.358403974] [camera.camera]: Device Serial No: 831612074153
[realsense2_camera_node-1] [INFO] [1715422078.358409875] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb4/4-4/4-4:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1715422078.358415946] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1715422078.358421526] [camera.camera]: Device Product ID: 0x0B07
[realsense2_camera_node-1] [INFO] [1715422078.358426834] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1715422078.370214251] [camera.camera]: Set ROS param depth_module.profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1715422078.376894653] [camera.camera]: Set ROS param rgb_camera.profile to default: 640x480x30
[realsense2_camera_node-1] [INFO] [1715422078.387960636] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1715422078.392028392] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1715422078.414874722] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1715422078.415916475] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1715422078.426174656] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1715422078.428280592] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1715422078.430819150] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1715422078.513705279] [camera.camera]: 

```

MartyG-RealSense, 2024-05-11:
After you launch ROS with 2.54.1, does it still become stuck?  Are you able to view streams in RViz after the **RealSense Node Is Up** message?

danielhertz, 2024-05-15:
yes i can, but wait, im still string to figure it up since im having issues with the TF, its not letting me c it in the all range(x,y,z) of the 3d map

MartyG-RealSense, 2024-05-16:
Okay, no problem.  Please let me know when you have further news to report from your tests.  Good luck!

MartyG-RealSense, 2024-05-23:
Hi @danielhertz  Do you have an update about this case that you can provide, please?  Thanks!

danielhertz, 2024-05-29:
im still working on it even theres an progresss i have failures, ill let u know as soon as i can, ty so much

MartyG-RealSense, 2024-05-29:
You are very welcome.  I look forward to your next update.  Good luck!

closed 2024-06-25 by MartyG-RealSense