realsenseai/realsense-ros #3093: Raspberry low framerate

TheSav1101, 2024-05-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435                          | 
| Firmware Version                   | latest (I don't have the camera right now as I will regain access to it on friday)| 
| Operating System & Version |    Ubuntu 22 [armbian jammy, should be equivalent] | 
| Kernel Version (Linux Only)    |  (6.6.23)                                                                       | 
| Platform                                 | Raspberry Pi 5                           |
| Librealsense SDK Version       |  2.54.1                                                   | 
| Language                               |  C++         | 
| Segment			         |  Robot                              | 
| ROS Distro			         |  Humble                                         | 
| RealSense ROS Wrapper Version   |  4.54.1                                        | 

### Issue Description
Hello, I am trying to use a raspberry pi 5 with realsense-ros to run some tflite models. I have a problem with the framerate since the messages from the camera node are sent at ~13.5Hz and not 30Hz as expected. Since this happens even when the camera node is the only node running I wanted to ask if there are known possible causes for this problem.

MartyG-RealSense, 2024-05-06:
Hi @TheSav1101  Are you using depth only or are depth and color enabled?  

If both depth and color are published, when you have the camera again please test whether the FPS frequency improves if you add **auto_exposure_priority:=false** to your launch instruction.  For example:

`ros2 launch realsense2_camera rs_launch.py auto_exposure_priority:=false`

TheSav1101, 2024-05-08:
Hello, @MartyG-RealSense I am currently publishing depth and color, but now that you mention it I probably don't need both, I will make a couple of tests on Friday with different settings. Sorry for the trouble, I didn't even think about that...

closed 2024-05-08 by TheSav1101

MartyG-RealSense, 2024-05-08:
It's no trouble at all.  Good luck!