realsenseai/realsense-ros #3099: ROS 2 Image Display Issues

ekfrckato, 2024-05-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D415   | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |   Linux | 
| Kernel Version (Linux Only)    |  6.5.0  | 
| Platform                                 | PC      |
| Librealsense SDK Version       |  { 2.55.1 }            | 
| Language                               | python   | 
| Segment			         |  Robot                       | 
| ROS Distro			         |  humble         | 
| RealSense ROS Wrapper Version   |                                              | 

### Issue Description

Until yesterday, everything was working fine. However, starting today, I can no longer load images through the Realsense SDK.

I confirmed that the Realsense camera is working properly in the viewer, so it does not seem to be a device recognition issue. When I open the Realsense camera, I use this command:
`ros2 launch realsense2_camera rs_launch.py clip_distance:=1.0 spatial_filter.enable:=true temporal_filter.enable:=true decimation_filter.enable:=true enable_color:=true rgb_camera.profile:=1280x720x30 initial_reset:=true enable_depth:=true`
The node reports that it has opened successfully.

To test whether the image is being received correctly, I enter `ros2 run image_tools showimage image:=/camera/color/image_raw` in another terminal window, and it outputs the following:

> :~$ ros2 run image_tools showimage image:=/camera/color/image_raw
> [WARN] [1715422644.252381392] [rcl]: Found remap rule 'image:=/camera/color/image_raw'. This syntax is deprecated. Use '--ros-args --remap image:=/camera/color/image_raw' instead.
> [INFO] [1715422644.381085594] [showimage]: Subscribing to topic 'image'

I think there is a problem in receiving the message. When I enter
`ros2 topic echo /camera/color/image_raw`
it only displays a blank space and there is no response.

Yesterday, I was working and changed the version of setuptools to 58.2.0 using` pip install setuptools==58.2.0 `because I encountered a 'setup.py' error during colcon build. There were alerts about version conflicts with OpenCV and TensorFlow, but I thought it would not affect my current task, which involves processing depth data through Realsense. Indeed, everything worked fine even after changing the version yesterday. I'm not sure if this is the cause of the issue, but it's the only change I made.

Today, in an attempt to resolve the issue, I upgraded the Realsense SDK, and the current version is as specified above. Please help! 😭

MartyG-RealSense, 2024-05-11:
Hi @ekfrckato  You should continue to use librealsense 2.54.1 with ROS wrapper 4.54.1, as there is not a wrapper for librealsense 2.55.1 at the time of writing this.  Ideally the librealsense version that should be used is the one listed in the release notes for a specific ROS wrapper verison.

Whenever the librealsense version is changed, the ROS wrapper needs to be built again.

ekfrckato, 2024-05-12:
Then, how can i change the version? 
I did `dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

`sudo mkdir -p /etc/apt/keyrings
curl -sSf https://librealsense.intel.com/Debian/librealsense.pgp | sudo tee /etc/apt/keyrings/librealsense.pgp > /dev/null`

`echo "deb [signed-by=/etc/apt/keyrings/librealsense.pgp] https://librealsense.intel.com/Debian/apt-repo 'lsb_release -cs' main" | \
sudo tee /etc/apt/sources.list.d/librealsense.list
sudo apt-get update`

and tried this `sudo apt-get install librealsense2-utils=2.54.1~0realsnese0.9591` 
> E: Version '2.54.1~0realsnese0.9591' for 'librealsense2-utils' was not found

`sudo apt-get install librealsense2-utils=2.54.1~0realsnese.tbd`
> E: Version '2.54.1~0realsnese.tbd' for 'librealsense2-utils' was not found

I am following this https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md and [#10526](https://github.com/IntelRealSense/librealsense/issues/10526). so confused which file has to be installed.

MartyG-RealSense, 2024-05-12:
If you are trying to install a specific version from packages then it is more difficult than building from a source code folder for a particular version.  It involves using the command **apt-cache showpkg librealsense2** to find the names of the packages for that version.  More information about this can be found at https://github.com/IntelRealSense/librealsense/issues/5773#issuecomment-581477785

The two compulsory packages that need to be installed are librealsense2-dkms and librealsense2-utils

ekfrckato, 2024-05-12:
Yes, I already found the packages.
  librealsense2-gl,librealsense2 2.54.1-0~realsense.9591
  librealsense2-dbg,librealsense2 2.54.1-0~realsense.9591
  librealsense2-utils,librealsense2 2.54.1-0~realsense.9591
  librealsense2-dev,librealsense2 2.54.1-0~realsense.9591
  librealsense2-gl,librealsense2 2.54.1-0~realsense.9588
  librealsense2-utils,librealsense2 2.54.1-0~realsense.9588
  librealsense2-dev,librealsense2 2.54.1-0~realsense.9588
  librealsense2-dbg,librealsense2 2.54.1-0~realsense.9588
but -dkms file isn't there.

and `sudo apt-get install librealsense2-utils=2.54.1~0realsnese0.9591` didn't work.

Is is `apt-cache showpkg librealsense2` whole of output

> Package: librealsense2
> Versions: 
> 2.55.1-0~realsense.12474 (/var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_jammy_main_binary-amd64_Packages)
>  Description Language: 
>                  File: /var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_jammy_main_binary-amd64_Packages
>                   MD5: d2bf76470afbccd342d993c3cbea3505
>  Description Language: 
>                  File: /var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_jammy_main_binary-amd64_Packages
>                   MD5: d2bf76470afbccd342d993c3cbea3505
>  Description Language: 
>                  File: /var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_jammy_main_binary-amd64_Packages
>                   MD5: d2bf76470afbccd342d993c3cbea3505
>  Description Language: 
>                  File: /var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_jammy_main_binary-amd64_Packages
>                   MD5: d2bf76470afbccd342d993c3cbea3505
>  Description Language: 
>                  File: /var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_jammy_main_binary-amd64_Packages
>                   MD5: d2bf76470afbccd342d993c3cbea3505
>  Description Language: 
>                  File: /var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_jammy_main_binary-amd64_Packages
>                   MD5: d2bf76470afbccd342d993c3cbea3505
>  Description Language: 
>                  File: /var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_jammy_main_binary-amd64_Packages
>                   MD5: d2bf76470afbccd342d993c3cbea3505
>  Description Language: 
>                  File: /var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_jammy_main_binary-amd64_Packages
>                   MD5: d2bf76470afbccd342d993c3cbea3505
> 
> 2.55.1-0~realsense.12429 (/var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_jammy_main_binary-amd64_Packages)
>  Description Language: 
>                  File: /var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_jammy_main_binary-amd64_Packages
>                   MD5: d2bf76470afbccd342d993c3cbea3505
>  Description Language: 
>                  File: /var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_jammy_main_binary-amd64_Packages
>                   MD5: d2bf76470afbccd342d993c3cbea3505
>  Description Language: 
>                  File: /var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_jammy_main_binary-amd64_Packages
>                   MD5: d2bf76470afbccd342d993c3cbea3505
>  Description Language: 
>                  File: /var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_jammy_main_binary-amd64_Packages
>                   MD5: d2bf76470afbccd342d993c3cbea3505
>  Description Language: 
>                  File: /var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_jammy_main_binary-amd64_Packages
>                   MD5: d2bf76470afbccd342d993c3cbea3505
>  Description Language: 
>                  File: /var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_jammy_main_binary-amd64_Packages
>                   MD5: d2bf76470afbccd342d993c3cbea3505
>  Description Language: 
>                  File: /var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_jammy_main_binary-amd64_Packages
>                   MD5: d2bf76470afbccd342d993c3cbea3505
>  Description Language: 
>                  File: /var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_jammy_main_binary-amd64_Packages
>                   MD5: d2bf76470afbccd342d993c3cbea3505
> 
> 2.55.1-0~realsense.12426 (/var/lib/apt/lists/librealsense.intel.com_Debian_apt-repo_dists_jammy_main_binary-amd64_Packages)
[... 185 lines cut ...]
>   librealsense2-dbg,librealsense2 2.53.1-0~realsense0.8251
>   librealsense2-utils,librealsense2 2.53.1-0~realsense0.8251
>   librealsense2-net,librealsense2 2.53.1-0~realsense0.8251
>   librealsense2-gl,librealsense2 2.53.1-0~realsense0.8251
>   librealsense2-dev,librealsense2 2.53.1-0~realsense0.8251
> Dependencies: 
> 2.55.1-0~realsense.12474 - libusb-1.0-0 (2 2:1.0.16) libudev1 (2 183) libc6 (2 2.34) libgcc-s1 (2 3.3.1) libstdc++6 (2 12) librealsense2-udev-rules (0 (null)) 
> 2.55.1-0~realsense.12429 - libusb-1.0-0 (2 2:1.0.16) libudev1 (2 183) libc6 (2 2.34) libgcc-s1 (2 3.3.1) libstdc++6 (2 12) librealsense2-udev-rules (0 (null)) 
> 2.55.1-0~realsense.12426 - libusb-1.0-0 (2 2:1.0.16) libudev1 (2 183) libc6 (2 2.34) libgcc-s1 (2 3.3.1) libstdc++6 (2 12) librealsense2-udev-rules (0 (null)) 
> 2.55.1-0~realsense.12423 - libusb-1.0-0 (2 2:1.0.16) libudev1 (2 183) libc6 (2 2.34) libgcc-s1 (2 3.3.1) libstdc++6 (2 12) librealsense2-udev-rules (0 (null)) 
[... cut ...]
> 2.54.1-0~realsense.9588 - libusb-1.0-0 (2 2:1.0.16) libudev1 (2 183) libc6 (2 2.34) libgcc-s1 (2 3.3.1) libstdc++6 (2 12) librealsense2-udev-rules (0 (null)) 
> 2.53.1-0~realsense0.8251 - libusb-1.0-0 (2 2:1.0.16) libudev1 (2 183) libc6 (2 2.34) libgcc-s1 (2 3.3.1) libstdc++6 (2 12) librealsense2-udev-rules (0 (null)) 
> Provides: 
> 2.55.1-0~realsense.12474 - 
> 2.55.1-0~realsense.12429 - 
> 2.55.1-0~realsense.12426 - 
> 2.55.1-0~realsense.12423 - 
> 2.54.2-0~realsense.10773 - 
> 2.54.1-0~realsense.9591 - 
> 2.54.1-0~realsense.9588 - 
> 2.53.1-0~realsense0.8251 - 
> Reverse Provides: 
>

ekfrckato, 2024-05-12:
I successfully changed the version. After running `sudo apt-get clean`, I installed the packages using the command `sudo apt-get install librealsense2-utils=2.54.1-0~realsense.9591 librealsense2=2.54.1-0~realsense.9591 librealsense2-gl=2.54.1-0~realsense.9591`

MartyG-RealSense, 2024-05-12:
That's great to hear that you were successful with the version change!

ekfrckato, 2024-05-12:
There's a new issue. I confirmed that the RealSense SDK 2.54.1 is installed and the RealSense camera is recognized properly using `realsense-viewer`. Since you mentioned that a ROS wrapper build was necessary after installing the SDK, I ran the following command in my ROS2 workspace:
`:~/ros2_ws$ colcon build --symlink-install`

But then I encountered this error:

> --- stderr: realsense2_camera_msgs
> failed to create symbolic link '/home/cae/ros2_ws/build/realsense2_camera_msgs/ament_cmake_python/realsense2_camera_msgs/realsense2_camera_msgs' because existing path cannot be removed: Is a directory
> gmake[2]: *** [CMakeFiles/ament_cmake_python_symlink_realsense2_camera_msgs.dir/build.make:70: CMakeFiles/ament_cmake_python_symlink_realsense2_camera_msgs] Error 1
> gmake[1]: *** [CMakeFiles/Makefile2:421: CMakeFiles/ament_cmake_python_symlink_realsense2_camera_msgs.dir/all] Error 2
> gmake[1]: *** Waiting for unfinished jobs....
> gmake: *** [Makefile:146: all] Error 2
> Failed <<< realsense2_camera_msgs [3.08s, exited with code 2]
> 
> Summary: 3 packages finished [3.41s]
> 1 package failed: realsense2_camera_msgs
> 1 package had stderr output: realsense2_camera_msgs
> 2 packages not processed

So I tried to reinstall the realsense2 package. **I removed the realsense-ros package from the ros2_ws** folder and deleted everything related to realsense in the build. I was following the installation for Ubuntu from this link:
https://github.com/IntelRealSense/realsense-ros?tab=readme-ov-file#installation-on-ubuntu 
But when I ran colcon build, it said **'Intel RealSense SDK 2.0 is missing.'** Wasn't the SDK installed correctly?

MartyG-RealSense, 2024-05-12:
In the Ubuntu installation instructions, did you use the **Option 1: Install debian package from ROS servers** instructions before then doing **Option 2: Install from source** afterwards?

When building the wrapper from source with these instructions, you are meant to ignore the Option 1 section and only do the tasks in Option 2.

ekfrckato, 2024-05-12:
Unfortunately, despite not having executed Option 1 when I initially posted my question, an error still occurred. Later, due to my mistake, I executed Option 1 and then Option 2, but the same error persists. If executing Option 1 caused something to go awry, I would like to uninstall and reinstall everything. Is there a way to completely remove the above wrapper? I would also like to seek further advice on resolving this issue. Thank you for your help.

ekfrckato, 2024-05-12:
[image]

When I executed Option 2, an error occurred during the Install dependencies section. Should I delete that file and reinstall it?
> ERROR: default sources list file already exists:
> 	/etc/ros/rosdep/sources.list.d/20-default.list

MartyG-RealSense, 2024-05-12:
RealSense SDK and wrapper installations based on DKMS package files can be removed from the computer with the purge instruction you mentioned earlier.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

If the ROS wrapper was built from source code then completely deleting the **/ros2_ws/src/** catkin workspace folder will remove it.

If the RealSense SDK was built with CMake with a source code folder then go to its 'librealsense > build' folder and from the build directory perform the command below to uninstall the SDK and clean up CMake.

**sudo make uninstall && make clean**

Advice for dealing with this rosdep error can be found at https://github.com/IntelRealSense/librealsense/issues/10562

In that case, the RealSense user deleted the sources file list as the text of the error suggested but then found that they had the same error.  What worked for them was to uninstall rosdep with the command below:

`sudo rm /etc/ros/rosdep/sources.list.d/20-default.list`

Then when they repeated the ROS wrapper installation process, they were able to reinstall the rosdep dependency with `sudo rosdep init`

ekfrckato, 2024-05-13:
I think I had **a problem with the source.list and installing a specific version of the realsense SDK.** 

**First,** by **deleting the rosdep source.list and reinstalling rosdep** as you mentioned
(`sudo rm /etc/ros/rosdep/sources.list.d/20-default.list`) ,
> ERROR: default sources list file already exists:
> /etc/ros/rosdep/sources.list.d/20-default.list

the error mentioned above was resolved. 

**Secondly,** there was still the same realsense missing issue during the `colcon build`, but this was resolved by **setting the PATH.**

The colcon build command I tried to solve the second problem was:
`colcon build --symlink-install --packages-select realsense2_camera_msgs --cmake-clean-cache`
This worked normally, but it printed the following along with a path warning:

> If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at runtime.
> If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at runtime may occur.
> 
> If you understand the risks and want to override a package anyways, add the following to the command line:
> --allow-overriding realsense2_camera_msgs
> 
> This may be promoted to an error in a future release of colcon-override-check.
> 

And colcon build also did not work due to the same missing error.
So the second method I tried involved the following two commands:

```
export COLCON_PREFIX_PATH=/home/<usrname>/ros2_ws/install:$COLCON_PREFIX_PATH
export CMAKE_PREFIX_PATH=/home/<usrname>/vcpkg/installed/x64-linux/share/realsense2:$CMAKE_PREFIX_PATH
```
By specifying the SDK's path through these two commands,` colcon build `now works properly.

Actually the colcon prefix path warning still appears, stating 

> [0.366s] colcon.colcon_core.prefix_path.colcon WARNING The path '/home/cae/ros2_ws/src/install' in the environment variable COLCON_PREFIX_PATH doesn't exist. 

However, aside from this one path warning, there are no more issues with building.

ekfrckato, 2024-05-13:
I think the ros-realsense package was built successfully. However, as shown in the picture below, I still can't tell what data is being published in the topic.
[image]

MartyG-RealSense, 2024-05-13:
I cannot tell what is in the above echo of camera/color/image_raw.

Can the streams be displayed in RViz?

ekfrckato, 2024-05-13:
**My goal is to receive depth data published by realsense-ros from a package I created.** I want to process depth data for a specific location. However, the code I wrote in my package is unable to display the data using imshow. It's hard to say it's a code issue, as it's the same code I've always used.

So, I wanted to check if the realsense-ros package is properly transmitting depth data and raw image data. To do this, I adopted the commands `ros2 run image_tools showimage image:=/camera/color/image_raw `and` ros2 topic echo /camera/color/image_raw` from the photo above. **If this is the wrong method of verification, what other method could I adopt?** Thanks as always for your help.

MartyG-RealSense, 2024-05-13:
I would agree that a topic echo such as `ros2 topic echo /camera/color/image_raw` is a correct way of quickly checking whether messages are being published on a topic.  Once the publishing of data on the topic is confirmed then I would recommend then trying to display the topic in RViz to confirm that there is a visual output.

ekfrckato, 2024-05-14:
I checked that the topic's image is displayed correctly through Rviz. Does this mean there is no issue with the msg output? Then, is it possible that the only problem is with the code that receives my msg? I haven't modified that particular Python file.. I'm not sure what steps to take next..

MartyG-RealSense, 2024-05-20:
Hi @ekfrckato  Do you require further assistance with this case, please?  Thanks!

ekfrckato, 2024-05-22:
I haven't solved the problem yet. I'll try again and reply if there's any problem or progress.

MartyG-RealSense, 2024-05-22:
Okay, thanks very much for the update.  Good luck!

ekfrckato, 2024-05-25:
I've solved my problem. It was a very simple issue and my mistake. For some reason, it seems that I modified the topic path of the code that was working well. Originally, it worked well when using `/camera/camera/depth/image_rect_raw`, but the issue was that I was subscribing to the `/camera/depth/image_rect_raw depth_image` topic.... Thank you so much for all the help during this time.

MartyG-RealSense, 2024-05-25:
You are very welcome.  It's great to hear that you found a solution.  Thanks very much for the update!

closed 2024-06-03 by MartyG-RealSense