realsenseai/realsense-ros #3103: Even after enabling gyro, accel values along with unite_imu argument neither is any data published in gyro or accel topics nor are the accel and gyro values are united as a single topic called IMU.

AjayAdit, 2024-05-15:
Hi all, even after enabling gyro, accel values along with unite_imu argument neither is any data published in gyro or accel topics (I checked using rostopic echo) nor are the accel and gyro values united as a single topic called IMU, but I am able to see live camera filed in Rviz. This was the result of running the following  launch file with arguments: 

`roslaunch realsense2_camera rs_camera.launch enable_accel:=true enable_gyro:=true unite_imu_method:=2`

My system configuration and logs for my error are as follows:

 **Required Info**                                                                                                                
|---------------------------------|---------------------------------------- |
| Camera Model                       | { D435i }                          | 
| Firmware Version                   | (2.55.1.0)                          | 
| Operating System & Version |   {  Linux (Ubuntu 20) } | 
| Kernel Version (Linux Only)    |  5.15.0-106-generic       | 
| Platform                                 | Ubuntu PC                         |
| Librealsense SDK Version       |  { LibRealSense v2.50.0}   | 
| Language                               |  {ROS 1 noetic ,python }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }       | 
| ROS Distro			         |  {Noetic}                                           | 
| RealSense ROS Wrapper Version   |  {v2.3.2}  | 

**Above said launch file (rs_camera.launch ) script:**
```

<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="false"/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="-1"/>
  <arg name="depth_height"        default="-1"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="true"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="848"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="-1"/>
  <arg name="color_height"        default="-1"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="-1"/>
  <arg name="depth_fps"           default="-1"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="-1"/>
  <arg name="gyro_fps"            default="-1"/>
  <arg name="accel_fps"           default="-1"/>
  <arg name="enable_gyro"         default="false"/>
  <arg name="enable_accel"        default="false"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>
[... 306 lines cut ...]
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/gyro/imu_info
/camera/gyro/metadata
/camera/gyro/sample
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static
```

I have tried to reinstall realsense and changed a couple of a parameter in launch files with no success so any inputs will be very helpful.

MartyG-RealSense, 2024-05-15:
Hi @AjayAdit  On ROS1 the **unite_imu_method** parameter uses 'linear_interpolation' instead of '2'.

`unite_imu_method:=linear_interpolation`

The warning **control_transfer returned** indicates that there is a communication problem between the camera and the computer, likely in the USB system (the USB port or the cable).  Sometimes though the streams will be displayed correctly in RViz even if the control_transfer returned warning is generating continuously in the log.

Does it make a difference if you reset the camera at launch by adding to the roslaunch instruction` initial_reset:=true`

AjayAdit, 2024-05-15:
Hi @MartyG-RealSense, Thanks for the heads up adding linear_interpolation merged gyro and accel into IMU as a single topic but it still does not publish imu data.

Additionally I did try to out the reset command but the same error of  control_transfer returned error appears.

New command I gave is:
`roslaunch realsense2_camera rs_camera.launch enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation initial_reset:=true`

launch file logs:
```

roslaunch realsense2_camera rs_camera.launch enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation initial_reset:=true
... logging to /home/ml/.ros/log/fbd63b56-129d-11ef-a689-9176d4dd17dd/roslaunch-ml-20-4517.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ml-20:35855/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
[... 90 lines cut ...]
[ INFO] [1715765528.303239170]: Motion Module was found.
[ INFO] [1715765528.303263100]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1715765528.303281980]: num_filters: 0
[ INFO] [1715765528.303293335]: Setting Dynamic reconfig parameters.
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ INFO] [1715765528.776190010]: Done Setting Dynamic reconfig parameters.
[ INFO] [1715765528.777055676]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1715765528.779149906]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1715765528.779214472]: gyro stream is enabled - fps: 200
[ INFO] [1715765528.779240429]: accel stream is enabled - fps: 63
[ INFO] [1715765528.779266023]: setupPublishers...
[ INFO] [1715765528.781865642]: Expected frequency for depth = 30.00000
[ INFO] [1715765528.809520867]: Expected frequency for color = 30.00000
[ INFO] [1715765528.817838921]: Start publisher IMU
[ INFO] [1715765528.818394054]: setupStreams...
 15/05 15:02:08,867 WARNING [139849252181760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1715765528.918994798]: SELECTED BASE:Depth, 0
[ INFO] [1715765528.936981600]: RealSense Node Is Up!
 15/05 15:02:09,170 WARNING [139849252181760] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 15/05 15:02:09,425 WARNING [139849252181760] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 15/05 15:02:55,587 WARNING [139849252181760] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
```

MartyG-RealSense, 2024-05-15:
If librealsense SDK 2.50.0 is being used then the camera firmware version should be **5.13.0.50** instead of 5.16.0.1.  Using newer firmwares with old SDK versions can result in errors.

AjayAdit, 2024-05-17:
Hi, actually was trying to implement the above suggestion for sometime now but faced the following issues. In order to ensure the compatibility of firmware and SKD in my case either the camera firmware should be downgraded or the SDK's should be updated. 

As realsense viewer only has an option for camera firmware upgrade not downgrade so explored the other possibility of upgrading the SKD to latest release. Then I found out that the binary installation is always little behind the latest SDK release so I tried mannual installation and build by following this [documentation](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md).

The last instruction in the above documentation is to build the librealsense using cmake ../ but I am face the following error.

The terminal log of trying to build librealsense is:
```

ml@ml-20:~/Downloads/librealsense-2.55.1/build$ cmake ../
-- Checking internet connection...
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.55.1
-- Setting Unix configurations
-- No output directory set; using /home/ml/Downloads/librealsense-2.55.1/build/Release/
-- CHECK_FOR_UPDATES depends on BUILD_GRAPHICAL_EXAMPLES flag, turning it off..
-- using RS2_USE_V4L2_BACKEND
-- Fetching nlohmann/json...
Cloning into 'json'...

```

Ps: I am trying to build the librealsense-2.55.1 version which is in my download folder.
[image]

I have tried to double check all the process I have done  all seems to be in order not sure what went wrong please suggest if other methods.

MartyG-RealSense, 2024-05-17:
The firmware version can be downgraded in the Viewer by first downloading a firmware **.bin** file from the link below and then going to the 'More' option near the top of the Viewer's options side-panel and selecting **Upgrade Firmware** from its drop-down menu to bring up a file selector that you can use to select the downloaded .bin and initiate the downgrade.

https://dev.intelrealsense.com/docs/firmware-releases-d400

However, RealSense cameras have a limit of 20 downgrades before downgrade becomes permanently locked and then you can only upgrade to a newer firmware than the currently installed one.

I cannot see anything wrong in your build procedure.  As 2.55.1 is not a suitable librealsense version to use with the 2.3.2 ROS1 wrapper though, it may be better to download the firmware bin file for 5.13.0.50 and then install it in your camera with the Viewer's 'Update Firmware' option.

Yrazu, 2024-05-17:
Hi @MartyG-RealSense and @AjayAdit, 

My issue was almost exactly the same as @AjayAdit and the only difference was my hardware firmware was too far forward.

My current Ubuntu version is **18.04**, Ros is **Melodic**. 

@MartyG-RealSense, I fixed my issue with your previous comment on pushing the **Hardware Firmware back to 5.12.0.50** for **librealsense v2.50.0** and **Realsense ROS Wrapper 2.3.2.**

 I am now seeing data being published on **/camera/gyro/sample and /camera/accel/sample** and when I use 

`roslaunch realsense2_camera rs_camera.launch enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation`

I see data on **/camera/imu** data stream.

@AjayAdit I recommend what @MartyG-RealSense  said with making sure you librealsense and ROS wrapper are correct version. (I did a clean install to make sure nothing conflicted) and then push your hardware firmware back to what he suggested.

MartyG-RealSense, 2024-05-18:
Hi @Yrazu  Thanks so much for sharing with @AjayAdit what worked for you!

AjayAdit, 2024-05-20:
I had actually run the below command and got the following response:
 
`roslaunch realsense2_camera rs_camera.launch enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation initial_reset:=true`

The launch log is:

```
roslaunch realsense2_camera rs_camera.launch enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation initial_reset:=true
... logging to /home/ml/.ros/log/f3546eee-1667-11ef-b3fc-17c9df78b9a1/roslaunch-ml-20-59146.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ml-20:46465/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
[... 100 lines cut ...]
[ INFO] [1716182129.222823044]: setupStreams...
[ WARN] [1716182129.679764771]: 
[ WARN] [1716182129.679851435]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
 20/05 10:45:29,785 WARNING [140597717341952] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1716182129.836160532]: SELECTED BASE:Depth, 0
[ INFO] [1716182129.840423305]: RealSense Node Is Up!
 20/05 10:45:29,907 WARNING [140597717341952] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/05 10:45:29,957 WARNING [140597717341952] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/05 10:45:30,009 WARNING [140597717341952] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/05 10:45:30,158 WARNING [140597717341952] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/05 10:45:30,209 WARNING [140597717341952] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/05 10:45:30,362 WARNING [140597717341952] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/05 10:45:30,413 WARNING [140597717341952] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/05 10:45:30,565 WARNING [140597717341952] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
 20/05 10:46:56,874 WARNING [140597717341952] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/05 10:47:02,930 WARNING [140597717341952] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/05 10:51:11,217 WARNING [140597717341952] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```

In this I am little sceptical of the control_transfer returned error in the end please suggest any resolve for that.

MartyG-RealSense, 2024-05-20:
Sometimes the topics will publish normally and be displayed correctly in RViz even if the **control_transfer returned** warning is generating continuously in the log.  If the streams are behaving normally then the warning (which indicates a possible commnication issue between camera and computer) can be safely ignored.

There is not a solution for removing the warnings.  At best, you can reduce the number of them.

AjayAdit, 2024-05-20:
Got it thank you for the support, Now all streams seem to be fine so will close this issue.

closed 2024-05-22 by AjayAdit

donipologhimire, 2025-03-08:
I had same issue, /camera/gyro/sample was not publishing any info. 
Downgrading to 5.13.0.50 worked. Thank you @AjayAdit @Yrazu .