realsenseai/realsense-ros #3107: Troubles with T265 IMU coordinate system.

alejandronespereira, 2024-05-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | t265 | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |   Ubuntu 22.04| 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  6.20.0-36-generic                          | 
| Language                            |      C++                      | 
| Segment			|  Robot | 

### Issue Description

 I am having trouble wrapping my head around the t265 IMU coordinate system. I am trying to integrate the IMU data as well as the odometry produced by the tracking camera. I see in the official documentation [here ](https://www.intelrealsense.com/how-to-getting-imu-data-from-d435i-and-t265/) that the coordinate system is a Right Hand convention with the Z pointing inwards the camera, and Y point upwards. 
 
However, the ROS2 wrapper is outputing the IMu in a frame that does not exist on the tf that is published, camera_imu_optical_frame. i will refer to the axes denoted on this image:
[image]
as XYZ (in caps) from now onwards, and the gravity vector G as pointing _towards_ the ground.

Here are the values for the linear acceleration depending on the camera orientation:

- X = G (side of the camera with the USB connection facing the ground): 
x: 9.8, y: 0, z: 0
- Y = G (camera upside down):
x: 0, y: -9.8, z: 0
- Z = G (camera facing the sky):
x: 0, y: 0, z: 9.8

I don''t understand why the Y axis is negative, but again the camera_imu_optical_frame has no connection to the camera that I know. 

What am I missing?

JTShuai, 2024-07-11:
@alejandronespereira Hi, I have the same trouble. Have you found any updates on it?

alejandronespereira, 2024-07-17:
Nope, sorry. I recall I ended up publishing some tfs myself and taking those into account for a while,  but as the camera is discontinued I ended up with another solution that did not use this model.

closed 2026-10-23 by MartyG-RealSense