realsenseai/realsense-ros #3108: Expected timeframe for Jazzy support

KolinGuo, 2024-05-24:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D415 / D435 / D435i / L515                          | 
| Firmware Version                   | 5.16.0.1                          | 
| Operating System & Version |  Ubuntu 24.04 Noble Numbat | 
| Kernel Version (Linux Only)    | 6.8.0         | 
| Platform                                 | PC                |
| Librealsense SDK Version       |  N/A                                                    | 
| Language                               |  C++ / python        | 
| Segment			         |  Robot     | 
| ROS Distro			         |  Jazzy / Rolling    | 
| RealSense ROS Wrapper Version   |  N/A       | 

### Issue Description
Please don't view this as any form of pressure. I'm just curious about when can I expect `realsense-ros` (and `librealsense`) to add support for Ubuntu 24.04 and ROS 2 Jazzy? 

Thank you for your attention and greatly appreciate the efforts in maintaining `realsense-ros`!

Nir-Az, 2024-05-26:
Hi @MohamedAlKayd ,
This work is planned, hopefully we can make it for the next release of LibRS + ROS.
Please keep track :)

gregoryw3, 2024-05-27:
With the removal of L515 is the current suggestion to compile 2.54.2 (or 2.50.0) and corresponding ros wrapper for Ubuntu 24.04 (or other Jazzy supported distro) if you need L515? Or potentially is it possible to use a Docker container to use the old SDK, Wrappers, and Ubuntu, just for L515 while the host system can use the latest SDK and Drivers?

MartyG-RealSense, 2024-05-27:
@gregoryw3  You should be able to use your L515 with an older Ubuntu version in a Docker container (such as 22.04) than the Ubuntu on the host, yes.  At the time of writing this, the ROS2 wrapper currently supports Humble or Iron on Ubuntu 22.04.

For a ROS1 setup you should ideally use Ubuntu 18.04, ROS1 Noetic, SDK 2.50.0 or 2.51.1 and wrapper 2.3.2.

MartyG-RealSense, 2024-06-03:
Does anyone require further assistance with this case, please?  Thanks!

MohamedAlKayd, 2024-06-03:
> Hi @MohamedAlKayd , This work is planned, hopefully we can make it for the next release of LibRS + ROS. Please keep track :)

Thanks for replying, greatly appreciated. I am currently working on a project with the Camera and the versions I told you about, do you know when the next release  of LibRS + ROS would be?

MartyG-RealSense, 2024-06-03:
Hi @MohamedAlKayd  Advance release dates for librealsense SDK or ROS wrapper updates are not provided.  In regard to librealsense, I recommend planning for a gap of roughly six months between releases.

New ROS wrapper releases usually do not occur until after a new librealsense version has been released.  However, the update model that the previous ROS wrapper 4.54.1 used was to apply regular updates to the source code version of the wrapper instead of waiting for the next librealsense release.  So installing the source code version of a wrapper instead of the package version should ensure that you have the latest updates.

MartyG-RealSense, 2024-06-10:
Case closed due to no further comments received.

closed 2024-06-10 by MartyG-RealSense

MohamedAlKayd, 2024-07-20:
@MartyG-RealSense @Nir-Az 
Any updates regarding the wrapper for the D415 for ROS Jazzy? It has been 2 months.
Regards.

MartyG-RealSense, 2024-07-20:
Hi @MohamedAlKayd  Earlier in this discussion at https://github.com/IntelRealSense/realsense-ros/issues/3108#issuecomment-2132114923 @Nir-Az stated that it is hoped that Jazzy support will be available in time for the next librealsense + ROS release.

2 weeks ago at https://github.com/IntelRealSense/librealsense/discussions/13072 it was further announced that the next release is planned for around the end of Q3 2024, which would be the end of September 2024.

Nir-Az, 2024-07-20:
I can also update that initial version was already published for Jazzy
Please check and update :-)

MohamedAlKayd, 2024-07-26:
@Nir-Az 
Thanks for the update, I was able to download it but it is not able to build. 

[image]

This is what I got when I downloaded it. I first try to build the realsense2_camera_msgs package and then the realsense2_camera package. I get a package issue when trying to build the first package.

Have you tested it with ros2 jazzy?

Nir-Az, 2024-07-28:
> @Nir-Az Thanks for the update, I was able to download it but it is not able to build. [...]

Are you building according to our manual using `colcon build` command?
Note that it is also available through Debain install on ROS Jazzy.

MohamedAlKayd, 2024-07-28:
@Nir-Az 
Where is your manual? I am using Colcon Build but did it by myself.
Could you please provide the manual for colcon build and the Debian command? That would be greatly appreciated.
Regards.

MartyG-RealSense, 2024-07-28:
Hi @MohamedAlKayd  You can find installation instructions for package build and colcon build by visiting the link below and clicking on **Step 3: Install Intel® RealSense™ ROS2 wrapper** to expand open the instructions.

https://github.com/IntelRealSense/realsense-ros?tab=readme-ov-file#installation-on-ubuntu

MohamedAlKayd, 2024-07-29:
@Nir-Az @MartyG-RealSense 
Thank you, I was able to get it working. I wanted to get topics published from the realsense2 on Live feed, Face Recognition, Object Recognition, Depth Map, and Contour Map. Are there existing functions that I can use or would it have to be developed from scratch?

[image]

MartyG-RealSense, 2024-07-30:
Typically you can find packages to add support in ROS for the functions that you need if they are not included by default in the RealSense ROS wrapper.  For example, for object detection / recognition **YOLO** is often used by RealSense users.  The link below has such a package.

https://github.com/exarchou/YOLO-Object-Recognition

Marco2929, 2025-12-11:
I set up the humble container but encountering the following error:

root@IPA-PC-327RFA5:/ws# ros2 launch realsense2_camera rs_launch.py depth_module.enable_auto_exposure:=false rgb_camera.enable_auto_exposure:=false align_depth.enable:=true
[INFO] [launch]: All log files can be found below /root/.ros/log/2025-12-11-10-22-59-892981-IPA-PC-327RFA5-220
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [221]
[realsense2_camera_node-1] 
[realsense2_camera_node-1] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[realsense2_camera_node-1] This error state is being overwritten:
[realsense2_camera_node-1] 
[realsense2_camera_node-1]   'string data is not null-terminated, at ./src/serdata.cpp:384'
[realsense2_camera_node-1] 
[realsense2_camera_node-1] with this new error message:
[realsense2_camera_node-1] 
[realsense2_camera_node-1]   'invalid data size, at ./src/serdata.cpp:384'
[realsense2_camera_node-1] 
[realsense2_camera_node-1] rcutils_reset_error() should be called after error handling to avoid this.
[realsense2_camera_node-1] <<<
[realsense2_camera_node-1] [INFO] [1765448579.957074889] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1765448579.957112104] [camera.camera]: Built with LibRealSense v2.54.2
[realsense2_camera_node-1] [INFO] [1765448579.957117971] [camera.camera]: Running with LibRealSense v2.54.2
[realsense2_camera_node-1] [INFO] [1765448580.025464007] [camera.camera]: Device with serial number f1421872 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1765448580.025591228] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1765448580.025618584] [camera.camera]: Device with name Intel RealSense L515 was found.
[realsense2_camera_node-1] [INFO] [1765448580.025976134] [camera.camera]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1765448580.026002581] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1765448580.029398155] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1765448580.029954092] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1765448580.029989714] [camera.camera]: Device Name: Intel RealSense L515
[realsense2_camera_node-1] [INFO] [1765448580.030011438] [camera.camera]: Device Serial No: f1421872
[realsense2_camera_node-1] [INFO] [1765448580.030033395] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1765448580.030056615] [camera.camera]: Device FW version: 1.5.4.1
[realsense2_camera_node-1] [INFO] [1765448580.030107618] [camera.camera]: Device Product ID: 0x0B64
[realsense2_camera_node-1] [INFO] [1765448580.030130039] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1765448580.041251544] [camera.camera]: Set ROS param depth_module.profile to default: 640x480x30
[realsense2_camera_node-1] [INFO] [1765448580.045481772] [camera.camera]: Set ROS param rgb_camera.profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1765448580.045891318] [camera.camera]: Set ROS param gyro_fps to default: 100
[realsense2_camera_node-1] [INFO] [1765448580.045986909] [camera.camera]: Set ROS param accel_fps to default: 200
[realsense2_camera_node-1] [WARN] [1765448580.056621675] [camera.camera]: Could not set param: pointcloud.stream_index_filter with -1 Range: [-1, -2147483648]: parameter 'pointcloud.stream_index_filter' could not be set: Parameter {pointcloud.stream_index_filter} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1765448580.056991399] [camera.camera]: Stopping Sensor: L500 Depth Sensor
[realsense2_camera_node-1] [INFO] [1765448580.066207283] [camera.camera]: Starting Sensor: L500 Depth Sensor
[realsense2_camera_node-1] [INFO] [1765448580.088864115] [camera.camera]: Open profile: stream_type: Confidence(0), Format: RAW8, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1765448580.088991016] [camera.camera]: Open profile: stream_type: Infra(0), Format: Y8, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1765448580.089035931] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1765448580.089362628] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1765448580.231661641] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1765448580.238726377] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1765448580.239432309] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1765448580.374271719] [camera.camera]: 
[realsense2_camera_node-1] [WARN] [1765448580.374410593] [camera.camera]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.

The topics are available, and the images of depth and color are visible in RViz via the Jazzy container, but they cannot be synchronized by time, and the depth image is not subscribable as a topic command. I am aware that this is officially no longer supported, but I am very interested in whether you think this is solvable. It seems that there is a problem with rmw_cyclonedd [... cut ...]
261.5 /ws/src/librealsense-2.51.1/src/libusb/libusb.h:13:82: error: extended character ’ is not valid in an identifier
261.5 /ws/src/librealsense-2.51.1/src/libusb/libusb.h:18:83: error: extended character ‘ is not valid in an identifier
261.5    18 | /usr/include/libusb-1.0/libusb.h:1260:4: warning: ISO C++ forbids zero-size array ‘iso_packet_desc’ [-Wpedantic]

But it worked with 4.54.1, that's also what I found in some repos on github

System/Setup:

- ROS2 Humble Container

- RealSense L515

- realsense2_camera v4.54.1

- librealsense2 v2.54.2

- RMW: rmw_cyclonedds_cpp

MartyG-RealSense, 2025-12-11:
I note that your L515's firmware driver version is 1.5.4.1 which is very old and was designed for use with librealsense 2.42.1.  For librealsense 2.54.2, the final L515 firmware version - **1.5.8.1** - is recommended.  As you have a mostly working setup with 2.54.2 / 4.54.1 it may not be necessary to update the firmware though.

If you add the parameter **enable_sync:=true**  to your launch instruction then the wrapper should gather the closest frames of different sensors (infrared, color and depth) to be sent with the same timetag.

The launch may be using the HARDWARE_CLOCK domain because on the L515 camera model, Global Time is disabled by default.  ROS has its own built-in time correction mechanism though which means that it may not be necessary to have Global Time enabled.  You could try adding **global_time_enabled:=true** to your launch instruction though to enable it.