realsenseai/realsense-ros #3110: No rgbd topic

pengzhenghao, 2024-05-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       |      D435i       | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |   Linux (Ubuntu 22)  | 
| Kernel Version (Linux Only)    |  (e.g. 5.4)                                                                       | 
| Platform                                 | PC + NVIDIA Jetson                            |
| Librealsense SDK Version       |  { 2.<?>.<?> }                                                    | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  {Iron/Humble/Rolling/etc.. }                                           | 
| RealSense ROS Wrapper Version   |  {4.51.1, 4.54.1, etc..}                                                  | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hi!

When I launched

```
ros2 launch realsense2_camera rs_launch.py depth_module.infra_profile:=848x480x15 rgb_camera.color_profile:=848x480x15 enable_rgbd:=true 
```

And `ros2 topic list | grep camera`, I only get

```
unitree@ubuntu:~$ ros2 topic list | grep camera
/camera/aligned_depth_to_color/image_raw
/camera/color/camera_info
/camera/color/image_raw
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/color/points
/camera/depth/image_rect_raw
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/imu
```

I tried to install `ros-foxy-rgbd-launch` but that package is not found with ros-foxy. I am expecting there is a new topic called `/camera/rgbd`? Thanks!

MartyG-RealSense, 2024-05-28:
Hi @pengzhenghao  The **RGBD** topic has a list of parameters that must be set to true in order to receive RGBD messages.  These are enable_rgbd, align_depth.enable, enable_sync, enable_depth and enable_color.

https://github.com/IntelRealSense/realsense-ros?tab=readme-ov-file#rgbd-topic

Please try the launch command below to see whether the RGBD topic is published.

`ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=848x480x15 depth_module.infra_profile:=848x480x15 rgb_camera.color_profile:=848x480x15 enable_rgbd:=true enable_sync:=true align_depth.enable:=true enable_color:=true enable_depth:=true `

pengzhenghao, 2024-05-28:
Hi, thanks for the reply! I have run the script with:

```bash
ros2 launch realsense2_camera rs_launch.py \
depth_module.depth_profile:=848x480x15 \
depth_module.infra_profile:=848x480x15 \
rgb_camera.profile:=848x480x15 \
enable_rgbd:=true \
enable_sync:=true \
align_depth.enable:=true \
enable_color:=true \
enable_depth:=true
```

And the valid topics are:

```
(ros_env) zhenghao@zhenghao-4090:~$ ros2 topic list | grep camera
/camera/aligned_depth_to_color/camera_info
/camera/aligned_depth_to_color/image_raw
/camera/aligned_depth_to_color/image_raw/compressed
/camera/aligned_depth_to_color/image_raw/compressedDepth
/camera/aligned_depth_to_color/image_raw/theora
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/theora
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/color/points
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/theora
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/imu
```

There is no RGBD topic. I wonder if it might be possible due to missing packages. So I've tried to install:
```bash
unitree@ubuntu:~$ sudo apt install ros-foxy-*rgbd*
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-foxy-*rgbd*
E: Couldn't find any package by glob 'ros-foxy-*rgbd*'
```

But it turn out that no packages are installed.

MartyG-RealSense, 2024-05-28:
There was a previous case at https://github.com/IntelRealSense/realsense-ros/issues/3070 where RGBD was not being published.  As the RGBD topic publishes depth and color in the same topic when align_depth.enable = true, I suggested to use the aligned depth-color topic `/camera/camera/aligned_depth_to_color/image_raw` instead of using the RGBD topic.

As your topics are named /camera and not /camera/camera, this suggests to me that your ROS2 wrapper may be outdated and not contain an update to how topics were named.  This could happen if you installed the wrapper from ROS packages instead of source code, as the source code procedure installs the latest wrapper version whilst the package version dates back to June 2023.

pengzhenghao, 2024-05-28:
Hi @MartyG-RealSense  Seems that the `ros-foxy-rgbd-launch` is missing for some reason. I am trying to workaround that. 

As I never know what the real RGBD message looks like, I am guessing it's a 4 channel or 6 channel images.

### 4 channel
If launch with
```bash
ros2 launch realsense2_camera rs_launch.py \
depth_module.profile:=848x480x15 rgb_camera.profile:=848x480x15 \
align_depth.enable:=true
```
[image]

### 6 channel
If launch with 
```bash
ros2 launch realsense2_camera rs_launch.py \
depth_module.profile:=848x480x15 rgb_camera.profile:=848x480x15 \
colorizer.enable:=true align_depth.enable:=true
```
[image]

### Questions

**The meaning of the image**: It's clear that in the colorized depth image the `black` means `out of detection scope.` But in the grayscale depth image, it seems `white` means far away? But all black means `out of the scope`? Would that sounds contradicting?

**4 or 6 channels in RGBD?**: In the native RGBD images, is it transmitting the 4 or 6 channel images?

**Do I launch correctly?**: As we can't transmit RGBD message for now, a workaround is to send two streams just like what I do above at the same time and combine them in the receiver end. So can I say that after using `align_depth.enable:=true` the `aligned_depth_to_color` and `color` streams are synced so I can just take these two streams?

**Any filter to be applied?**: This is a off-topic question sorry for putting it here. I saw there are many postprocessing filter such as hole-filling. I am not sure what's the best practice the community always use. Should I enable `temporal filter`, `spacial filter`, `decimation filter` and `hole filing filter` all together?

**Depth image black out**:  I sometime notice the depth images will black out for a few moment and return normal. Should I apply the temporal filter to avoid this?

Thank you again for the greatly and promptly help!

pengzhenghao, 2024-05-28:
Oh I just noticed we reply at the same time, Thanks @MartyG-RealSense I will checkout the latest ROS wrapper and get back to you!

pengzhenghao, 2024-05-28:
I am trying to build ROS wrapper from source, but it seems that `realsense2-gl` is missing...

```console
unitree@ubuntu:~/ros2_ws$ colcon build
Starting >>> realsense2_camera_msgs
Finished <<< realsense2_camera_msgs [4.56s]                     
Starting >>> realsense2_camera
Starting >>> realsense2_description             
Finished <<< realsense2_description [1.53s]                                                                
--- stderr: realsense2_camera                          
CMake Warning at CMakeLists.txt:122 (find_package):
  By not providing "Findrealsense2-gl.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "realsense2-gl", but CMake did not find one.

  Could not find a package configuration file provided by "realsense2-gl"
  (requested version 2.54.1) with any of the following names:

    realsense2-glConfig.cmake
    realsense2-gl-config.cmake

  Add the installation prefix of "realsense2-gl" to CMAKE_PREFIX_PATH or set
  "realsense2-gl_DIR" to a directory containing one of the above files.  If
  "realsense2-gl" provides a separate development package or SDK, be sure it
  has been installed.

CMake Error at /opt/ros/foxy/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:77 (message):
  ament_target_dependencies() the passed package name 'realsense2-gl' was not
  found before
Call Stack (most recent call first):
  CMakeLists.txt:256 (ament_target_dependencies)

---
Failed   <<< realsense2_camera [3.06s, exited with code 1]

Summary: 2 packages finished [11.5s]
  1 package failed: realsense2_camera
  1 package had stderr output: realsense2_camera
```

I just followed the instruction:

```
# Install ROS2 wrapper from source, don't do: sudo apt install ros-humble-realsense2-*
sudo apt remove ros-foxy-realsense2-*

mkdir -p ~/ros2_ws/src
cd ~/ros2_ws/src/
git clone https://github.com/IntelRealSense/realsense-ros.git -b ros2-development
cd ~/ros2_ws
sudo apt-get install python3-rosdep -y

# Foxy:
sudo rosdep init --include-eol-distros
sudo rosdep update --include-eol-distros

rosdep install -i --from-path src --rosdistro $ROS_DISTRO --skip-keys=librealsense2 -y

# Build
colcon build 
colcon build --cmake-args '-DBUILD_ACCELERATE_GPU_WITH_GLSL=ON'
```

pengzhenghao, 2024-05-28:
For the future reader who fails to do `colcon build --cmake-args '-DBUILD_ACCELERATE_GPU_WITH_GLSL=ON'` and build ROS wrapper in ros-foxy from source:

 
```
sudo apt-get install librealsense2-gl-dev
sudo apt-get install libglfw3-dev
```

I don't know why `librealsense2-gl-dev` is not install with `sudo apt install librealsense2-*` but this is it.

MartyG-RealSense, 2024-05-29:
Thanks so much @pengzhenghao for sharing your detailed advice for the benefit of other RealSense ROS users!